MichaIng/DietPi #7181: QBittorrent Zero Peer Connection

sureshk75, 2024-08-05:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `bookworm 0`
- SBC model | `RPi 5 Model B (aarch64)`
- Power supply used | `RPi 5 Official`
- SD card used | `SanDisk ultra 32Gb`

#### Additional Information (if applicable)
- Software title | `QBittorrent`
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi? `fresh install`

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. On startup or when restarts initiated by _dietpi-software_

#### Expected behaviour

- QBittorrent should start connecting to peers after the device starts up or when the qbittorrent-nox restarts via dietpi-software

#### Actual behaviour

- QBittorrent has no peer connections (0 DHT) resulting in stalled torrents. This is resolved by manually restarting the service 'systemctl restart qbittorrent.service`

#### Extra details

- This problem exists on RPi4 too when I was using it as my main device before. It wasn't much of a bother then (nor now actually) but figure I better highlight this for your reference.

sureshk75, 2024-08-05:
[image]

MichaIng, 2024-08-06:
Please check service logs after reboot:
```sh
journalctl -u qbittorrent
```
Do you use WiFi or Ethernet, and DHCP or a static IP? Maybe the network is not fully up yet when qBittorrent starts and it remains stuck in some false/none initialized state.

kwilkins, 2025-01-12:
I am also seeing this behavior. On device startup the qBittorrent client indicates the yellow fire icon (in op screenshot) that the connection status is "firewalled". However qbittorrent on my other machine works fine, no extra config on router was needed.

For my issue:
- I am using WiFi
- I am using DHCP
- If I run `systemctl restart qbittorrent.service`, the connection does not work.

The service logs after rebooting the service do not seem to indicate any issues.

```
Jan 12 07:03:42 DietPi systemd[1]: Started qbittorrent.service - qBittorrent (DietPi).
Jan 12 07:05:03 DietPi qbittorrent-nox[619]: Catching signal: SIGTERM
Jan 12 07:05:03 DietPi qbittorrent-nox[619]: Exiting cleanly
Jan 12 07:05:03 DietPi systemd[1]: Stopping qbittorrent.service - qBittorrent (DietPi)...
Jan 12 07:05:03 DietPi qbittorrent-nox[619]: WebUI will be started shortly after internal preparations. Please wait...
Jan 12 07:05:03 DietPi qbittorrent-nox[619]: ******** Information ********
Jan 12 07:05:03 DietPi qbittorrent-nox[619]: To control qBittorrent, access the WebUI at: http://localhost:1340
Jan 12 07:05:03 DietPi systemd[1]: qbittorrent.service: Deactivated successfully.
Jan 12 07:05:03 DietPi systemd[1]: Stopped qbittorrent.service - qBittorrent (DietPi).
Jan 12 07:05:03 DietPi systemd[1]: Started qbittorrent.service - qBittorrent (DietPi).
```

Any tips here?

MichaIng, 2025-01-13:
Did you open/forward port 6881?

DHT is enabled and shown in GUI, but with zero nodes 🤔. Probably the torrent does not support it, or there is really peer available?

kwilkins, 2025-01-14:
I didn't do any additional port configuration, I thought that would be handled automatically? I didn't have to do anything for my other machine with the router either.

I have gotten qBittorrent to work now though through a combination other things.
1. Have hardwired for ethernet, was always the plan but was configuring it for the first time over Wifi
2. Select the following settings in qBittorrent options
  - Advanced/Network interface: eth0 (default value is "Any interface")
  - Advanced/Optional IP address to bind to: All IPv4 addresses (default value is "All addresses")
[image]

I tried similar settings while using Wifi, selecting the other network interface "Io", but could not get it to work. Something wrong with the Wifi configuration qBittorrent thinks it has?