MichaIng/DietPi #7182: Error with "apt-get update"

DrCWO, 2024-08-05:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye

- Kernel version | `uname -a`
Linux rooExtend 5.15.52-rockchip64 #22.05.4 SMP PREEMPT Mon Jul 4 09:40:29 CEST 2022 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
NanoPi R2S (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
5V 15W

- SD card used | (EG: SanDisk ultra)
Sandisk ultra 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
apt

- Was the software title installed freshly or updated/migrated?
No, but worked until now on my customer's systems

- Can this issue be replicated on a fresh installation of DietPi?
I don't know, this is a production system distributed to many customers

#### Steps to reproduce

- Open ssh
- Enter:
`# apt-get update`

- Result:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://download.docker.com/linux/debian bullseye InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Ign:7 https://rooextend.com/rooextend-test-R2S agnostic InRelease
Hit:8 https://rooextend.com/rooextend-test-R2S agnostic Release
Ign:6 http://armbian.tnahosting.net/apt bullseye InRelease
Err:9 http://fi.mirror.armbian.de/apt bullseye Release
  404  Not Found [IP: 65.21.120.247 80]
Reading package lists... Done
E: The repository 'http://apt.armbian.com bullseye Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

#### Expected behaviour
No error message

#### Actual behaviour
Error message see above. My distibuted client systems dan't be updated any more

#### Extra details
This seems to be an issue that more than once happed and was more than once fixed.
I saw that in the net. But I have no idea to whom I can address this issue.
The ambios in git has no issue tracker :-(
Any ideas how you can help me so my customer's systems can be updated again?
My update script catches this error on the remote systems and stops working so I have
no chance to fix anything on the customer's systems out in the field :-(
I guess it is a repository error but who is able to fix it????
At the moment this is a dead end street for me and I'm a little desperate.

Joulinar, 2024-08-05:
Looks like an issue with the specific apt mirror of Armbian. You can wait and hope they will fix the missing release file as it seems to be missing for Bookworm as well.

DrCWO, 2024-08-05:
Do you have the chance to ping the Armbian guys and point them to this issue?

DrCWO, 2024-08-06:
Thank you for your effort. I also recognized today that it works again 👍

closed 2024-08-06 by DrCWO

MichaIng, 2024-08-06:
Update to latest DietPi, which replaces the Armbian APT server with our own.

DrCWO, 2024-08-07:
Great news 👍