MichaIng/DietPi #7187: DietPi-Software | Folding@Home - Automatic latest download URL failed

cruzadernl, 2024-08-07:
### Creating a bug report/issue
- [ X ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.6.1
- Distro version | 7
- Kernel version | 
- SBC model | Odroid N2(+)

#### Additional Information (if applicable)
- Software title | Folding@Home (list nr: 2)
- Was the software title installed:  freshly 
- Can this issue be replicated: yes

#### Extra details
The requested URL returned error 404

MichaIng, 2024-08-09:
Right, thanks for opening the issue. With client version 8, the download file structure changed. But version 8 changed a lot of other things, so me or someone needs to find time to rework it.

If you need a quick solution, it is possible to adjust the download URL to install the older version 7, which works with other parts of our setup.

Otherwise the `.deb` packages for version 8 can be downloaded here: https://foldingathome.org/start-folding/
It however requires to register a new account at their side, as far as I checked so far.

MichaIng, 2024-10-16:
I implemented the correct installation for v7 for now: https://github.com/MichaIng/DietPi/commit/d9b8a0b

MichaIng, 2024-10-16:
v8 implementation tracked in new issue: #7244

closed 2024-10-16 by MichaIng