MichaIng/DietPi #7188: Error after updating HomeAssistant

Dynamic5912, 2024-08-10:
After updating HomeAssistant to 2024.8.0 (and subsequently 2024.8.1), the following error message is displayed after completion of the upgrade:

**ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
aiohttp-fast-url-dispatcher 0.3.0 requires aiohttp<3.10,>=3.8.5, but you have aiohttp 3.10.2 which is incompatible.
Successfully installed aiohttp-3.10.2 homeassistant-2024.8.1**

Is anything required to be done here to resolve this?

Many thanks!

baslking, 2024-08-11:
Additionally this same package generates a steady stream of errors:
~~~
  Invalid method encountered:
Aug 11 09:37:46 odroid hass[2672591]: aiohttp.http_exceptions.BadStatusLine: 400, message:
Aug 11 09:37:46 odroid hass[2672591]:   File "aiohttp/_http_parser.pyx", line 557, in aiohttp._http_parser.HttpParser.feed_data
Aug 11 09:37:46 odroid hass[2672591]:                                    ^^^^^^^^^
Aug 11 09:37:46 odroid hass[2672591]:     assert self._request_parser is not None
Aug 11 09:37:46 odroid hass[2672591]:   File "/var/lib/hass/.venv/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 355, in data_received
Aug 11 09:37:46 odroid hass[2672591]: Traceback (most recent call last):
Aug 11 09:37:46 odroid hass[2672591]: 2024-08-11 09:37:46.583 ERROR (MainThread) [aiohttp.server] Error handling request
Aug 11 09:36:47 odroid hass[2672591]:       ^
Aug 11 09:36:47 odroid hass[2672591]:     b'\x10T'

~~~

Joulinar, 2024-08-11:
pls can you share the entire log and not just a part? This would be helpful.

And the missing system information

#### Required 
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- Architecture | `dpkg --print-architecture`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

MichaIng, 2024-08-11:
> `/var/lib/hass/.venv`

You did not install Home Assistant via `dietpi-software`, did you? The path does not match and we use `pyenv`, not `venv`.

No idea how an incompatible version of this particular module could have been installed in the first place. Maybe `aiohttp-fast-url-dispatcher` added the upper version limit retrospectively, after this incompatibility has been found. Go into your venv and downgrade `aiohttp` to v3.9 manually, then restart Home Assistant.

Dynamic5912, 2024-08-11:
> You did not install Home Assistant via `dietpi-software`, did you? The path does not match and we use `pyenv`, not `venv`.
>
>
>
> No idea how an incompatible version of this particular module could have [...]

And for me (OP)?

Let me know if you require any logs from my side.

MichaIng, 2024-08-12:
I just did a fresh install, and checked the module dependency versions pulled by the `homeassistant` module:
```console
homeassistant@VM-Bookworm:~$ pip3 freeze
...
aiohttp==3.10.2
aiohttp-cors==0.7.0
aiohttp-fast-zlib==0.1.1
...
async_upnp_client==0.40.0
-rw-r--r-- 1 homeassistant homeassistant   88 Aug 12 19:15 WHEEL
homeassistant@VM-Bookworm:~$ ls -dl /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp*
drwxr-xr-x 4 homeassistant homeassistant 4096 Aug 12 19:15 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp
drwxr-xr-x 2 homeassistant homeassistant 4096 Aug 12 19:15 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp-3.10.2.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 Aug 12 19:15 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_cors
drwxr-xr-x 2 homeassistant homeassistant 4096 Aug 12 19:15 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_cors-0.7.0.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 Aug 12 19:15 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_fast_zlib
drwxr-xr-x 2 homeassistant homeassistant 4096 Aug 12 19:15 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_fast_zlib-0.1.1.dist-info
```
So it is v3.10.2, and nothing prevents it from being this version. But I also do not have `aiohttp-fast-url-dispatcher`, also not in `/mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/` (where modules installed by Home Assistant itself are stored).

As can be seen `async_upnp_client` is v0.40.0, which has no conflict with the `aiohttp` version, and updating it was also the solution for the similar report on HA forum.

@Dynamic5912 
Can you check this:
```sh
sudo -u homeassistant bash
source /home/homeassistant/pyenv-activate.sh
pip3 freeze | grep 'aiohttp'
ls -dl /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp*
```

Dynamic5912, 2024-08-12:
> I just did a fresh install, and checked the module dependency versions pulled by the `homeassistant` module:
>
> ```console
>
> homeassistant@VM-Bookworm:~$ pip3 freeze
>
> ...
>
> aiohttp==3.10.2 [...]

As requested:

homeassistant@dietpi:~$ pip3 freeze | grep 'aiohttp'

aiohttp==3.10.2
aiohttp-cors==0.7.0
aiohttp-fast-url-dispatcher==0.3.0
aiohttp-fast-zlib==0.1.1
aiohttp-isal==0.3.1
aiohttp-session==2.12.0
aiohttp-zlib-ng==0.3.1

homeassistant@dietpi:~$ ls -dl /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp*

drwxr-xr-x 4 homeassistant homeassistant 4096 Aug 10 19:36 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp
drwxr-xr-x 2 homeassistant homeassistant 4096 Aug 10 19:36 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp-3.10.2.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 Feb 25 21:46 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_cors
drwxr-xr-x 2 homeassistant homeassistant 4096 Feb 25 21:46 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_cors-0.7.0.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 Feb 25 21:46 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_fast_url_dispatcher
drwxr-xr-x 2 homeassistant homeassistant 4096 Feb 25 21:46 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_fast_url_dispatcher-0.3.0.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 Jul  3 19:58 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_fast_zlib
drwxr-xr-x 2 homeassistant homeassistant 4096 Jul  3 19:58 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_fast_zlib-0.1.1.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 May  7 07:50 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_isal
drwxr-xr-x 2 homeassistant homeassistant 4096 May  7 07:50 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_isal-0.3.1.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 May  1 20:14 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_session
drwxr-xr-x 2 homeassistant homeassistant 4096 May  1 20:14 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_session-2.12.0.dist-info
drwxr-xr-x 3 homeassistant homeassistant 4096 Feb 25 21:46 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_zlib_ng
drwxr-xr-x 2 homeassistant homeassistant 4096 Feb 25 21:46 /mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/aiohttp_zlib_ng-0.3.1.dist-info_

MichaIng, 2024-08-12:
Ah, I missed that we create `/mnt/dietpi_userdata/homeassistant/deps` as symlink to `/home/homeassistant/.pyenv/versions/3.*` in the meantime, so that both contain the same packages. That makes it easier. Please try to remove `aiohttp-fast-url-dispatcher` and paste the output of `pip3`. It would say if a dependency is missing:
```sh
pip3 uninstall aiohttp-fast-url-dispatcher
exit # exit the sudo session
```
If it does not throw any error, restart HA, access its web interface, and see whether you still see the error. If any integration you use needed `aiohttp-fast-url-dispatcher`, it would be installed automatically:
```sh
sudo systemctl restart home-assistant
```

Dynamic5912, 2024-08-12:
OK that's removed now and HA restarted.

No errors thus far and it hasn't reinstalled again...

MichaIng, 2024-08-12:
Okay great. It is possible that it gets re-added when you actually use particular integrations or their GUI parts. I am not 100% sure how/when the HA internal automatic dependency installs are done in all cases. So just in case it gets, and you see again the error, please paste again the output of
```sh
sudo -u homeassistant bash
source /home/homeassistant/pyenv-activate.sh
pip3 freeze | grep 'aiohttp'
```
to verify the version incompatibility, and try to remember the integration you use last. It is not impossible that there is simply a conflict between bare HA deps and an integration's deps, or that a lower `aiohttp` is possible, but it is not automatically downgraded if needed for some later installed integration (Python module). `pip` at least indeed allows to install/uninstall modules even if there are conflicts or unresolved dependencies, opposed to how strict `apt` handles this. We can then try whether a manual downgrade works, and in case report to the developer of the integration, asking for an update of the `aiohttp-fast-url-dispatcher` dependency, to avoid this kind of conflict with HA core.

I'll however mark the issue as closed for now.

closed 2024-08-12 by MichaIng