MichaIng/DietPi #7189: Continuous issue dmesg warning about unprivileged eBPF being enabled and vulnerable..despite disabling.

zappydood, 2024-08-14:
### Creating a bug report/issue
- [yes] I have searched the existing open and closed issues

#### Required Information
- DietPi Version | cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro Version | Bookworm

- Kernel Version | uname -a
Linux DietPiii 5.10.160-legacy-rk35xx #1 SMP Wed May 15 03:04:45 UTC 2024 aarch64 GNU/Linux

- SBC Model | echo $G_HW_MODEL_NAME
Orange Pi 5 Plus (aarch64)

- Power Supply Used | 
Included 5V 4A USB power adapter (purchased from Amazon, provided by the Orange Pi seller)

- SD Card Used | 
None (Operating system is installed on NVMe for better speed)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
None specifically.
- Was the software title installed freshly or updated/migrated?
n/a
- Can this issue be replicated on a fresh installation of DietPi?
Not sure, because its my only opi5+ and I don't have this issue on my nanopi r5c's

- Bug report ID | `echo $G_HW_UUID`
n/a

#### Steps to reproduce

Honestly this popped up a while ago and I've been trying to get it to go AWAY, but no matter how many times I set it in sysctl and various other changes, it will continue to alert me in dmesg:
"WARNING: Unprivileged eBPF is enabled, data leaks possible via Spectre v2 BHB attacks!"
Each time its listed 3 times in a row in dmesg. every. single. time. And it hasn't gone away since setting it to disabled. I added it as extraargs in dietpiEnv.txt and upon reboot its STILL doing it in the same exact way it has been three warnings in dmesg in a row. 

#### Expected behaviour

It shouldn't be enabled, as far as I know, and as far as I can tell I've disabled it in various ways but the persistent warning in dmesg is driving me nuts at this point because I can't figure out what the cause is...
#### Actual behaviour

Persistent warnings about unprivileged eBPF being enabled and vulnerable to spectre v2 despite disabling it many times now, and confirming it shows as disabled. 
#### Extra details

Honestly this issue has had me stumped. And Its been an occurence in the logs for a while, and I really need some help figuring out what the cause of this is from. I currently only run AdGuard Home, Samba Server, and use my two ethernet ports bonded as a failover since my switch doesnt support LA anyway. It works well, otherwise, but this will NOT go away. I've been near just backing up all my confs individually and wiping but I really really want to avoid doing that if possible and can't find the time sometimes just to remember to login and do an apt update, let alone nuking it and starting fresh.

MichaIng, 2024-08-14:
Is it the same with the vendor kernel?
```sh
apt install linux-{image,dtb}-vendor-rk35xx
reboot
```
We will offer the upgrade anyway with upcoming release.

zappydood, 2024-08-18:
Yes!! thank you man, you and dietpi are just awesome. It immediately fixed it. So it looks like as per usual, dietpi is doing great and will continue as has been my experience. Truly appreciate your time and I really appreciate your work. I notice how much you work on this all and it makes me happy to know that there is support even when its a silly issue like mine. :)

closed 2024-08-18 by zappydood