MichaIng/DietPi #7190: Slow loading of the home assistant after installation

gabochko, 2024-08-16:
Hello, I have the latest version of dietpi at the moment. The problem is that I installed home assistant, and through htop I monitored the download speed of additional files. Raspberry pi zero board, not W with a shield with internet and USB ports. In a day and a half, I have downloaded only 57mb of 1.5gb. Help please

gabochko, 2024-08-16:
Maybe there is some command that will accelerate the speed

gabochko, 2024-08-17:
-m pip install --quiet zeroconf==0.132.2 --timeout 60 --upgrade

MichaIng, 2024-08-23:
You mean Home Assistant is downloading additional Python modules/dependencies very slowly? Did you compare the download speed with some CLI tool like `curl` or `wget`? If for the `zeroconf` module on an RPi Zero (1) e.g.:
```sh
cd /tmp
curl https://piwheels.org/simple/zeroconf/zeroconf-0.132.2-cp311-cp311-manylinux_2_36_armv7l.whl
```

closed 2025-06-02 by MichaIng