MichaIng/DietPi #7192: nextcloud memories, Missing Line in nginx-nextcloud.conf

droogi, 2024-08-19:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
- dietpi 9.6

#### Additional Information (if applicable)
- nextcloud memories

#### Steps to reproduce
Install nginx, nextcloud,nc  App memories,  lets-encrypt

#### Expected behaviour
Memories should Display thumbs and images

#### Actual behaviour
Thumbs are generated, but are Gray and images are Not displayed

#### Extra details
Solution is to add Line in /etc/nginx/sites-available/dietpi-nextcloud.conf

    add_header Cross-Origin-Embedder-Policy credentialless;

I added It twice, IT works AS IT should now

Regards, droogi

droogi, 2024-08-22:
Maps Work again also (App Maps opened, but No layer was displayed)

MichaIng, 2024-08-23:
Hmm, this does not seem to be supported widely among browsers, e.g. Safari and certain mobile browsers do not support it at all:
- https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Cross-Origin-Embedder-Policy
- https://caniuse.com/?search=Cross-Origin-Embedder-Policy%3A%20credentialless

Isn't it possible to do all this with CSP? Nextcloud has an API for this, which can be used by apps, if I am not mistaken.

This does not seem suitable to add for all Nextcloud instances, also it should be probably added for the very specific URLs used by this app only, instead of disabling CORS entirely for the whole Nextcloud instance. This all can be done much better with CSP from the backend/app code, instead of globally via webserver configuration. I suggest you talk to its developer/maintainer about this.

Also, why did you add it twice?

droogi, 2024-10-01:
I added it twice in two locations in this config file: `/etc/nginx/sites-dietpi/dietpi-nextcloud.conf`
1, location `^~ /nextcloud {`  
2, location `~ \.(?:css|js|mjs|svg|gif|png|jpg|ico|wasm|tflite|map)$ {`

Yesterday i updated to nextcould 30.0.0 (via beta channel) in dietpi 9.7.1
I commented (#) both entries in the config file and restarted nginx, 
- memories displays both preview and images **as expected**.
- maps (and phonetrack) do **not** display tiles
  maps works, when `add_header Cross-Origin-Embedder-Policy credentialless;` is added to  location `^~ /nextcloud {` 

I did not know about CSP. I did not touch any nextcloud files (but the nextcloud.conf). CSP editor says " no custom policy set".  Isn´t the dietpi-nextcloud.conf only for nextcloud but not for global webserver config?

MichaIng, 2024-10-03:
I just did a test installation of Nextcloud with Nginx, installed the maps app, and it works OOTB here, if you mean map tiles:

[image]

Can you check in your browser console/developer tools, which rule exactly denies loading of this resources? Probably your Nginx globally overrides CSP or so, breaking the apps.

droogi, 2024-10-03:
I tried it with two browsers and i thinks these are the errors for not loading tiles: 

**firefox131.0:** multiprocess: errors: `Die Ressource unter "https://b.tile.openstreetmap.org/10/543/346.png" wurde aufgrund ihres Cross-Origin-Resource-Policy-Headers (oder dessen Fehlens) blockiert. Weitere Informationen unter https://developer.mozilla.org/docs/Web/HTTP/Cross-Origin_Resource_Policy_(CORP)#`

**edge 129.0.2792.65:** `Failed to load resource: net::ERR_BLOCKED_BY_RESPONSE.NotSameOriginAfterDefaultedToSameOriginByCoep` [https://c.tile.openstreetmap.org/9/284/168.png]

Copilot says Copilot
Antworten für Sie werden generiert…

Gesendet von Copilot:

> 
> Hier ist eine Erklärung der Fehlermeldung aus der DevTools-Konsole:
> 
> Fehlermeldung: Die Meldung “Failed to load resource: net::ERR_BLOCKED_BY_RESPONSE.NotSameOriginAfterDefaultedToSameOriginByCoep” bedeutet, dass eine Ressource aufgrund von Cross-Origin Resource Sharing (CORS)-Richtlinien blockiert wurde.
> Ursache: Diese Fehlermeldung tritt auf, wenn eine Ressource von einer anderen Domain geladen wird und die Cross-Origin-Embedder-Policy (COEP) nicht korrekt konfiguriert ist.
> Lösung: Stellen Sie sicher, dass der Server die richtigen CORS-Header sendet, insbesondere Access-Control-Allow-Origin. Überprüfen Sie auch die COEP-Einstellungen auf dem Server.

MichaIng, 2024-10-03:
Hmm, that header should not be set by default. Can you check this:
```sh
grep -ri 'Cross-Origin-Resource-Policy' /etc/nginx
```

droogi, 2024-10-03:
`grep -ri 'Cross-Origin-Resource-Policy' /etc/nginx ` gives no result

MichaIng, 2024-10-03:
Hmm, and when you run this:
```sh
curl -Ik https://localhost
curl -Ik https://localhost/nextcloud/
```

droogi, 2024-10-03:
curl -Ik https://localhost
```

> HTTP/2 200
> server: nginx
> date: Thu, 03 Oct 2024 20:56:48 GMT
> content-type: text/html; charset=utf-8
> content-length: 82
> last-modified: Thu, 30 May 2024 19:34:06 GMT
> etag: "6658d4ae-52"
> accept-ranges: bytes

```

curl -Ik https://localhost/nextcloud/
```
HTTP/2 302
server: nginx
date: Thu, 03 Oct 2024 20:57:21 GMT
content-type: text/html; charset=UTF-8
location: https://localhost/nextcloud/login
set-cookie: oc_sessionPassphrase=1tShT%2BFLEUg7r4PzumeXj4C0bmJcf4uDQw11nKZtaBNl8mciTdOxhOXp2gXwaGpezxZeoG9DcDjd7zak7tgOfWg1ch3qDHWLG2CoqPCd5%2Bj%2BBZr1j%2BanNAbGRNxvR0EN; path=/nextcloud; secure; HttpOnly; SameSite=Lax
content-security-policy: default-src 'self'; script-src 'self' 'nonce-K26Up9dkRrVcH3jAeJi01TiAGxILuJRKelbilnvEPRw='; style-src 'self' 'unsafe-inline'; frame-src *; img-src * data: blob:; font-src 'self' data:; media-src *; connect-src *; object-src 'none'; base-uri 'self';
set-cookie: nc_sameSiteCookielax=true; path=/nextcloud; httponly;secure; expires=Fri, 31-Dec-2100 23:59:59 GMT; SameSite=lax
set-cookie: nc_sameSiteCookiestrict=true; path=/nextcloud; httponly;secure; expires=Fri, 31-Dec-2100 23:59:59 GMT; SameSite=strict
set-cookie: oc7tw63zjg8i=ud4nal0jtcci8a66hbo0pu638m; path=/nextcloud; secure; HttpOnly; SameSite=Lax
referrer-policy: no-referrer
x-content-type-options: nosniff
x-download-options: noopen
x-frame-options: SAMEORIGIN
x-permitted-cross-domain-policies: none
x-robots-tag: noindex, nofollow
x-xss-protection: 1; mode=block
```

droogi, 2024-10-03:
i restarted service nginx, memories do not load again, no previews for all pictures, but grey boxes are displayed. When clicking the box on top (last year), the image is displayed. When clicking boxes on bottom (timeline), no image is displayed.

MichaIng, 2024-10-03:
Hmm, no `Cross-Origin-Resource-Policy` or `Cross-Origin-Embedder-Policy` set. Do you use a proxy to access your Nextcloud instance, which could add one of these headers?

droogi, 2024-10-04:
There ist pihole running on the same Pi. IT IS configured in the Fritzbox as DNS.

droogi, 2024-10-04:
There is pihole on the same device. It is configured in the Fritzbox as DNS.
I use lets-encrypt to enable https only.

MichaIng, 2024-10-04:
When opening the app in browser, can you check the response headers like this

[image]

Hit F12 to open developer tools, switch to Network tab, in case reload page with F5. The the top of the request list, select `maps/`, then select "Headers" tab in the newly opened frame. Take a screenshot of the headers section, but please mask/remove some identifiers for your privacy, like the request ID, request URL and remote address.

As you can see in my case, there is no `Cross-Origin-Resource-Policy-Headers` header set, and the CSP header explicitly permits to embed the needed `openstreetmap.org` URLs.

droogi, 2024-10-05:
i cannot find the left side bar (Name), i tried with edge and firefox
[image]

mistakes: 

- The source list for the Content Security Policy directive 'connect-src' contains an invalid source: '://'. It will be ignored.

- c.tile.openstreetmap.org/11/1086/692.png:1                

`        GET https://c.tile.openstreetmap.org/11/1086/692.png net::ERR_BLOCKED_BY_RESPONSE.NotSameOriginAfterDefaultedToSameOriginByCoep 200 (OK)`

droogi, 2024-10-05:
it is preferences- network - group networkprotokol by frame
Cross-Origin-Embedder-Policy: require-corp
Cross-Origin-Opener Policy: same-origin

[image]

droogi, 2024-10-05:
I searched for Cross-Origin-Embedder-Policy: require-corp and came to 
[richdocuments](https://github.com/nextcloud/richdocuments/issues/3258)

So i disabled and uninstalled Nextcloud office 8.5.1 and both memories and maps work as expected.
I installed again, both apps do not load external content, when disabling Nextcloud Office, they work fine.

MichaIng, 2024-10-07:
Oh great find, so it is the Nextcloud Office app which sets this globally. That is of course a problem. I'll check back with those guys, as this seems to be wrong to me.

MichaIng, 2025-06-02:
Closing this issue. Let's try to push the issue at Nextcloud Office where it belongs.

closed 2025-06-02 by MichaIng