MichaIng/DietPi #7195: Please add support for Orange Pi 5 Max

DiegoBM, 2024-08-25:
### Creating an image request
Orange Pi released not long ago the Orange Pi 5 Max, which is similar to the 5 plus but with wifi and bluetooth already included

#### Formal device information
- Device name | Orange Pi 5 Max
- Official product URL | http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-Max.html
- GitHub resource URL (if available) | https://github.com/orangepi-xunlong/orangepi-build
- Image download URL | http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-5-Max.html

#### Is the SBC officially supported by the Debian installer?
- Not that I know?

#### If not, is a reliable 3rd party Debian image available for this SBC?
- [This one is the reference implementation](https://github.com/Joshua-Riek/ubuntu-rockchip)

#### If not, are there install instructions for Debian available?
- N/A

DiegoBM, 2024-08-25:
Hehehe same here, I was about to start working with it and got to my attention how well DietPi worked with Orange Pi 3B, so I was considering using it my set of 5 Maxs

MichaIng, 2024-10-17:
First step done: https://github.com/MichaIng/DietPi/commit/a4d273d

StephanStS, 2024-12-23:
Doc change: See https://github.com/MichaIng/DietPi-Docs/pull/1040

closed 2024-12-25 by MichaIng