MichaIng/DietPi #7198: dietpi-software has incorrect download URL version for nomachine

tzvi208, 2024-08-26:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.6.1
- Distro version | bookworm 0
- Kernel version | `Linux rpi5-01 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`
- SBC model | RPi 5 Model B (aarch64)
- Power supply used | n/a
- SD card used | SanDisk 256GB
#### Additional Information (if applicable)
- Software title | nomachine
- Was the software title installed freshly or updated/migrated? **No**
- Can this issue be replicated on a fresh installation of DietPi? **Probably**

- Bug report ID | d2f560e4-1fd5-47a4-ae41-52fae7d8c10f

#### Steps to reproduce

1. `dietpi-software install 30`

#### Expected behaviour

- `dietpi-software` should install **nomachine**.

#### Actual behaviour

- installation fails
[dietpi-software_install_30.txt](https://github.com/user-attachments/files/16753710/dietpi-software_install_30.txt)
```
Reading package lists...
E: Invalid archive signature
E: Internal error, could not locate member control.tar{.zst,.lz4,.gz,.xz,.bz2,.lzma,}
E: Could not read meta data from /tmp/DietPi-Software/nomachine_8.11.3_3_arm64.deb
E: The package lists or status file could not be parsed or opened.
[FAILED] DietPi-Software | APT install ./nomachine_8.11.3_3_arm64.deb
```

#### Extra details

| what | URL | Comment |
| --- | --- | --- |
| URL asserted by `dietpi-software` | `https://download.nomachine.com/download/8.11/Arm/nomachine_8.11.3_3_arm64.deb` | downloads an HTML page pointing to v8.13 |
| URL that succeeds | `https://download.nomachine.com/download/8.13/Arm/nomachine_8.13.1_1_arm64.deb` | from `https://downloads.nomachine.com/download/?id=114&distro=ARM` |

MichaIng, 2024-08-26:
Many thanks for reporting. Fixed with: https://github.com/MichaIng/DietPi/commit/40050a2

This is the first time they really removed the old downloads. Makes it more valuable to detect the latest available version automatically, but so far I could not find a good API or method to do that for their website, also since different architectures have different download pages, and sometimes different latest versions as well.

tzvi208, 2024-08-27:
> This is the first time they really removed the old downloads. Makes it more valuable to detect the latest available version automatically, but so far I could not find a good API or method to do that [...]

Would it be reasonable for `dietpi-software` to parse the href value from `https://downloads.nomachine.com/download/?id=114&distro=ARM` for the href in `<a id="link_download" href="YADA">Download</a>` each time it installs **nomachine**?

tzvi208, 2024-08-29:
Based on a conversation I had with Nomachine support, here are static URLs for the latest versions:
| `uname -m` | URL |
| --- | --- |
| i386 | https://www.nomachine.com/free/linux/32/deb/nomachine_latest_i386.deb |
| amd64 | https://www.nomachine.com/free/linux/64/deb/nomachine_latest_amd64.deb |
|armhf | https://www.nomachine.com/free/arm/v7/deb/nomachine_latest_armhf.deb |
| arm64 | https://www.nomachine.com/free/arm/v8/deb/nomachine_latest_arm64.deb | 

Are there any other architectures you need?

MichaIng, 2024-08-29:
Thanks a lot! Did these exist already did they just set it up? Sometimes it just makes sense to ask 🙂.

tzvi208, 2024-08-29:
> We need the `armv6hf` architecture as well, for Raspberry Pi 1 and Zero models.

I do not see  `armv6hf` in the list of architectures supported by Debian, please see [here](https://packages.debian.org/search?suite=bookworm&arch=armel&searchon=names&keywords=gnome).
Searching raspbian.org for `armv6hf` returns only [one](https://www.google.com/search?q=armv6hf+site%3Araspbian.org&oq=armv6hf+site%3Araspbian.org&gs_lcrp=EgZjaHJvbWUyBggAEEUYOdIBCTExNjIxajBqNKgCALACAQ&sourceid=chrome&ie=UTF-8) hit.

What is the return from `dpkg --print-architecture` on these models running bookworm?

tzvi208, 2024-08-29:
> Did these exist already, or did they just set it up? Sometimes it just makes sense to ask 🙂.

I don't see these URL's on their web site, but I would be astonished if they set them up only on my query.

MichaIng, 2024-08-29:
This is why "Raspbian" exists 😉. Here are the NoMachine downloads for these RPi models, respectively such which use the Raspbian package repository (instead of Debian): https://downloads.nomachine.com/linux/?id=29&distro=Raspberry

tzvi208, 2024-08-29:
> This is why "Raspbian" exists 😉. Here are the NoMachine downloads for these RPi models, respectively such which use the Raspbian package repository (instead of Debian): https://downloads.nomachine.com/linux/?id=29&distro=Raspberry

- Perhaps this was the case at one time, but today I examined all the V6 DEB links [there](https://downloads.nomachine.com/linux/?id=29&distro=Raspberry) and they all point to download.nomachine.com.
- Likewise, I manually examined [raspbian.org](https://archive.raspbian.org/raspbian/) and failed to find any mention nomachine there.

Meanwhile, I continued my conversation with the Support Team at NoMachine support to inquire if they have a static link for ARM6 packages; I'll report their reply.

MichaIng, 2024-08-29:
@tzvi208 
Not sure what you want to say with that? What's wrong with `download.nomachine.com`? Of course Raspbian does not distribute NoMachine, since it's closed source commercial product, and as we know now, not allowed to be redistributed without permission. Raspbian is, like Debian, a source code repository, with binary packages compiled by the distributors. They do not distribute packages compiled elsewhere. NoMachine however provides packages compatible with the special ARMv6 hardware-floating-point architecture of the first RPi models, like Raspbian does, but Debian not.

tzvi208, 2024-08-31:
Hi @MichaIng

I misunderstood your remark:
> This is why "Raspbian" exists 😉. Here are the NoMachine downloads for these RPi models, respectively such which use the Raspbian package repository (instead of Debian): https://downloads.nomachine.com/linux/?id=29&distro=Raspberry

When you mentioned **Raspibian**, I thought you were talking about `http://archive.raspbian.org/raspbian`. Now, I understand that you're referring to what NoMachine calls a `distro=Raspberry`. Sorry about the confusion :confused: : .

NoMachine does offer both [`distro=Arm`](https://downloads.nomachine.com/linux/?id=30&distro=Arm) and  [`distro=Raspberry`](https://downloads.nomachine.com/linux/?id=29&distro=Raspberry)
When examining these distros, I find the same two architectures in each distro: `armhf` and `arm64`.

Also, the package that NoMachine offers for `raspberry_v6` is identical to `raspberry_v7`, including the fact that both assert `Architecture: armhf` in their respective package control files.

Although I do find mention of `armv6hf` in this  [Knowledge Base](https://kb.nomachine.com/AR07N00896) article from 2016, I don't think NoMachine supports ARMv6 anymore. :disappointed:

tzvi208, 2024-09-01:
Hi @MichaIng 

This is just a separate comment to generate notifications that I significantly revised these two previous comments, and that you should review them before closing the implementation of any fix from this issue #7198:

- [2024-08-29](https://github.com/MichaIng/DietPi/issues/7198#issuecomment-2316776183) filled in table of packages.
- [2024-08-31](https://github.com/MichaIng/DietPi/issues/7198#issuecomment-2323018143) corrected mistaken conclusions.

MichaIng, 2024-09-01:
Many thanks. I edited the last line where I think you wanted to show that as well the "arm" and "raspbian" ARMv8 downloads are the same. This is also what we found (and which makes just sense since RPi 2 and above are ARM fully compatible with plain Debian) and hence used the "arm" downloads for all ARMv7 and ARMv8 boards.

MichaIng, 2024-09-01:
Implemented: https://github.com/MichaIng/DietPi/commit/3942854

Hmm, now we have a bug fix and enhancement about the very same thing 😄. Well, it is both indeed. The code for installing NoMachine is now ridiculously simple/slim, not that I want to complain. It might grow again since we might need to add some EULA/licensing/personal-use-only notice, to get an official permission distributing it.

closed 2024-10-18 by MichaIng