MichaIng/DietPi #7202: Unable to run dietpi-update on old DietPi version (v8.14.2)

abstractionmage, 2024-08-27:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
Unknown 0
```
- Kernel version | `uname -a`
```
Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
RPi 4 Model B (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)
```
Official Raspberry Pi 4B power supply
```
- SD card used | 
```
Intenso micro SD XC 1CLASS 10 128 GB
```

#### Additional Information (if applicable)
- Software title | dietpi-update
- Was the software title installed freshly or updated/migrated? | updated
- Can this issue be replicated on a fresh installation of DietPi? | obviously not, but that's not the point

#### Steps to reproduce

1. Install DietPi v8.14.2 on an arm-based system
2. Run `dietpi-update` as root

#### Expected behaviour

- The DietPi update should succeed

#### Actual behaviour

- The DietPi update fails because it tries to fall back onto a no longer existing git branch called `jessie-support`

#### Extra details
I also can't update any debian packages anymore because they have all automatically been held back.

Here's the full output log:
```
root@DietPi:/home/dietpi
$ dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/stretch/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 7.9.4
[ INFO ] DietPi-Update | Your Debian version is too low to update to the current branch: Unknown (0)
[ INFO ] DietPi-Update | We're switching to an alternative branch: jessie-support
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/jessie-support/.update/version
curl: (22) The requested URL returned error: 404
[FAILED] DietPi-Update | Unable to get latest version.
[ INFO ] DietPi-Update | Please check the target Git repository and your network connection, then rerun dietpi-update.
[ INFO ] DietPi-Update | If this error persists, please report at: https://github.com/MichaIng/DietPi/issues
[ INFO ] DietPi-Update | APT update, please wait...
Hit:2 http://ftp.de.debian.org/debian sid InRelease
Hit:3 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:4 https://deb.debian.org/debian bullseye InRelease
Hit:5 http://download.zerotier.com/debian/buster buster InRelease
Hit:6 https://deb.debian.org/debian bullseye-updates InRelease
Hit:7 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:8 https://deb.debian.org/debian bullseye-backports InRelease
Hit:9 https://download.docker.com/linux/debian bookworm InRelease
Hit:1 https://packages.kopia.io/apt stable InRelease
Reading package lists...
W: http://download.zerotier.com/debian/buster/dists/buster/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | Storing number of available APT upgrades to file: /run/dietpi/.apt_updates
```

So since the `dietpi-update` command no longer works, how can I update the dietpi version on my raspberry pi now?

MichaIng, 2024-08-27:
Hmm, you have a very mixed OS, actually I am unsure which Debian version it really is, but there are APT sources for Sid/testing, Bullseye and Buster mixed, and our script is not able to detect the version.

Can you show the following:
```sh
cat /etc/debian_version /etc/os-release
dpkg -l base-files
```

abstractionmage, 2024-08-27:
Of course:
```
$ cat /etc/debian_version /etc/os-release
trixie/sid
PRETTY_NAME="Debian GNU/Linux trixie/sid"
NAME="Debian GNU/Linux"
VERSION_CODENAME=trixie
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
$ dpkg -l base-files
sh: 1: less: not found
dpkg-query: error: showing package list on pager subprocess returned error exit status 127
```

```
$ ls -l /etc/apt/sources.list.d
total 16
-rw-r--r-- 1 root root 62 Mar 11  2023 docker.list
-rw-r--r-- 1 root root 94 Sep 29  2023 kopia.list
-rw-r--r-- 1 root root 58 Aug 12  2021 raspi.list
-rw-r--r-- 1 root root 59 Sep 17  2021 zerotier.list
```

Joulinar, 2024-09-03:
can you share content of your source list

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

MichaIng, 2024-09-03:
You have `http://ftp.de.debian.org/debian sid` in your `sources.list`, and the OS hence seems to have upgraded to Debian Sid/unstable (shown as upcoming release Trixie/sid). DietPi v8.14 is indeed not able to detect this Debian version yet. This was added with v8.19.

We could manually put support inside:
```sh
G_CONFIG_INJECT 'G_DISTRO=' 'G_DISTRO=8' /boot/dietpi/.hw_model
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
dietpi-update 1
```
should do it. But the major problem is that you are on an unstable Sid system, which is highly discouraged, volatile, potentially insecure, basically a playground for developers, not intended to be used by end users. Did you add `http://ftp.de.debian.org/debian sid` intentionally to your `sources.list` or did some misleading guide instruct to do so?

Since Debian cannot really be downgraded, I would actually set up a fresh DietPi Bookworm system. You could also remove the Debian Sid entry from APT sources, and replace the Bookworm sources with Trixie ones. If you did not run `apt upgrade` for a longer time, most packages should be upgraded to Debian Trixie/testing (which is much better than Sid/unstable), a safer and more stable version.

abstractionmage, 2024-09-21:
Thanks, with the commands above I was indeed able to update to the latest version of DietPi (v9.7.1).

And you're right, I don't remember when or why but it appears that I have indeed added sid sources to my `/etc/apt/sources.list` at some point to get something to work. 

Since I haven't updated my system for a few months at this point I removed the sid sources and replaced bookworm sources with trixie ones as you suggested, which seems to have worked for the most part. I still ran into a few dependency problems when upgrading my system, but nothing that I couldn't resolve by purging a few packages I didn't need anymore anyway (mainly GTK and QT libraries and the programs depending on them, but on my headless raspberry pi I don't really need those anyway).

Thanks again, I just want to say that I'm deeply grateful for your support and your helpful suggestions. This level of assistance really goes above and beyone everything I could've imagined from the owner of a somewhat popular linux distro 💙

closed 2024-09-21 by abstractionmage

MichaIng, 2024-09-23:
Great that it worked. I can imagine a bunch of the package conflicts was due to the 64-bit time_t transition Debian currently does: https://wiki.debian.org/ReleaseGoals/64bit-time
Basically all libraries are renamed to have a `t64` suffix. I fear the moment when all this renaming is reverted 🙈.