MichaIng/DietPi #7207: Jellyfin v10.9.10 on DietPi 9.7 not starting up

sureshk75, 2024-09-02:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm 0`
- Kernel version | `Linux myrpi5 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`
- SBC model | `RPi 5 Model B (aarch64)`
- Power supply used | 5V 5A RPi5 Official
- SD card used | SanDisk ultra 32Gb

#### Additional Information (if applicable)
- Software title | Jellyfin Media Server (Fresh)

#### Steps to reproduce
Brand new installation. Besides the defaults from DietPi 9.7, only **mdadm** was installed before proceeding to install Jellyfin via dietpi-software.

#### Expected behaviour
Jellyfin should work without hitches.. it worked before with v10.9.9 and even got upgraded to v10.9.10 without issues. This system was reset cause I upgraded my storage drives.

#### Actual behaviour
Failed to start jellyfin.service - Jellyfin Media Server.
[image]

sureshk75, 2024-09-02:
Never mind... it appears to be a permission issue.. I fixed it by applying jellyfin:adm on /etc/jellyfin as mentioned in the jellyfin github site.

WolfganP, 2024-09-02:
> Never mind... it appears to be a permission issue.. I fixed it by applying jellyfin:adm on /etc/jellyfin as mentioned in the jellyfin github site.

It happened to me as well. Check https://github.com/jellyfin/jellyfin-packaging/issues/32# and where to properly config /etc/jellyfin permissions after the program updates.

closed 2024-10-18 by MichaIng