MichaIng/DietPi #7211: DietPi-Config - Unable to clear proxy username/password

dipisoft, 2024-09-04:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=9 | G_DIETPI_VERSION_SUB=7 | G_DIETPI_VERSION_RC=1 | G_GITBRANCH='master' | G_GITOWNER='MichaIng'
- Distro version | bookworm
- Kernel version | Linux DietPi 6.1.0-23-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux
- SBC model | Native PC (x86_64)
- Power supply used | 12V
- SD card used | SSD M2 16Go

#### Additional Information (if applicable)
- Software title | dietpi-config
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. launch dietpi-config
2. select option "7 : Network Options: Adapters"
3. select option "Proxy"
4. select option "Username", enter anything, "toto" for example and validate with "Ok"
5. re-select option "Username", clear the entered value and validate with "Ok"
6. the "_[FAILED] An input value was not entered, please try again..._" error message appears, despite the presence of the mention "Leave blank if not required"
7. same behavior for the "Password" option

#### Expected behaviour
- as the "Leave blank if not required" mentions, it should be possible to empty the previously entered values ​​in "Username" and "Password"

#### Actual behaviour
- it is currently not possible to clear previously entered values ​​in "Username" and "Password"

MichaIng, 2024-09-04:
Thanks for reporting. At some point we added a condition to our inputbox function to require some input, which we previously handled around the function call in various cases. We missed to adjust the call in this case.

For now, you can either adjust `CONFIG_PROXY_USERNAME` and `CONFIG_PROXY_PASSWORD` and re-apply settings by opening that menu and just select "Ok", or you edit the actual login script which applies the proxy settings: /etc/bashrc.d/dietpi-proxy.sh

I'll implement proper handling for variable inputbox conditions now.

MichaIng, 2024-09-04:
This should solve it: https://github.com/MichaIng/DietPi/commit/f7096ee
If you want to test, you can switch to `dev` branch this way:
```sh
G_DEV_BRANCH dev
```
And to revert
```sh
G_DEV_BRANCH master
```

dipisoft, 2024-09-04:
Wow, what a responsiveness! 😮

Thank you very much! I'll try to test this tomorrow and get back to you.

dipisoft, 2024-09-05:
Tested and it works, **except in one particular case**: if the content of the field started with spaces then emptying the field is not taken into account. You have to operate in two steps: remove the spaces, validate, re-edit the field to remove the rest of the characters, validate. Only then will the field be truly emptied.

closed 2024-10-18 by MichaIng