MichaIng/DietPi #7213: orange pi 5 plus emmc not detected

artifex69, 2024-09-04:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`  

- Distro version | `bookworm`  
- Kernel version | `Linux DietPi 6.1.75-vendor-rk35xx #1 SMP Thu Jul 25 14:36:35 UTC 2024 aarch64 GNU/Linux`
- SBC model | `Orange Pi 5 Plus (aarch64)`
- Power supply used | `5V 4A  power source`
- SD card used | `kingston 16gb`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?  
yes  

- Bug report ID | `bc1d3cee-51d7-475d-98f0-cc521f079dbe`
#### Steps to reproduce

1. Install dietpi
2. boot

#### Expected behaviour

- eMMC card detected

#### Actual behaviour

- eMMC card is not detected

#### Extra details

it doesn't seem to be a faulty card, booting official image "Orangepi5plus_1.0.8_debian_bookworm_server_linux6.1.43" does detect the eMMC card, kernel `Linux orangepi5plus 6.1.43-rockchip-rk3588 #1.0.8 SMP Sat Feb  3 00:04:38 CST 2024 aarch64 GNU/Linux`

**dietpi dmesg and lsblk**
```
root@DietPi:~# dmesg|grep mmc
[   14.844738] dwmmc_rockchip fe2c0000.mmc: No normal pinctrl state
[   14.844763] dwmmc_rockchip fe2c0000.mmc: No idle pinctrl state
[   14.844829] dwmmc_rockchip fe2c0000.mmc: IDMAC supports 32-bit address mode.
[   14.844868] dwmmc_rockchip fe2c0000.mmc: Using internal DMA controller.
[   14.844887] dwmmc_rockchip fe2c0000.mmc: Version ID is 270a
[   14.844948] dwmmc_rockchip fe2c0000.mmc: DW MMC controller at irq 105,32 bit host data width,256 deep fifo
[   14.845037] dwmmc_rockchip fe2c0000.mmc: Looking up vmmc-supply from device tree
[   14.846046] sdhci-dwcmshc fe2e0000.mmc: Looking up vmmc-supply from device tree
[   14.846061] sdhci-dwcmshc fe2e0000.mmc: Looking up vmmc-supply property in node /mmc@fe2e0000 failed
[   14.855478] sdhci-dwcmshc fe2e0000.mmc: Looking up vqmmc-supply from device tree
[   14.855491] sdhci-dwcmshc fe2e0000.mmc: Looking up vqmmc-supply property in node /mmc@fe2e0000 failed
[   14.855566] dwmmc_rockchip fe2c0000.mmc: Looking up vqmmc-supply from device tree
[   14.861432] mmc0: CQHCI version 5.10
[   14.884021] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[   14.894516] mmc0: SDHCI controller on fe2e0000.mmc [fe2e0000.mmc] using ADMA
[   14.931096] mmc_host mmc1: Bus speed (slot 0) = 49500000Hz (slot req 50000000Hz, actual 49500000HZ div = 0)
[   14.931819] mmc1: new high speed SDHC card at address 1234
[   14.932962] mmcblk1: mmc1:1234 SA16G 14.4 GiB 
[   14.943460]  mmcblk1: p1
[   15.799661] EXT4-fs (mmcblk1p1): mounted filesystem with ordered data mode. Quota mode: none.
[   17.587316] EXT4-fs (mmcblk1p1): re-mounted. Quota mode: none.
```
```
root@DietPi:/boot# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0    16M  0 disk 
mmcblk1     179:0    0  14,4G  0 disk 
└─mmcblk1p1 179:1    0  14,4G  0 part /
nvme0n1     259:0    0 953,9G  0 disk 
└─nvme0n1p1 259:1    0 953,9G  0 part 
``` 

orange pi 1.0.8 **dietpi dmesg and lsblk**  

eMMC detected `[    6.776179] mmcblk0: mmc0:0001 A3A551 28.9 GiB `

```
oot@DietPi:/boot# cat dmesg-orangepi.txt|grep mmc
[    6.324261] dwmmc_rockchip fe2c0000.mmc: No normal pinctrl state
[    6.324272] dwmmc_rockchip fe2c0000.mmc: No idle pinctrl state
[    6.324334] dwmmc_rockchip fe2c0000.mmc: IDMAC supports 32-bit address mode.
[    6.324351] dwmmc_rockchip fe2c0000.mmc: Using internal DMA controller.
[    6.324361] dwmmc_rockchip fe2c0000.mmc: Version ID is 270a
[    6.324396] dwmmc_rockchip fe2c0000.mmc: DW MMC controller at irq 146,32 bit host data width,256 deep fifo
[    6.343203] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    6.356871] mmc0: SDHCI controller on fe2e0000.mmc [fe2e0000.mmc] using ADMA
[    6.402652] mmc_host mmc1: Bus speed (slot 0) = 148500000Hz (slot req 150000000Hz, actual 148500000HZ div = 0)
[    6.420761] dwmmc_rockchip fe2c0000.mmc: Successfully tuned phase to 155
[    6.420774] mmc1: new ultra high speed SDR104 SDHC card at address 0007
[    6.421082] mmcblk1: mmc1:0007 SD16G 14.5 GiB 
[    6.433940]  mmcblk1: p1 p2
[    6.775402] mmc0: Host Software Queue enabled
[    6.775417] mmc0: new HS400 Enhanced strobe MMC card at address 0001
[    6.776179] mmcblk0: mmc0:0001 A3A551 28.9 GiB 
[    6.777544] mmcblk0boot0: mmc0:0001 A3A551 4.00 MiB 
[    6.778468] mmcblk0boot1: mmc0:0001 A3A551 4.00 MiB 
[    6.779390] mmcblk0rpmb: mmc0:0001 A3A551 16.0 MiB, chardev (235:0)
[    7.539076] EXT4-fs (mmcblk1p2): mounted filesystem with writeback data mode. Quota mode: none.
[    9.499239] EXT4-fs (mmcblk1p2): re-mounted. Quota mode: none.
[   12.166013] FAT-fs (mmcblk1p1): utf8 is not a recommended IO charset for FAT filesystems, filesystem will be case sensitive!
```
```
root@orangepi5plus:~# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0     31:0    0    16M  0 disk 
mmcblk1      179:0    0  14,5G  0 disk 
├─mmcblk1p1  179:1    0     1G  0 part /boot
└─mmcblk1p2  179:2    0  13,3G  0 part /var/log.hdd
                                       /
mmcblk0      179:32   0  28,9G  0 disk 
mmcblk0boot0 179:64   0     4M  1 disk 
mmcblk0boot1 179:96   0     4M  1 disk 
zram0        254:0    0   7,8G  0 disk [SWAP]
zram1        254:1    0   200M  0 disk /var/log
nvme0n1      259:0    0 953,9G  0 disk 
└─nvme0n1p1  259:1    0 953,9G  0 part 
```

I also tried to boot with the orangepi kernel, but the result was the same, eMMC card not detected.

MichaIng, 2024-10-18:
Sorry for late reply. Does this issue still exist?

artifex69, 2024-10-22:
Thank you for the reply.

Yes, this issue still is there but I figured out what is the problem and it's the u-boot.
Booting the orange pi with dietpi default uboot (vendor) leads to eMMC card detection failure.
Switching to u-boot-orangepi5-plus-legacy and installing it solves the problem.
So, in order to solve the issue it is necessary to do:
apt install u-boot-orangepi5-plus-legacy
cd /usr/lib/linux-u-boot-legacy-orangepi5-plus
dd if=idbloader.img of=/dev/mmcblk1 seek=64 conv=notrunc
588+0 records in
588+0 records out
301056 bytes (301 kB, 294 KiB) copied, 0.00487433 s, 61.8 MB/s

dd if=u-boot.itb of=/dev/mmcblk1 seek=16384 conv=notrunc
2740+0 records in
2740+0 records out
1402880 bytes (1.4 MB, 1.3 MiB) copied, 0.0183882 s, 76.3 MB/s

Attached both boot logs, maybe there is some voltage failure there
vendor:
DM: v2
mmc@fe2c0000: 1, mmc@fe2e0000: 0
Card did not respond to voltage select!

[boot-linux-u-boot-orangepi5-plus-legacy.txt](https://github.com/user-attachments/files/17480862/boot-linux-u-boot-orangepi5-plus-legacy.txt)
[boot-linux-u-boot-orangepi5-plus-vendor.txt](https://github.com/user-attachments/files/17480864/boot-linux-u-boot-orangepi5-plus-vendor.txt)

MichaIng, 2024-10-26:
Thanks for the info. Hmm good to know and bad. Actually I thought that both U-Boot packages use the same source. But the versions are different, if course. I'll test our as well, and see whether I can identify the related change in the source code.

daileyo, 2026-08-31:
I'm probably the only one, but I was a bit lost on the solution. So I figured I'd add some info in case it helps others... as well as to get someone to chime in if I misunderstood the solution.

### Solution

- In **dietpi-config** -> **Advanced Options**
- Run **SPI bootloader** to flash the current U-Boot to SPI storage
- reboot.

### TLDR;

While I was ultimately wanting to boot from the eMMC; the problem was that it would not show up, when running DietPi. It kind of did with official Orange Pi images... but I stress kind of... and that is a tangent. Also the steps to get the system to boot from the eMMC after it's visible is another tangent.

As the ghost and Michalng mentioned; it seems to be an older bootloader; but I didn't have to downgrade anything to get it to work. I was able to resolve  the missing eMMC on an Orange Pi 5 Max running the current DietPi image. I just needed to update teh SPI bootloader.

On my Orange Pi 5 Max, mmc0 initialized but the eMMC never enumerated. The SPI flash contained an older Orange Pi U-Boot:

**U-Boot SPL 2017.09-orangepi (Jun 24 2024)**

DietPi already had a SPI image installed at:

**_/usr/lib/linux-u-boot-vendor-orangepi5-max/rkspi_loader.img_**

In dietpi-config -> Advanced Options, I ran Update SPI bootloader to flash the current U-Boot to SPI storage and rebooted.

After that, the eMMC immediately enumerated.

I also tested Update MMC bootloader before updating SPI. That successfully updated the bootloader on the SD card, but the eMMC still did not enumerate afterward. It was specifically after Update SPI bootloader and reboot that the eMMC appeared.

This got me past the hump of my eMMC device not showing up; other tangents later; I was able to get DietPi running from the eMMC.

MichaIng, 2026-08-31:
It is most likely that the Orange Pi hardware boots vendor bootloaders with higher priority, and SD card > eMMC > SPI only as a secondary priority order. This means:
- With legacy Orange Pi image from Xunlong on eMMC and legacy SPI bootloader => it loads the bootloader from eMMC
- With modern DietPi/Armbian image on eMMC and legacy SPI bootloader => it loads the bootloader from SPI => boot fails
- With modern image and mainline U-Boot on SPI => it loads the bootloader from eMMC

We found the same annoying and IMO unnecessary vendor <> mainline priority difference on some FriendlyELEC boards as well. So when flashing a different bootloader onto either SPI or SD/eMMC, it can potentially render the system unbootable, as it can change the boot(loader) priority for no good reason.

E.g. NanoPi R5/R6 boards:
- They ship without SPI bootloader, but instead with internal fixed eMMC.
- The eMMC is by default flashed with FriendlyWRT, hence with a legacy vendor bootloader.
- If you flash and insert an SD card with a FriendlyELEC image, with legacy bootloader, that boots, as one would expect: SD card > eMMC priority.
- But if you flash DietPi or Armbian or any other mainline U-Boot image, it keeps booting from the eMMC.
- They all have a MASK button, which prevents the eMMC from being enumerated, similar to how this works with the SPI flash on many boards. The Orange Pi 5 Plus e.g. has a MASKROM button as well. The Orange Pi 5 Max does note.
- However, masking the eMMC means that it is not available from the OS booted from SD card either. So you have no way to flash it from there. Hence the user is partly stuck.
- Only way to solve is to boot into FriendlyWRT, and erase the boot sectors via `dd` from there.