MichaIng/DietPi #7214: orange pi 5 plus, console is bigger than the screen

artifex69, 2024-09-04:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`  

- Distro version | `bookworm`  
- Kernel version | `Linux DietPi 6.1.75-vendor-rk35xx #1 SMP Thu Jul 25 14:36:35 UTC 2024 aarch64 GNU/Linux`
- SBC model | `Orange Pi 5 Plus (aarch64)`
- Power supply used | `5V 4A  power source`
- SD card used | `kingston 16gb`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?  
yes  

- Bug report ID | `bc1d3cee-51d7-475d-98f0-cc521f079dbe`
#### Steps to reproduce

1. Install dietpi
2. boot

#### Expected behaviour

- console tty fit the screen, i mean the console lines number should not be greater than the lines visible on the screen

#### Actual behaviour

the console is bigger than the screen, the last lines are not visible at the bottom of the screen

#### Extra details

artifex69, 2024-09-08:
The problem seems to be the driver.  Monitor resolution is detected correctly but the monitor is set to full hd when it should be 1920x1200

`fbset -i`
```
mode "1920x1200"
    geometry 1920 1200 1920 1200 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode

Frame buffer device information:
    Name        : rockchipdrmfb
    Address     : 0
    Size        : 9216000
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 7680
    Accelerator : No
```
setting the console resolution to 1080  (´fbset-yres 1080´)will make it fit the monitor but with image stretching.

artifex69, 2024-10-22:
it seems that the kernel and drivers are not adequate, besides failing graphics NFS does not work either, the module isn't compiled too.
As a workaround I installed [Joshua-Riek](https://github.com/Joshua-Riek)  kernel and  modules and  https://launchpad.net/~liujianfeng1994 panfork mesa, both compiled with dietpi bookworm.
Now  all I've tried is working, GPU acceleration for firefox and chromium, jellyfin transcoding etc

So, for me that issue is resolved

MichaIng, 2024-10-26:
Panfrost and NFS kernel divers should both be there. But maybe some X11 config is needed to enable it for X. Though JianFeng also has own Chromium etc packages, which you use I guess?

artifex69, 2024-10-29:
Hi MichaIng, sorry for the delay, I was doing some tests this days.
You are right, NFS driver is there, it was another kernel that lacks the driver, Maybe the orange pi official ones.
The chromium package is the dietpi one. 
I cannot get to work the GPU acceleration nor in Chromiun neither firefox with panfrost.  chromium chrome://gpu and firefox about:support shows no HW render.

However, I get the HW acceleration by changing the following packages  to the JianFeng panfork version: 
libegl-mesa0
libgbm1
libgl1-mesa-dri
libglapi-mesa
libglx-mesa0 
 and installing mali-g610-firmware.

Chromium need to be started without --use-gl=egl flag in the file /etc/chromium.d/dietpi file though.

there is still de problem with the screen resolution that it's not working with some of resolutions and shows the weird behaviour on console

MichaIng, 2025-01-05:
Actually, Mesa 24.1 should have brought G610 support. And it is available via Bookworm backports:
```sh
cat << '_EOF_' > /etc/apt/preferences.d/dietpi-mesa
Package: src:mesa
Pin: release a=bookworm-backports
Pin-Priority: 500
_EOF_
apt upgrade

closed 2025-06-22 by MichaIng