MichaIng/DietPi #7216: Option to set a different default user then "dietpi"

ikem-krueger, 2024-09-06:
Within the configuration file `dietpi.txt` I want to be able to set a default user like this:

```
# Global user to be applied for the system
# - During first run setup, the user is created
# - The user has sudo capabilities
AUTO_SETUP_GLOBAL_USER=custom
```

That way a hacker would have it harder to get into my appliance.

Joulinar, 2024-09-06:
This feature we have on our agenda already to switch from DietPi user to a custom user. But ETA yet.

MichaIng, 2024-09-19:
Jep, actually my plan is to not create an other user by default, but only as option, via `AUTO_SETUP_EXTRA_USER=` and first run setup dialogue, or something like that.

ikem-krueger, 2024-09-19:
Can I then disable the default user?

MichaIng, 2024-09-19:
There would be none by default, only root, and you could skip creating one on first login.

ikem-krueger, 2024-09-19:
I meant root. Sorry.

MichaIng, 2024-09-19:
The root user can be theoretically locked for regular logins. For SSH you can do that already now.

Yoghoo, 2025-05-13:
What is the status of this feature request? Just took a first look at DietPi and really like it but not able to create your own default user in dietpi.txt is a big let down. As other scripts still heavily depend on this user even if you create other users by hand and disable the dietpi user. Creating a mount with the dietpi program for example will just mount it with permission dietpi:dietpi in the /etc/fstab. For example Raspberry Pi OS doesn't depend on the pi user anymore for some time now. So hope this will be implemented soon.

MichaIng, 2025-05-13:
For now, you could do that all with the custom script feature in `dietpi.txt`. Rarely anything depends on the `dietpi` user. I turned most references to be conditional already, left are only Syncthing and Google AIY installation options, and the latter (Voice Kit device) is EOL since 4 years, including the related software repositories. So that one we probably just remove.

Yoghoo, 2025-05-13:
Of course I can (and did) script it but it's quite cumbersome as I want user id and group id (1000:1000) also for my own account. And as I know that the dietpi user is still hardcoded in several places I need to usermod/groupmod this user first. Just removing it will probably cause issues somewhere. I hope this will be picked up soon. For now I will take a look at Armbian or go back to Rasberry Pi OS as this just feels kind of hacky and does not feel secure. And I know you don't agree with that but then again I am an end user and don't know (and want to know) why this was done in the first place and why it's still needed. Please don't take it personally btw as I only want to share my opinion and it's absolutely fine if you don't agree with that. ;)

MichaIng, 2025-05-13:
> And as I know that the dietpi user is still hardcoded in several places

As said, it is not. All DietPi systems I manage have the `dietpi` user removed. You can do that first with your custom script, and then create your own user with ID 1000. The only thing which is still more often used by software installs is the `dietpi` group, so if you install downloaders, media players and such via `dietpi-software`, you might need a `dietpi` group for the installation to go through without error.