MichaIng/DietPi #7220: Corrupted System Caused by Update Failure (dietpi-update): Run now to update DietPi from v9.6.1 to v9.7.1

SystemTheory, 2024-09-07:
**Problem**

I login to dietpi using ssh from powershell.

After running dietpi-update as root the update failed as shown in the output below. I cannot login as root or user from another powershell tab; and I cannot run commands as root on dietpi via ssh! 

Is my system corrupted beyond repair? How can I recover data from the hard drive? Can I plug it into another linux OS?

**Terminal Output Update Failure**
```
DietPi v9.6.1 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (aarch64)
 - CPU temp : 45 °C / 113 °F : Optimal temperature
 - LAN IP : 192.168.1.6 (wlan0)
 - MOTD : DietPi v9.7 is here. Check out all changes:
          https://dietpi.com/docs/releases/v9_7/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Patreon Legends : Chris Gelatt, ADSB.im
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v9.6.1 to v9.7.1

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@dpi4 /root
# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.7.1
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v9.6.1
[ INFO ] DietPi-Update | Latest version  : v9.7.1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=907957 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
[... 113 lines cut ...]
/boot/dietpi/func/dietpi-globals: line 473: ((: <= 120 : syntax error: operand expected (error token is "<= 120 ")
/boot/dietpi/func/dietpi-globals: line 479: ((: >= 7 : syntax error: operand expected (error token is ">= 7 ")
/boot/dietpi/func/dietpi-globals: line 775: /usr/bin/rm: Input/output error
/boot/dietpi/func/dietpi-globals: line 1052: /usr/bin/rm: Input/output error
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.
/boot/dietpi/func/dietpi-globals: line 148: /usr/bin/rm: Input/output error
[FAILED] DietPi-Update | Failed to remove scripts working directory: /tmp/DietPi-Update
```
```
root@dpi4 /root
# cd opt
-bash: cd: opt: Input/output error

root@dpi4 /root
# tree -L 1
-bash: tree: command not found

root@dpi4 /root
# uptime
-bash: uptime: command not found

root@dpi4 /root
# ls
-bash: ls: command not found
```

Joulinar, 2024-09-08:
Are you using an SD card by any chance? If so, it could be that the card is slowly giving up. We can try to see if the following helps to repair defective blocks.

```
> /forcefsck
reboot
# then after reboot
journalctl -t systemd-fsck
```

SystemTheory, 2024-09-08:
I am using an external SSD as boot and data drive.

Joulinar, 2024-09-08:
How is it powered? It is known that the RPi4 is not able to power HDD/SSD correctly when using the RPI usb ports to power the drive. Therefore, we strongly recommend using an external power supply for such drives.

SystemTheory, 2024-09-08:
Thank you. I will consider options for disk power supply. The SSD is powered from USB port. The system has been stable for a couple of years while running 24/7. There was one update problem some time ago and the uptime gets disrupted at times but I think that is due to brown-out conditions in the local power grid.

MichaIng, 2024-10-18:
Is this still an issue, or did a dedicates PSU for the SSD help?

MichaIng, 2025-05-31:
Closing this for now. Feel free to reopen if issue persists.

closed 2025-05-31 by MichaIng