MichaIng/DietPi #7221: after installing klipper with kiauh: curl gives error `curl: error while loading shared libraries: �......�: cannot open shared object file: Error 36`

alber70g, 2024-09-11:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

<details>
<summary>The full error with history</summary>

        /=======================================================\
        |     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
        |        Klipper Installation And Update Helper         |
        |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
        \=======================================================/

        ###### Initializing Klipper installation ...

        /=======================================================\
        | Please select your preferred Python version.          |
        | The recommended version is Python 3.x.                |
        |-------------------------------------------------------|
        |  1) [Python 3.x]  (recommended)                       |
        |  2) [Python 2.7]  (legacy)                            |
        |-------------------------------------------------------|
        |                       B) « Back                       |
        \=======================================================/
        ###### Select Python version: 1
          [➔] Python 3.x

        /=======================================================\
        | Please select the number of Klipper instances to set  |
        | up. The number of Klipper instances will determine    |
        | the amount of printers you can run from this host.    |
        |                                                       |
        | WARNING:                                              |
        | Setting up too many instances may crash your system.  |
        |-------------------------------------------------------|
        |                       B) « Back                       |
        \=======================================================/
        ###### Number of Klipper instances to set up: 1
          [➔] Instance count: 1

        ###### Installing single Klipper instance ...

        ###### Checking for the following dependencies:
        ● git
        [✓ OK] Dependencies already met!

        ###### Cloning Klipper from https://github.com/Klipper3d/klipper ...
        Cloning into '/home/dietpi/klipper'...
        error: git-remote-https died of signal 11

        #=======================================================#
        Cloning Klipper from
        https://github.com/Klipper3d/klipper
        failed!
        #=======================================================#

        dietpi@DietPi:~$ shutdown -r now
        -bash: shutdown: command not found
        dietpi@DietPi:~$ sudo shutdown -r now
        Failed to connect to bus: No such file or directory
        Connection to 192.168.1.7 closed by remote host.
        Connection to 192.168.1.7 closed.

        20:36:07 ~ on 𝌋 unraid
        took 52m21s ❯ ssh dietpi@192.168.1.7

        The programs included with the Debian GNU/Linux system are free software;
        the exact distribution terms for each program are described in the
        individual files in /usr/share/doc/*/copyright.

        Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
[... 119 lines cut ...]
  I tried sending bug report, but it obviously fails now that `curl` is broken

#### Steps to reproduce

1. install fresh dietpi on Raspberry Pi Zero 2W
2. install git from the menu
3. clone dw-0/kiauh.git in the dietpi user home dir
4. run `./kiauh/kiauh.sh` (use non beta version)
5. select install (1)
6. select klipper (1)
7. select defaults for python and amount
8. go again to install
9. install moonraker

#### Expected behaviour

- installation and clone etc continues

#### Actual behaviour

- curl error as described above

#### Extra details

- ...

MichaIng, 2024-09-12:
I am not sure what this klipper installer does, but it looks like either a corrupted `curl` binary or one of its shared libraries.

Does this work without issues?
```sh
curl --version
curl https://dietpi.com/geoip
```

alber70g, 2024-09-15:
No this doesn't work. Both give the error like before. I suppose it does corrupt `curl` somehow. 
I can do some debugging if you want. Let me know what you need to see. For what I know, it's only using apt-get to install stuff nothing special. That shouldn't corrupt any installed software I'd say...

MichaIng, 2024-09-18:
Please try to reinstall `curl` and its library:
```sh
apt install --reinstall curl libcurl4
curl --version
```

closed 2025-06-02 by MichaIng