MichaIng/DietPi #7223: RPi 3B errors on clean install inc. Apt failing

mdr92, 2024-09-18:
### Creating a bug report/issue
- [Y] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.7.1
- Distro version | bookworm
- Kernel version | 6.1.21-v8+ 
- SBC model | RPi 3 Model B
- Power supply used | 5v 3A
- SD card used | Samsung Evo
- Build | DietPi_RPi-ARMv8-Bookworm.img.xz

#### Additional Information (if applicable)
- Happens on a clean install even with no software on a Raspberry Pi 3B, I've also tried an native pc version and that works ok
- I've researched but can only find examples caused by software, not on a clean install

#### Steps to reproduce

1. Clean install 9.7.1 and boot
2. After initial setup and updates working fine any subsequent boots result in errors in the banner and apt failing.

#### Expected behaviour
No errors in banner and apt should work

#### Actual behaviour
Banner has error 
"curl: error while loading shared libraries: ▒▒▒▒"
Pages of  ▒▒▒▒ symbol then
": cannot open shared object file: Error 36"

Apt fails with the below
Reading package lists... Error!
E: Problem parsing dependency 20 of python3-rtimulib-dbgsym:none=1.22y:0.164.3095+gitba97c73ibuutil3linux (= 2.2.3-1~bpo12+
E: Error occurred while processing python3-rtimulib-dbgsym (NewVersion2)
E: Problem with MergeList /var/lib/apt/lists/archive.raspberrypi.com_debian_dists_bookworm_main_binary-arm64_Packages.xz
E: The package lists or status file could not be parsed or opened.

[image]
[image]

#### Extra details
Tried multiple clean installs, happens even without any additional software installed, have also tried a week apart and with isp dns and the dietpi one's set with the static ip option in the dietpi.txt
Thanks

MichaIng, 2024-09-18:
This actually looks like your SD card is about to die, if you run into similar errors on multiple fresh installs. Do you have a different one to test with?

Another thing, while 3A should be enough for an RPi 3, if it is a no-name/low-quality PSU, or more intended for charging than for stable voltage operation, it may be not. Can you check for kernel errors?
```sh
dmesg -l 0,1,2,3
```

mdr92, 2024-09-18:
I can try a different card, it's the same SD card and PSU that's been running my old DietPi 7.8.4 setup for years though and that's working without issue, other than obviously been well out of support. I'll be back shortly after trying another card.

MichaIng, 2024-09-18:
> So I presume no errors?

Jep that looks good. Hmm, so PSU should then be fine, and at least there were no filesystem errors during the initial setup (or did you reboot?). Could be theoretically also the SD card reader, though most tools check content after flashing it. If you have another Linux system to plug the SD card after flashing, to `fsck` the ext4 partition, could be done to check this.

However
> `Problem with MergeList /var/lib/apt/lists/archive.raspberrypi.com_debian_dists_bookworm_main_binary-arm64_Packages.xz`

This file was written after first boot, so if it is corrupted, that happened afterwards 🤔.

mdr92, 2024-09-18:
It looks like it is the SD card so good spot! Weird how my 7.8.4 seems stable though and 9.7.1 is fine for the full setup and works inc. apt update until the first reboot. I also did a h2testw run on it and the whole card verified fine. Anyway, it has to be the SD card because it works ok with a different one.
Thanks for the help

closed 2024-09-19 by MichaIng