MichaIng/DietPi #7226: Download/Extraction of docker-ce-cli_27.3.1-1~debian.12~bookworm_arm64.deb fails

DerSpatz, 2024-09-21:
When I try to install docker via `apt install docker-ce`, the installation always fails, because the package `docker-ce-cli_27.3.1-1~debian.12~bookworm_arm64.deb` does not extract properly.

When I manually download the package from the official docker repo using `curl -O` in `/var/cache/apt/archives/` the installation works.

I am using an Odroid C4 and the latest DietPi version.

Joulinar, 2024-09-21:
did you already tried to install Docker via `dietpi-software`?

DerSpatz, 2024-09-21:
Yes, that was what I tried first, but it gave the same error, so that's when I tried `apt` directly.

At first the error happened a yesterday, when I first set up the system, and the error was there with `ce-cli_27.3.1~debian.12~bookworm_arm64.deb`, and was resolvable like described above, and I assumed it was a one off.
But today there was an update to from `27.3.1` to `27.3.1-1` and the same error occurred again, so I decided to post here.

Sorry for mentioning those details earlier...

Joulinar, 2024-09-21:
Can you share the entire output and error message

DerSpatz, 2024-09-21:
I did a fresh install to test this, with a formatted µSD card.

I edited `dietpi.txt` to German locale settings, and in `dietpiEnv.txt` I set `docker_optimizations=on`.

On the first automatic update/upgrade run, apt throws an error when trying to update initramfs, but retrying as suggested by the tool runs through successfully. Then, after changing passwords, I disabled the serial interface as suggested, and then I was able to install docker via `dietpi-config`.

Then I got the following output:

```
Selecting previously unselected package containerd.io.^M
Preparing to unpack .../1-containerd.io_1.7.22-1_arm64.deb ...^M
Unpacking containerd.io (1.7.22-1) ...^M
dpkg-deb: error: <decompress> subprocess was killed by signal (Segmentation fault)^M
dpkg: error processing archive /tmp/apt-dpkg-install-NNJoOi/1-containerd.io_1.7.22-1_arm64.deb (--unpack):^M
 cannot copy extracted data for './usr/bin/containerd' to '/usr/bin/containerd.dpkg-new': unexpected end of file or
stream^M
Selecting previously unselected package docker-ce-cli.^M
Preparing to unpack .../2-docker-ce-cli_5%3a27.3.1-1~debian.12~bookworm_arm64.deb ...^M
Unpacking docker-ce-cli (5:27.3.1-1~debian.12~bookworm) ...^M
dpkg-deb (subprocess): decompressing archive
'/tmp/apt-dpkg-install-NNJoOi/2-docker-ce-cli_5%3a27.3.1-1~debian.12~bookworm_arm64.deb' (size=13610412) member
'data.tar': lzma error: compressed data is corrupt^M
dpkg-deb: error: <decompress> subprocess returned error exit status 2^M
dpkg: error processing archive
/tmp/apt-dpkg-install-NNJoOi/2-docker-ce-cli_5%3a27.3.1-1~debian.12~bookworm_arm64.deb (--unpack):^M
 cannot copy extracted data for './usr/bin/docker' to '/usr/bin/docker.dpkg-new': unexpected end of file or stream^M
Selecting previously unselected package libip6tc2:arm64.^M
Preparing to unpack .../3-libip6tc2_1.8.9-2_arm64.deb ...^M
Unpacking libip6tc2:arm64 (1.8.9-2) ...^M
Selecting previously unselected package libnfnetlink0:arm64.^M
Preparing to unpack .../4-libnfnetlink0_1.0.2-2_arm64.deb ...^M
Unpacking libnfnetlink0:arm64 (1.0.2-2) ...^M
Selecting previously unselected package libnetfilter-conntrack3:arm64.^M
Preparing to unpack .../5-libnetfilter-conntrack3_1.0.9-3_arm64.deb ...^M
Unpacking libnetfilter-conntrack3:arm64 (1.0.9-3) ...^M
Selecting previously unselected package libnftnl11:arm64.^M
Preparing to unpack .../6-libnftnl11_1.2.4-2_arm64.deb ...^M
Unpacking libnftnl11:arm64 (1.2.4-2) ...^M
Selecting previously unselected package iptables.^M
Preparing to unpack .../7-iptables_1.8.9-2_arm64.deb ...^M
Unpacking iptables (1.8.9-2) ...^M
Selecting previously unselected package docker-ce.^M
Preparing to unpack .../8-docker-ce_5%3a27.3.1-1~debian.12~bookworm_arm64.deb ...^M
Unpacking docker-ce (5:27.3.1-1~debian.12~bookworm) ...^M
Errors were encountered while processing:^M
 /tmp/apt-dpkg-install-NNJoOi/1-containerd.io_1.7.22-1_arm64.deb^M
 /tmp/apt-dpkg-install-NNJoOi/2-docker-ce-cli_5%3a27.3.1-1~debian.12~bookworm_arm64.deb^M
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Retrying did not solve the problem, but `apt --fix-broken install`  at least installed docker-ce-cli, while containerd.io still exited with a failed extraction:

```
Vorbereitung zum Entpacken von .../containerd.io_1.7.22-1_arm64.deb ... _{preparing unpacking}_
Entpacken von containerd.io (1.7.22-1) ... _{unpacking}_
dpkg-deb: Fehler: »<Dekomprimierung>«-Unterprozess wurde durch Signal (Speicherzugriffsfehler) getötet _{Error: unpacking subprocess was killed by signal (memory access error}_
dpkg: Fehler beim Bearbeiten des Archivs /var/cache/apt/archives/containerd.io_1.7.22-1_arm64.deb (--unpack): _{error unpacking archive}_
 Extrahierte Daten für »./usr/bin/containerd« können nicht nach »/usr/bin/containerd.dpkg-new« kopiert werden: Unerwartetes Ende der Datei oder des Datenstroms _{unexpected EOF or stream}_
Vorbereitung zum Entpacken von .../docker-ce-cli_5%3a27.3.1-1~debian.12~bookworm_arm64.deb ...
Entpacken von docker-ce-cli (5:27.3.1-1~debian.12~bookworm) ... _{no error unpacking}_
Fehler traten auf beim Bearbeiten von:
 /var/cache/apt/archives/containerd.io_1.7.22-1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

After that I tried:

```
rm /var/cache/apt/archives/containerd.io_1.7.22-1_arm64.deb
apt --fix-broken install
```

But this did not solve the problem.

```
rm /var/cache/apt/archives/containerd.io_1.7.22-1_arm64.deb
cd /var/cache/apt/archives/
curl -O https://download.docker.com/linux/ubuntu/dists/jammy/pool/stable/arm64/containerd.io_1.7.22-1_arm64.deb
dpkg -i /var/cache/apt/archives/*deb
```

But this also did not work (unexpected EOF). Honestly, I am pretty confused right now. Last time, the docker-ce-cli did not work, now it's containerd.io, and manual install also does noct work now...

Joulinar, 2024-09-21:
Do you have a specific network security device, gateway or firewall? Sometimes such device could have unwanted side effects.

MichaIng, 2024-09-21:
Also, is this always the same SD card, or did you try a different one? And please check for kernel errors:
```sh
dmesg -l 0,1,2,3
```

DerSpatz, 2024-09-21:
It was a different SD card for the second install, but I decided to try amrbian now...

MichaIng, 2024-09-21:
> But this also did not work (unexpected EOF). Honestly, I am pretty confused right now. Last time, the docker-ce-cli did not work,

Was this "unexpected EOF" already during the `curl` download, or when doing the `dpkg` install? Either it aborts the download at some point, of it fails to write the file correctly to disk. The latter would mean there must be a related kernel error, e.g. low voltage, filesystem corruption, any such, which can be checked with above `dmesg` command. If the download fails already, then it could be a network issue, as indicated by Joulinar. The packages are downloaded from the same official Docker APT repository, so those are fine.

MichaIng, 2025-05-31:
Closing this for now. Feel free to reopen if issue persists.

closed 2025-05-31 by MichaIng