MichaIng/DietPi #7228: Please add Home Assistant Yellow (CM4)

bwims, 2024-09-24:
### Creating an image request

I run Home Assistant via Docker on an RPi4, but was recently gifted a Home Assistant Yellow which is a carrier board for a CM4, well populated with networking and storage upgrade capability, but with a flashed standalone operating system that precludes running from Docker (though it is container-based).
 
I am a linux user with some experience but not enough to deal with creating my own version of dietpi.  I have no idea how much work would be required to do so, but I thought, no harm in asking!

#### Formal device information
- Device name | Home Assistant Yellow - a carrier board for RPi CM4
- Official product URL | https://www.home-assistant.io/yellow/
- GitHub resource URL (if available) | 
- Image download URL | https://yellow.home-assistant.io/power-supply/

The image is supplied via the Raspberry Pi Imager

#### Is the SBC officially supported by the Debian installer?
- ..No

#### If not, is a reliable 3rd party Debian image available for this SBC?
- I don't know

#### If not, are there install instructions for Debian available?

- No

MichaIng, 2024-09-24:
Does our RPi 4 image not work OOTB on it?

bwims, 2024-09-25:
Not yet.  (a) I've been travelling with a family member who gave me the kit for my birthday (!) 
(b) whilst researching the possibilities I came across [this ](https://www.reddit.com/r/homeassistant/comments/167s4ym/success_installing_dietpi_on_home_assistant_yellow/)thread which outlined what looked like a complicated procedure, and you actually commented on it ( ! - probably slipped your mind) .

So unless you have made some major changes in the past year that would accommodate the differences, I'm guessing it wouldn't work OOB?

B.

MichaIng, 2024-09-26:
Ah, I indeed forgot about that. Good find 👍🙂. So makes sense to test our image, which should boot at least. Then we could go through the device tree and patches and see whether we can create an overlay which enables some or all of the baseboard features.

bwims, 2024-09-27:
Of course, pretty silly of me to think you would have access to one of these bits of hardware.  I'm loathe to overwrite the OOB CM4, so what I'll need to do is get another CM4 to play with.  So until then, I'll close this and reopen later.

Incidentally, I am such a Linux noob, that I don't even understand the bit about device tree and patches (nor did I with the article I referenced - just saw your name there which gave me hope!)

Thanks!

closed 2024-09-27 by bwims