MichaIng/DietPi #7230: Update kernel?

danergo, 2024-10-01:
Folks, I know you are not doing any kernel developments.

I have updated to latest version, and now my SC16IS752 (UART to SPI) started behaving strange (using it for full uart: rx,tx,cts,rts).

It runs for hours (10-12) but then usually (9 og 10) when there is some 'bursty' data, it just seems dropping it.

This converter chip is utilized with dts by the kernel.

Kernel version is 6.1.21-v18+.

I found some interesting kernel source commits for this chip's driver here:
https://github.com/torvalds/linux/commits/master/drivers/tty/serial/sc16is7xx.c

Some also mentions invalid fifo fixes, tx fifo corruption fix, etc. These are rather recent commits dating back only to a few months.

Unfortunately I couldn't find out the date of my kernel, moreover I can't guess the latest included commit of this source file. Is it even possible to query this information from a running system?

Also: how can I know which kernel version contains which version (commit) of this sc16is7xx.c file?

Basically which kernel source is dietpi based on?

Joulinar, 2024-10-01:
> Basically which kernel source is dietpi based on?
Depends on SBC used. You are running RPi?

danergo, 2024-10-01:
Sorry forgot to add this. Yes, Rpi3.

Joulinar, 2024-10-01:
On the RPi SBC we use the original kernel as provided by the PRi Foundation. There is no difference to the RPi OS at kernel level. Except for one limitation we have with standard images. We are currently using kernel 6.1.21 by default, as we have not yet switched to the new kernel/file system structure for all users. However, there is a BETA path to upgrade to the new RPi kernel. See  https://github.com/MichaIng/DietPi/issues/6676 Please note the corresponding WARNING in the other post.

danergo, 2024-10-20:
Thanks guys. It seems (very likely), that root cause was not due to kernel.

closed 2024-10-20 by danergo