MichaIng/DietPi #7231: Home Assistant 2024.10 failed after update from 2024.9.3

whyisthisbroken, 2024-10-03:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'
- Distro version bookworm 0
- Kernel version Linux HomeNetDNS 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model RPI 4 8GB

#### Additional Information (if applicable)
- Software title | Home Assistant
- Home Assistant 2024.9.3 / Frontend 20240909.1
- Can this issue be replicated on a fresh installation of DietPi? I dont know

#### Steps to reproduce

1. /home/homeassistant/homeassistant-update.sh
2. Home Assistant update in background much depencies..
3. Home Assistant starts but could not load anything.

#### Expected behaviour

- Update and work like before :)

#### Actual behaviour

- Update and not working like before :(

#### Extra details
- This Error comes at evry update depencie:
  error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment

After Update and HA want start this comes...
- Oct 03 08:15:58 HomeNetDNS Home Assistant[1936]: 2024-10-03 08:15:58.319 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of 'recovery_mode'. Setup failed for dependencies: ['frontend', 'cloud']
Oct 03 08:15:58 HomeNetDNS Home Assistant[1936]: 2024-10-03 08:15:58.320 ERROR (MainThread) [homeassistant.setup] Setup failed for 'recovery_mode': Could not setup dependencies: frontend, cloud

mostlyvirtual, 2024-10-03:
I have a very similar or identical issue, but I don't have as many modules failing. I even tried purging the whole installation and starting fresh, but still get the errors. I submitted a BugReport via the dietpi-bugreport feature, and am including some logs here to help with this.

```bash
Oct 03 10:51:34 pi3 Home Assistant[18272]: 2024-10-03 10:51:34.594 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration tapo_control which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 03 10:51:34 pi3 Home Assistant[18272]: 2024-10-03 10:51:34.598 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration midea_ac_lan which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 03 10:51:34 pi3 Home Assistant[18272]: 2024-10-03 10:51:34.601 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration tapo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 03 10:51:34 pi3 Home Assistant[18272]: 2024-10-03 10:51:34.604 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 03 10:51:41 pi3 Home Assistant[18272]: 2024-10-03 10:51:41.803 ERROR (SyncWorker_1) [homeassistant.util.package] Unable to install package pyotp==2.8.0: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:51:46 pi3 Home Assistant[18272]: 2024-10-03 10:51:46.791 ERROR (SyncWorker_1) [homeassistant.util.package] Unable to install package pyotp==2.8.0: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:51:51 pi3 Home Assistant[18272]: 2024-10-03 10:51:51.732 ERROR (SyncWorker_1) [homeassistant.util.package] Unable to install package pyotp==2.8.0: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:51:56 pi3 Home Assistant[18272]: 2024-10-03 10:51:56.662 ERROR (SyncWorker_1) [homeassistant.util.package] Unable to install package PyQRCode==1.2.1: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:01 pi3 Home Assistant[18272]: 2024-10-03 10:52:01.600 ERROR (SyncWorker_1) [homeassistant.util.package] Unable to install package PyQRCode==1.2.1: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:06 pi3 Home Assistant[18272]: 2024-10-03 10:52:06.555 ERROR (SyncWorker_1) [homeassistant.util.package] Unable to install package PyQRCode==1.2.1: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:06 pi3 Home Assistant[18272]: 2024-10-03 10:52:06.557 ERROR (MainThread) [homeassistant.bootstrap] Home Assistant core failed to initialize. Further initialization aborted
Oct 03 10:52:06 pi3 Home Assistant[18272]: 2024-10-03 10:52:06.558 WARNING (MainThread) [homeassistant.bootstrap] Unable to set up core integrations. Activating recovery mode
Oct 03 10:52:11 pi3 Home Assistant[18272]: 2024-10-03 10:52:11.667 ERROR (SyncWorker_6) [homeassistant.util.package] Unable to install package pyotp==2.8.0: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:16 pi3 Home Assistant[18272]: 2024-10-03 10:52:16.646 ERROR (SyncWorker_6) [homeassistant.util.package] Unable to install package pyotp==2.8.0: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:21 pi3 Home Assistant[18272]: 2024-10-03 10:52:21.652 ERROR (SyncWorker_6) [homeassistant.util.package] Unable to install package pyotp==2.8.0: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:26 pi3 Home Assistant[18272]: 2024-10-03 10:52:26.677 ERROR (SyncWorker_6) [homeassistant.util.package] Unable to install package PyQRCode==1.2.1: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:31 pi3 Home Assistant[18272]: 2024-10-03 10:52:31.687 ERROR (SyncWorker_6) [homeassistant.util.package] Unable to install package PyQRCode==1.2.1: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:36 pi3 Home Assistant[18272]: 2024-10-03 10:52:36.751 ERROR (SyncWorker_6) [homeassistant.util.package] Unable to install package PyQRCode==1.2.1: error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
Oct 03 10:52:36 pi3 Home Assistant[18272]: 2024-10-03 10:52:36.753 ERROR (MainThread) [homeassistant.bootstrap] Home Assistant core failed to initialize. Further initialization aborted
```

domenicoblanco, 2024-10-03:
I've solved my issue logging in as root and installing uv, creating a new venv with `uv venv` and finally editing all the files in `/home/homeassistant`:
- homeassistant-start.sh: changing the first line from dash to bash
- homeassistant-update.sh:
```bash
#!/bin/bash
exec sudo -u homeassistant bash -c '. /home/homeassistant/pyenv-activate.sh; exec uv pip install -U homeassistant'
```
- pyenv-activate.sh
```bash
#!/bin/bash
if [ $(whoami) != 'homeassistant' ]; then
        echo '[FAILED] This pyenv must be activated as user "homeassistant". Aborting...'
        kill -INT $$
fi
cd /home/homeassistant
export PYENV_ROOT='/home/homeassistant/.venv'
export PATH="$PYENV_ROOT/bin:$PATH"
source .venv/bin/activate
[ -f '.cargo/env' ] && . .cargo/env
```

MichaIng, 2024-10-03:
Looks like one needs to pass `--system` now if it is installed without virtual environment (it is, as we use `pyenv`, but that one is is not detected as such, obviously).

InnovoDeveloper, 2024-10-03:
is adding --system a quick fix till there is a patch?  if so where should it be added to?  or is the @domenicoblanco method the quick fix?
Thanks for being fast to response.

InnovoDeveloper, 2024-10-03:
looks like restore is your uncle here till there is a perm fix :)

MichaIng, 2024-10-03:
Hmm, the `VIRTUAL_ENV` variable is passed through, but it does not seem to be respected as described here:
- https://github.com/astral-sh/uv/issues/1374
- https://github.com/astral-sh/uv/blob/main/docs/pip/environments.md

As if the directory would not complain PIP 405: https://peps.python.org/pep-0405/#specification
But it is a complete common Python environment 🤔.

MichaIng, 2024-10-03:
Hmm
```console
homeassistant@VM-Bookworm:~$ echo $VIRTUAL_ENV
/mnt/dietpi_userdata/homeassistant/deps
homeassistant@VM-Bookworm:~$ uv pip install pyotp==2.8.0
error: No virtual environment found; run `uv venv` to create an environment, or pass `--system` to install into a non-virtual environment
homeassistant@VM-Bookworm:~$ uv pip install --system pyotp==2.8.0
Resolved 1 package in 141ms
Prepared 1 package in 25ms
Installed 1 package in 1ms
 + pyotp==2.8.0
homeassistant@VM-Bookworm:~$
```
HA passes `--target /mnt/dietpi_userdata/homeassistant/deps`, which results in the same error, despite `VIRTUAL_ENV` passed. Need to check back with `uv` devs, how it evaluates whether it is a virtual env or not.

Dynamic5912, 2024-10-03:
> looks like restore is your uncle here till there is a perm fix :)  

How do you roll it back to a previous version?

And does this actually fix it?

MichaIng, 2024-10-03:
Thanks @Yougoshatenshi this is exactly what I would have suggested for now :)

Dynamic5912, 2024-10-03:
Downgrading worked a treat - thanks!

Just have to resist the urge to update again now... 😁

InnovoDeveloper, 2024-10-03:
I am getting 
`No matching distribution found for homeassistant==2024.9.3`
 when I try to roll it back.  What am I missing ?

MichaIng, 2024-10-03:
Let's see whether there is a different solution than adding an actual `pyenv virtuelenv`: https://github.com/astral-sh/uv/issues/7907

InnovoDeveloper, 2024-10-03:
> > I am getting
> > `No matching distribution found for homeassistant==2024.9.3`
> > when I try to roll it back.  What am I missing ?
> 
> Are you running the other commands as well?

yes, I am first running 
sudo -u homeassistant bash
. /home/homeassistant/pyenv-activate.sh

MichaIng, 2024-10-03:
@InnovoDeveloper 
Which SBC and architecture do you use?
```sh
echo $G_HW_MODEL_NAME $G_HW_ARCH_NAME $G_DISTRO_NAME
```

InnovoDeveloper, 2024-10-03:
NanoPi R5S/R5C (aarch64) aarch64 bookworm

> @InnovoDeveloper Which SBC and architecture do you use?
> 
> ```shell
> echo $G_HW_MODEL_NAME $G_HW_ARCH_NAME $G_DISTRO_NAME
> ```

****
NanoPi R5S/R5C (aarch64) aarch64 bookworm

InnovoDeveloper, 2024-10-03:
I am getting the following error, so this could be due to a different python version that I am running.  
python3 --version returns 3.11.2
I did dietpi-software reinstall 130 but that still gives me the same version. 

`
Defaulting to user installation because normal site-packages is not writeable
ERROR: Ignored the following versions that require a different python version: 2024.10.0 Requires-Python >=3.12.0; 2024.10.0b0 Requires-Python >=3.12.0; 2024.10.0b1 Requires-Python >=3.12.0; 2024.10.0b10 Requires-Python >=3.12.0; 2024.10.0b11 Requires-Python >=3.12.0; 2024.10.0b12 Requires-Python >=3.12.0; 2024.10.0b2 Requires-Python >=3.12.0; 2024.10.0b3 Requires-Python >=3.12.0; 2024.10.0b4 Requires-Python >=3.12.0; 2024.10.0b5 Requires-Python >=3.12.0; 2024.10.0b6 Requires-Python >=3.12.0; 2024.10.0b7 Requires-Python >=3.12.0; 2024.10.0b8 Requires-Python >=3.12.0; 2024.10.0b9 Requires-Python >=3.12.0; 2024.4.0 Requires-Python >=3.12.0; 2024.4.0b0 Requires-Python >=3.12.0; 2024.4.0b1 Requires-Python >=3.12.0; 2024.4.0b2 Requires-Python >=3.12.0; 2024.4.0b3 Requires-Python >=3.12.0; 2024.4.0b4 Requires-Python >=3.12.0; 2024.4.0b5 Requires-Python >=3.12.0; 2024.4.0b6 Requires-Python >=3.12.0; 2024.4.0b7 Requires-Python >=3.12.0; 2024.4.0b8 Requires-Python >=3.12.0; 2024.4.0b9 Requires-Python >=3.12.0; 2024.4.1 Requires-Python >=3.12.0; 2024.4.2 Requires-Python >=3.12.0; 2024.4.3 Requires-Python >=3.12.0; 2024.4.4 Requires-Python >=3.12.0; 2024.5.0 Requires-Python >=3.12.0; 2024.5.0b0 Requires-Python >=3.12.0; 2024.5.0b1 Requires-Python >=3.12.0; 2024.5.0b2 Requires-Python >=3.12.0; 2024.5.0b3 Requires-Python >=3.12.0; 2024.5.0b4 Requires-Python >=3.12.0; 2024.5.0b5 Requires-Python >=3.12.0; 2024.5.0b6 Requires-Python >=3.12.0; 2024.5.0b7 Requires-Python >=3.12.0; 2024.5.1 Requires-Python >=3.12.0; 2024.5.2 Requires-Python >=3.12.0; 2024.5.3 Requires-Python >=3.12.0; 2024.5.4 Requires-Python >=3.12.0; 2024.5.5 Requires-Python >=3.12.0; 2024.6.0 Requires-Python >=3.12.0; 2024.6.0b0 Requires-Python >=3.12.0; 2024.6.0b1 Requires-Python >=3.12.0; 2024.6.0b2 Requires-Python >=3.12.0; 2024.6.0b3 Requires-Python >=3.12.0; 2024.6.0b4 Requires-Python >=3.12.0; 2024.6.0b5 Requires-Python >=3.12.0; 2024.6.0b6 Requires-Python >=3.12.0; 2024.6.0b7 Requires-Python >=3.12.0; 2024.6.0b8 Requires-Python >=3.12.0; 2024.6.0b9 Requires-Python >=3.12.0; 2024.6.1 Requires-Python >=3.12.0; 2024.6.2 Requires-Python >=3.12.0; 2024.6.3 Requires-Python >=3.12.0; 2024.6.4 Requires-Python >=3.12.0; 2024.7.0 Requires-Python >=3.12.0; 2024.7.0b0 Requires-Python >=3.12.0; 2024.7.0b1 Requires-Python >=3.12.0; 2024.7.0b10 Requires-Python >=3.12.0; 2024.7.0b11 Requires-Python >=3.12.0; 2024.7.0b2 Requires-Python >=3.12.0; 2024.7.0b3 Requires-Python >=3.12.0; 2024.7.0b4 Requires-Python >=3.12.0; 2024.7.0b5 Requires-Python >=3.12.0; 2024.7.0b6 Requires-Python >=3.12.0; 2024.7.0b7 Requires-Python >=3.12.0; 2024.7.0b8 Requires-Python >=3.12.0; 2024.7.0b9 Requires-Python >=3.12.0; 2024.7.1 Requires-Python >=3.12.0; 2024.7.2 Requires-Python >=3.12.0; 2024.7.3 Requires-Python >=3.12.0; 2024.7.4 Requires-Python >=3.12.0; 2024.8.0 Requires-Python >=3.12.0; 2024.8.0b0 Requires-Python >=3.12.0; 2024.8.0b1 Requires-Python >=3.12.0; 2024.8.0b2 Requires-Python >=3.12.0; 2024.8.0b3 Requires-Python >=3.12.0; 2024.8.0b4 Requires-Python >=3.12.0; 2024.8.0b5 Requires-Python >=3.12.0; 2024.8.0b6 Requires-Python >=3.12.0; 2024.8.0b7 Requires-Python >=3.12.0; 2024.8.0b8 Requires-Python >=3.12.0; 2024.8.0b9 Requires-Python >=3.12.0; 2024.8.1 Requires-Python >=3.12.0; 2024.8.2 Requires-Python >=3.12.0; 2024.8.3 Requires-Python >=3.12.0; 2024.9.0 Requires-Python >=3.12.0; 2024.9.0b0 Requires-Python >=3.12.0; 2024.9.0b1 Requires-Python >=3.12.0; 2024.9.0b2 Requires-Python >=3.12.0; 2024.9.0b3 Requires-Python >=3.12.0; 2024.9.0b4 Requires-Python >=3.12.0; 2024.9.0b5 Requires-Python >=3.12.0; 2024.9.1 Requires-Python >=3.12.0; 2024.9.2 Requires-Python >=3.12.0; 2024.9.3 Requires-Python >=3.12.0
ERROR: Could not find a version that satisfies the requirement homeassistant==2024.9.3 (from versions: 0.7.0rc1, 0.7.0, 0.7.0.1, 0.7.1, 0.7.2, 0.7.3, 0.7.4, 0.7.5, 0.7.6, 0.7.7, 0.8.0, 0.9.0, 0.9.1, 0.10.0, 0.10.1, 0.11.0, 0.11.1, 0.12.0, 0.13.0, 0.13.1, 0.14.0, 0.14.1, 0.14.2, 0.15.0, 0.16.0, 0.16.1, 0.17.0, 0.17.1, 0.17.2, 0.17.3, 0.18.2, 0.19, 0.19.1, 0.19.2, 0.19.3, 0.19.4, 0.20.0, 0.20.1, 0.20.2, 0.20.3, 0.21.0, 0.21.1, 0.21.2, 0.22.0, 0.22.1, 0.23.0, 0.23.1, 0.24.0, 0.24.1, 0.25.0, 0.25.1, 0.25.2, 0.26.0, 0.26.1, 0.26.2, 0.26.3, 0.27.0, 0.27.1, 0.27.2, 0.28.0, 0.28.1, 0.28.2, 0.29.0, 0.29.1, 0.29.2, 0.29.3, 0.29.4, 0.29.5, 0.29.6, 0.29.7, 0.30.0, 0.30.1, 0.30.2, 0.31.0, 0.31.1, 0.32.0, 0.32.1, 0.32.2, 0.32.3, 0.32.4, 0.33.0, 0.33.1, 0.33.2, 0.33.3, 0.33.4, 0.34.0, 0.34.1, 0.34.2, 0.34.3, 0.34.4, 0.34.5, 0.35.0, 0.35.1, 0.35.2, 0.35.3, 0.36.0, 0.36.1, 0.37.0, 0.37.1, 0.38.0, 0.38.1, 0.38.2, 0.38.3, 0.38.4, 0.39.0, 0.39.1, 0.39.2, 0.39.3, 0.40.0, 0.40.1, 0.40.2, 0.41.0, 0.42.0, 0.42.1, 0.42.2, 0.42.3, 0.42.4, 0.43.0, 0.43.1, 0.43.2, 0.44.0, 0.44.1, 0.45.0, 0.45.1, 0.46.0, 0.46.1, 0.47.0, 0.47.1, 0.48.0, 0.48.1, 0.49.0, 0.49.1, 0.50.1, 0.50.2, 0.51.0, 0.51.1, 0.51.2, 0.52.0, 0.52.1, 0.53.0, 0.53.1, 0.54.0, 0.55.0, 0.55.1, 0.56.0, 0.56.1, 0.56.2, 0.57.0, 0.57.1, 0.57.2, 0.57.3, 0.58.0, 0.58.1, 0.59.0, 0.59.1, 0.59.2, 0.60.0, 0.60.1, 0.61.0, 0.61.1, 0.62.0, 0.62.1, 0.63.0, 0.63.1, 0.63.2, 0.63.3, 0.64.0, 0.64.1, 0.64.2, 0.64.3, 0.64.4b0, 0.65.0, 0.65.1, 0.65.2, 0.65.3, 0.65.4 [... cut ...] 4.1.0b8, 2024.1.0, 2024.1.1, 2024.1.2, 2024.1.3, 2024.1.4, 2024.1.5, 2024.1.6, 2024.2.0b0, 2024.2.0b1, 2024.2.0b2, 2024.2.0b3, 2024.2.0b4, 2024.2.0b5, 2024.2.0b6, 2024.2.0b7, 2024.2.0b8, 2024.2.0b9, 2024.2.0b10, 2024.2.0b11, 2024.2.0, 2024.2.1, 2024.2.2, 2024.2.3, 2024.2.4, 2024.2.5, 2024.3.0b0, 2024.3.0b1, 2024.3.0b2, 2024.3.0b3, 2024.3.0b4, 2024.3.0b5, 2024.3.0b6, 2024.3.0b7, 2024.3.0b8, 2024.3.0, 2024.3.1, 2024.3.2, 2024.3.3)
ERROR: No matching distribution found for homeassistant==2024.9.3
`

zanieb, 2024-10-03:
👋 I've provided some details on uv's virtual environment compliance checks and some possible workarounds at https://github.com/astral-sh/uv/issues/7907#issuecomment-2392184750

MichaIng, 2024-10-03:
@InnovoDeveloper 
You are mixing up the system Python instance and `pyenv` Python instance, that is used explicitly for Home Assistant. After running
```sh
sudo -u homeassistant bash
. /home/homeassistant/pyenv-activate.sh
```
can you run this again
```sh
python3 --version
```
It must be Python 3.12, otherwise Home Assistant 3.10 could not have been installed in the first place.

InnovoDeveloper, 2024-10-03:
> @InnovoDeveloper You are mixing up the system Python instance and `pyenv` Python instance, that is used explicitly for Home Assistant. After running
> 
> ```shell
> sudo -u homeassistant bash
> . /home/homeassistant/pyenv-activate.sh
> ```
> 
> can you run this again
> 
> ```shell
> python3 --version
> ```
> 
> It must be Python 3.12, otherwise Home Assistant 3.10 could not have been installed in the first place.

Yes, I was mixing up the two.  the python in the envelope is 3.12.3.  still cannot roll back.  I'll wait for a fix.

MichaIng, 2024-10-03:
This should fix it: https://github.com/MichaIng/DietPi/commit/e7d4053

And to not wait for the update:
```sh
sed -i '/^export /a\export UV_SYSTEM_PYTHON=1' /home/homeassistant/pyenv-activate.sh
```

Yougoshatenshi, 2024-10-03:
> This should fix it: [e7d4053](https://github.com/MichaIng/DietPi/commit/e7d4053)
> 
> And to not wait for the update:
> 
> ```shell
> sudo sed -i '/^export /a\export UV_SYSTEM_PYTHON=1' /home/homeassistant/pyenv-activate.sh
> ```

Thanks. Running 2024.10.0 with no issues.

Dynamic5912, 2024-10-03:
> This should fix it: https://github.com/MichaIng/DietPi/commit/e7d4053
> 
> 
> 
> And to not wait for the update:
> 
> ```sh
> 
> sudo sed -i '/^export /a\export UV_SYSTEM_PYTHON=1' /home/homeassistant/pyenv-activate.sh
> 
> ```
> 
> 
> 
> Running automated tests, though since some builds are running, it will take a few hours: https://github.com/MichaIng/DietPi/actions/runs/11166786788

Do we run this before upgrading HA or afterwards?

Or does it not make a difference?

Yougoshatenshi, 2024-10-03:
> 
> Do we run this before upgrading HA or afterwards?
> 
> Or does it not make a difference?

Before, so HA can initialize directly after the upgrade. Otherwise it does not make difference.

Dynamic5912, 2024-10-03:
Tested and working OK here now!

Dynamic5912, 2024-10-03:
Not sure if related - but now the Ring entity won't load in HA..

Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 22:09:58 (5 occurrences)
Last logged: 22:12:59

Error setting up entry XXX@XXX.XXX for ring
Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1264, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1296, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 994, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/components/ring/binary_sensor.py", line 48, in <module>
    capability=RingCapability.DING,
               ^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'RingCapability' has no attribute 'DING'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/components/ring/__init__.py", line 100, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/config_entries.py", line 2142, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1177, in async_get_platforms
    import_future.result()
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1177, in async_get_platforms
    import_future.result()
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1177, in async_get_platforms
    import_future.result()
  [Previous line repeated 6 more times]
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1165, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1090, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/loader.py", line 1284, in _load_platform
    raise ImportError(
ImportError: Exception importing homeassistant.components.ring.binary_sensor

MichaIng, 2024-10-03:
Yes there is another thing which needs to be changed. In the previous HA version, the `homeassistant/deps` directory was the directory of the internally used Python instance. Now after `uv pip install ... --target .../homeassistant/deps` is used, it is the `site-packages` directory, which previously was at `homeassistant/deps/lib/python3.12/site-packages`. So the symlink we create to merge the `pyenv` and HA internal Python package dirs, needs to be adjusted:

```sh
sudo rm -R /mnt/dietpi_userdata/homeassistant/deps
sudo ln -sf "/home/homeassistant/.pyenv/versions/3.12.*/lib/python3.12/site-packages" /mnt/dietpi_userdata/homeassistant/deps
sudo systemctl restart home-assistant
```

Dynamic5912, 2024-10-03:
> Yes there is another thing which needs to be changed. In the previous HA version, the `homeassistant/deps` directory was the directory of the internally used Python instance. Now after `uv pip3 install ... --target .../homeassistant/deps` is used, it is the `site-packages` directory, which previously was at `homeassistant/deps/lib/python3.12/site-packages`. So the symlink we create to merge the `pyenv` and HA internal Python package dirs, needs to be adjusted:
> 
> 
> 
> ```sh
> 
> sudo rm -R /mnt/dietpi_userdata/homeassistant/deps
> 
> sudo ln -sf "/home/homeassistant/.pyenv/versions/3.12.*/lib/python3.12/site-packages" /mnt/dietpi_userdata/homeassistant/deps
> 
> sudo systemctl restart home-
> 
> ```

Well that did something.. now HA won't start again after running the restart command...

MichaIng, 2024-10-03:
"won't" start? The symlink is valid? And can you show logs again?
```sh
ls -l /mnt/dietpi_userdata/homeassistant/deps
journalctl -u home-assistant
```

lukaszsobala, 2024-10-03:
The fix works for me as well. 

What does this `uv` thing do anyway?

whyisthisbroken, 2024-10-04:
So for my understanding:

1.
sudo sed -i '/^export /a\export UV_SYSTEM_PYTHON=1' /home/homeassistant/pyenv-activate.sh

2.
sudo rm -R /mnt/dietpi_userdata/homeassistant/deps
sudo ln -sf "/home/homeassistant/.pyenv/versions/3.12.*/lib/python3.12/site-packages" /mnt/dietpi_userdata/homeassistant/deps
sudo systemctl restart home-assistant

3.
Upgrade HA to the newest Version.

Is this the right sequence to fix it ?

Dynamic5912, 2024-10-04:
> "won't" start? The symlink is valid? And can you show logs again?
> 
> ```sh
> 
> ls -l /mnt/dietpi_userdata/homeassistant/deps
> 
> journalctl -u home-assistant
> 
> ```

_dietpi@dietpi:~$ ls -l /mnt/dietpi_userdata/homeassistant/deps
lrwxrwxrwx 1 root root 71 Oct  3 22:58 /mnt/dietpi_userdata/homeassistant/deps -> '/home/homeassistant/.pyenv/versions/3.12.*/lib/python3.12/site-packages'
dietpi@dietpi:~$_

Running journalctl.... yields far too many lines to post back here and takes forever to run

Dynamic5912, 2024-10-04:
Journal logs from today:

Oct 04 01:28:50 dietpi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
Oct 04 01:28:53 dietpi Home Assistant[470738]: Fatal Error: Unable to create library directory /mnt/dietpi_userdata/homeassistant/deps: [Errno 17] File exists: '/mnt/dietpi_userdata/homeassistant/deps'
Oct 04 01:28:53 dietpi systemd[1]: home-assistant.service: Main process exited, code=exited, status=1/FAILURE
Oct 04 01:28:53 dietpi systemd[1]: home-assistant.service: Failed with result 'exit-code'.
Oct 04 01:28:53 dietpi systemd[1]: home-assistant.service: Consumed 3.029s CPU time.
Oct 04 07:48:10 dietpi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
Oct 04 07:48:12 dietpi Home Assistant[471589]: Fatal Error: Unable to create library directory /mnt/dietpi_userdata/homeassistant/deps: [Errno 17] File exists: '/mnt/dietpi_userdata/homeassistant/deps'
Oct 04 07:48:12 dietpi systemd[1]: home-assistant.service: Main process exited, code=exited, status=1/FAILURE
Oct 04 07:48:12 dietpi systemd[1]: home-assistant.service: Failed with result 'exit-code'.
Oct 04 07:48:12 dietpi systemd[1]: home-assistant.service: Consumed 2.774s CPU time.
Oct 04 07:49:28 dietpi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
Oct 04 07:49:30 dietpi Home Assistant[471778]: Fatal Error: Unable to create library directory /mnt/dietpi_userdata/homeassistant/deps: [Errno 17] File exists: '/mnt/dietpi_userdata/homeassistant/deps'
Oct 04 07:49:30 dietpi systemd[1]: home-assistant.service: Main process exited, code=exited, status=1/FAILURE
Oct 04 07:49:30 dietpi systemd[1]: home-assistant.service: Failed with result 'exit-code'.
Oct 04 07:49:30 dietpi systemd[1]: home-assistant.service: Consumed 2.753s CPU time.
Oct 04 07:52:22 dietpi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
Oct 04 07:52:25 dietpi Home Assistant[471979]: Fatal Error: Unable to create library directory /mnt/dietpi_userdata/homeassistant/deps: [Errno 17] File exists: '/mnt/dietpi_userdata/homeassistant/deps'
Oct 04 07:52:25 dietpi systemd[1]: home-assistant.service: Main process exited, code=exited, status=1/FAILURE
Oct 04 07:52:25 dietpi systemd[1]: home-assistant.service: Failed with result 'exit-code'.
Oct 04 07:52:25 dietpi systemd[1]: home-assistant.service: Consumed 2.767s CPU time.
dietpi@dietpi:~$

whyisthisbroken, 2024-10-04:
Got the same errors with deps.
First try thought it runs but the whole journal is spammed with errors.

MichaIng, 2024-10-04:
Guys there were unintended double-quotes in my comment. Please repair like this (also if you run into the original issue for the first time):
```sh
sudo systemctl stop home-assistant
grep -q 'UV_SYSTEM_PYTHON' /home/homeassistant/pyenv-activate.sh || sudo sed -i '/^export /a\export UV_SYSTEM_PYTHON=1' /home/homeassistant/pyenv-activate.sh
sudo rm -R /mnt/dietpi_userdata/homeassistant/deps
sudo ln -sf /home/homeassistant/.pyenv/versions/3.12.*/lib/python3.12/site-packages /mnt/dietpi_userdata/homeassistant/deps
sudo systemctl restart home-assistant
```

Dynamic5912, 2024-10-04:
Hero - works like a charm now!!

7knips, 2024-10-05:
> ```shell
> sudo systemctl stop home-assistant
> sudo grep -q 'UV_SYSTEM_PYTHON' /home/homeassistant/pyenv-activate.sh || sudo sed -i '/^export /a\export UV_SYSTEM_PYTHON=1' /home/homeassistant/pyenv-activate.sh
> sudo rm -R /mnt/dietpi_userdata/homeassistant/deps
> sudo ln -sf /home/homeassistant/.pyenv/versions/3.12.*/lib/python3.12/site-packages /mnt/dietpi_userdata/homeassistant/deps
> sudo systemctl restart home-assistant
> ```

Works fine, thank you!

`export UV_SYSTEM_PYTHON=1` is duplicated in my pyenv-activate.sh - have removed redundant one:

                                                                                                                                                  
```diff
#!/bin/dash                                                                                                                                                                        
if [ $(whoami) != 'homeassistant' ]; then                                                                                                                                          
        echo '[FAILED] This pyenv must be activated as user "homeassistant". Aborting...'                                                                                          
        kill -INT $$                                                                                                                                                               
fi                                                                                                                                                                                 
cd /home/homeassistant                                                                                                                                                             
export PYENV_ROOT='/home/homeassistant/.pyenv'                                                                                                                                     
- export UV_SYSTEM_PYTHON=1                                                                                     
export PATH="$PYENV_ROOT/bin:$PATH"                                                                                                                                                
export UV_SYSTEM_PYTHON=1                                                                                                                                                          
eval "$(pyenv init --path)"                                                                                                                                                        
eval "$(pyenv init -)"                                                                                                                                                             
[ -f '.cargo/env' ] && . .cargo/env
```

domenicoblanco, 2024-10-05:
> The problem is `--system` would need to be added to the `uv` call itself, which is in HA code. An alternative is to export `VIRTUAL_ENV`, but it seems along the `pyenv` > `hass` > `uv` calls, the environment is lost, or at least this variable, so it does not work either.
> 
> A bigger problem might be that this way, also the Rust environment would be lost, potentially breaking installs on 32-bit ARM and RISC-V systems.
> 
> @domenicoblanco method means a completely different Python instance environment, not using `pyenv` at all, but the system Python instance, which also does not work on Bullseye, and will break future HA versions on current Debian, as fast as HA drops older Python support. So this is not a solution we can consider. If at all, we would create a true virtualenv within `pyenv`, but I would like to avoid that double-env wrapping, which is nonsense here, as this `pyenv` instance is exclusive for HA already.

Sorry for the delay, but technically my method uses a virtual environment, the only global package is uv.
I edited the actual `pyenv-activate.sh` to start the venv and install packages within it.
I agree that's not the most elegant solution, but I hadn't so much time and went straightforward

MichaIng, 2024-10-05:
> Sorry for the delay, but technically my method uses a virtual environment, the only global package is uv.

Yes, but a `venv` of the system Python instance, instead of the `pyenv` (which is a complete dedicated Python instance). With Debian Bookworm and Trixie it currently works that way, but it will break in the future, and it does not work on Debian Bullseye. But no problem as long as it works for you, I just explained why this is no general fix, or one we can consider, as someone was asking 🙂.

domenicoblanco, 2024-10-05:
Oh okay, I didn't notice 😅
I added that comment just to understand if I wrote something wrong or maybe a misread c:

machejc, 2024-10-08:
I reverted back to 9.3. How many weeks before this is sorted out and I can just upgrade from the dietpi server like I normally do?

MichaIng, 2024-10-08:
Either follow this: https://github.com/MichaIng/DietPi/issues/7231#issuecomment-2393485026
Or wait until tomorrow, latest Thursday, for next DietPi release.

machejc, 2024-10-09:
Thanks, I will wait until you release a DietPi update.

Dynamic5912, 2024-10-09:
> Thanks, I will wait until you release a DietPi update.
> 
> 
> 
> 

Pretty sure the DietPi update will just contain the above fix anyway...

So either way you'll get the same end result.

MichaIng, 2024-10-09:
The DietPi update will have `dietpi-software` updated so that `dietpi-software reinstall 157` will setup everything correctly. We should probably add this to the changelog.

`/home/homeassistant/homeassistant-update.sh` really only updates the `homeassistant` Python module, and in case needed dependencies. But if there are structural changes required, or a newer Python version, a reinstall as of above needs to be done.

Above set of commands is hence a faster fix, bypassing the Python re-compilation etc. But it would be too much magic to add all this into the `dietpi-update` patch script, including a check which HA version is used, whether it has been patched for `uv` support already or is currently in broken state, etc.

KoriKosmos, 2024-10-16:
Sorry to necro what seems to be a resolved issue, but I get a problem when trying to do the above fix:

`Oct 16 20:08:02 DietPi systemd[1]: home-assistant.service: Consumed 1min 14.850s CPU time.
Oct 16 20:08:02 DietPi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
Oct 16 20:08:09 DietPi Home Assistant[85236]: 2024-10-16 20:08:09.330 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 16 20:08:10 DietPi Home Assistant[85236]: 2024-10-16 20:08:10.481 WARNING (MainThread) [aiohttp_fast_zlib] zlib_ng and isal are not available, falling back to zlib, performance will be degraded.
Oct 16 20:08:11 DietPi Home Assistant[85236]: 2024-10-16 20:08:11.364 ERROR (ImportExecutor_0) [homeassistant.components.camera.img_util] Error loading libturbojpeg; Camera snapshot performance will be sub-optimal
Oct 16 20:08:11 DietPi Home Assistant[85236]: Traceback (most recent call last):
Oct 16 20:08:11 DietPi Home Assistant[85236]:   File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/site-packages/homeassistant/components/camera/img_util.py", line 100, in __init__
Oct 16 20:08:11 DietPi Home Assistant[85236]:     TurboJPEGSingleton.__instance = TurboJPEG()
Oct 16 20:08:11 DietPi Home Assistant[85236]:                                     ^^^^^^^^^^^
Oct 16 20:08:11 DietPi Home Assistant[85236]:   File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/site-packages/turbojpeg.py", line 300, in __init__
Oct 16 20:08:11 DietPi Home Assistant[85236]:     self.__find_turbojpeg() if lib_path is None else lib_path)
Oct 16 20:08:11 DietPi Home Assistant[85236]:     ^^^^^^^^^^^^^^^^^^^^^^^
Oct 16 20:08:11 DietPi Home Assistant[85236]:   File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/site-packages/turbojpeg.py", line 967, in __find_turbojpeg
Oct 16 20:08:11 DietPi Home Assistant[85236]:     raise RuntimeError(
Oct 16 20:08:11 DietPi Home Assistant[85236]: RuntimeError: Unable to locate turbojpeg library automatically. You may specify the turbojpeg library path manually.
Oct 16 20:08:11 DietPi Home Assistant[85236]: e.g. jpeg = TurboJPEG(lib_path)`

RPi 4 Model B (aarch64) aarch64 bookworm

MichaIng, 2024-10-16:
The camera component requires the libjpeg library:
```sh
apt install libjpeg62-turbo
systemctl restart home-assistant
```

closed 2024-10-18 by MichaIng