MichaIng/DietPi #7237: OrangePI 5+ error after kernel update to 24.11.0-trunk-dietpi1 (cgroup: fork rejected by pids controller...)

ctraber, 2024-10-11:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version 9.7.1
- Distro version bookworm
- Kernel version 6.1.75-vendor-rk35xx
- SBC model Orange Pi 5 Plus (aarch64)

After update to new kernel packages:
armbian-firmware 24.11.0-trunk-dietpi1 
linux-dtb-vendor-rk35xx 24.11.0-trunk-dietpi1
linux-image-vendor-rk35xx 24.11.0-trunk-dietpi1

k3s cannot start additional processes. dmesg e.g.:
cgroup: fork rejected by pids controller in /kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podf03920d0_4917_430d_a15e_a25e63ff86bb.slice/cri-containerd-b2a0a20a1bf54f3630971193120d6f66f096cbbe447ae006c519bb35bad2ab97.scope

downgrade to 24.8.0 solved the problem.

MichaIng, 2024-10-12:
For now I took the kernel off the repository, since it broke booting from NVMe SSD as well. Will start to investigate for next release.

ctraber, 2024-10-14:
Booting from NVMe worked without problem here.

MichaIng, 2024-10-14:
There is a bug in the latest vendor kernel sources which breaks NVMe boot on Orange Pi 5 and 5 Plus, not sure whether this one is related. Armbian however also froze the kernel to 24.8 for those two SBCs (and some others): https://github.com/armbian/build/commit/edbf678

So we did the same, monitoring the freeze at Armbian. Sadly, the original commit was done without any comment, and also subsequent commits did not explain why it is actually frozen. So not sure about the underlying issue. Armbian's vendor kernel sources for RK3588 are based on Radxa sources, so either ROCK 5 family will work fine, but commits they do on top of the base source from Rockchip can theoretically always break (features on) any other RK3588 board. Mainline kernel would enhance the situation, but it still lacks a lot of features.

I'll close the issue, and as fast as Armbian unfreezes the kernel for these boards, we'll provide a package for testing, and provide it via APT in case it does not cause the two known issues (NVMe boot and K3s/cgroups).

ctraber, 2025-01-03:
After upgrade to dietpi 9.9 kernel 24.11.0-trunk-dietpi1 is back and is causing the same problems as before.

MichaIng, 2025-01-04:
Means the NVMe boot issue (which is solved with this build) was not related to your K3s issue.

ctraber, 2025-01-04:
Thanks for the quick reply.

Tried the new kernel and there are no error messages with 6.1.84-vendor-rk35xx #1 SMP Tue Dec 24 13:36:04 UTC 2024 aarch64 GNU/Linux. 

Seems to work. :-)
Will monitor...

```
systemctl cat k3s-agent
...
[Service]
Type=notify
EnvironmentFile=-/etc/default/%N
EnvironmentFile=-/etc/sysconfig/%N
EnvironmentFile=-/etc/systemd/system/k3s-agent.service.env
KillMode=process
Delegate=yes
# Having non-zero Limit*s causes performance problems due to accounting overhead
# in the kernel. We recommend using cgroups to do container-local accounting.
LimitNOFILE=1048576
LimitNPROC=infinity
LimitCORE=infinity
TasksMax=infinity
TimeoutStartSec=0
Restart=always
...
```

```
sysctl kernel.pid_max
kernel.pid_max = 4194304
```

MichaIng, 2025-01-04:
Ah great, so it was fixed upstream or at Armbian already. I'll push that build to the APT repo ASAP.

MichaIng, 2025-01-04:
I pushed the new kernel to our APT repository.

closed 2025-02-09 by MichaIng