MichaIng/DietPi #7238: Radarr stopped working

shlatchz, 2024-10-12:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm
- Kernel version | `uname -a`
Linux DietPi 6.1.0-26-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Native PC (x86_64)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`
3b0cd5c5-8000-40c6-ac83-14972cc6877b

```
root@DietPi:~# journalctl -u radarr.service --since today
Journal file /var/log/journal/455a81b1b9734a369117844dfd7e4a58/system.journal is truncated, ignoring file.
Oct 12 14:14:07 DietPi systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 12 14:14:07 DietPi (Radarr)[674]: radarr.service: Failed to locate executable /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:07 DietPi (Radarr)[674]: radarr.service: Failed at step EXEC spawning /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:07 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=203/EXEC
Oct 12 14:14:07 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Scheduled restart job, restart counter is at 1.
Oct 12 14:14:08 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:14:08 DietPi systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 12 14:14:08 DietPi (Radarr)[733]: radarr.service: Failed to locate executable /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:08 DietPi (Radarr)[733]: radarr.service: Failed at step EXEC spawning /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=203/EXEC
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Scheduled restart job, restart counter is at 2.
Oct 12 14:14:08 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:14:08 DietPi systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 12 14:14:08 DietPi (Radarr)[739]: radarr.service: Failed to locate executable /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:08 DietPi (Radarr)[739]: radarr.service: Failed at step EXEC spawning /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=203/EXEC
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Scheduled restart job, restart counter is at 3.
Oct 12 14:14:08 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:14:08 DietPi systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 12 14:14:08 DietPi (Radarr)[750]: radarr.service: Failed to locate executable /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:08 DietPi (Radarr)[750]: radarr.service: Failed at step EXEC spawning /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=203/EXEC
Oct 12 14:14:08 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:14:09 DietPi systemd[1]: radarr.service: Scheduled restart job, restart counter is at 4.
Oct 12 14:14:09 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:14:09 DietPi systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 12 14:14:09 DietPi (Radarr)[755]: radarr.service: Failed to locate executable /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:09 DietPi (Radarr)[755]: radarr.service: Failed at step EXEC spawning /opt/radarr/Radarr: No such file or directory
Oct 12 14:14:09 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=203/EXEC
Oct 12 14:14:09 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:14:09 DietPi systemd[1]: radarr.service: Scheduled restart job, restart counter is at 5.
Oct 12 14:14:09 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:14:09 DietPi systemd[1]: radarr.service: Start request repeated too quickly.
Oct 12 14:14:09 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:14:09 DietPi systemd[1]: Failed to start radarr.service - Radarr (DietPi).
[... 172 lines cut ...]
Oct 12 14:53:59 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:53:59 DietPi systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 12 14:53:59 DietPi (Radarr)[9903]: radarr.service: Failed to locate executable /opt/radarr/Radarr: No such file or directory
Oct 12 14:53:59 DietPi (Radarr)[9903]: radarr.service: Failed at step EXEC spawning /opt/radarr/Radarr: No such file or directory
Oct 12 14:53:59 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=203/EXEC
Oct 12 14:53:59 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:54:00 DietPi systemd[1]: radarr.service: Scheduled restart job, restart counter is at 4.
Oct 12 14:54:00 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:54:00 DietPi systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 12 14:54:00 DietPi (Radarr)[9904]: radarr.service: Failed to locate executable /opt/radarr/Radarr: No such file or directory
Oct 12 14:54:00 DietPi (Radarr)[9904]: radarr.service: Failed at step EXEC spawning /opt/radarr/Radarr: No such file or directory
Oct 12 14:54:00 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=203/EXEC
Oct 12 14:54:00 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:54:00 DietPi systemd[1]: radarr.service: Scheduled restart job, restart counter is at 5.
Oct 12 14:54:00 DietPi systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 12 14:54:00 DietPi systemd[1]: radarr.service: Start request repeated too quickly.
Oct 12 14:54:00 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:54:00 DietPi systemd[1]: Failed to start radarr.service - Radarr (DietPi).
Oct 12 14:54:02 DietPi systemd[1]: radarr.service: Start request repeated too quickly.
Oct 12 14:54:02 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:54:02 DietPi systemd[1]: Failed to start radarr.service - Radarr (DietPi).
Oct 12 14:54:04 DietPi systemd[1]: radarr.service: Start request repeated too quickly.
Oct 12 14:54:04 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 12 14:54:04 DietPi systemd[1]: Failed to start radarr.service - Radarr (DietPi).
```

shlatchz, 2024-10-12:
reinstalling didn't help

MichaIng, 2024-10-12:
Hmm, the executable `/opt/radarr/Radarr` is missing or not seen by the service in your case. Can you check this:
```sh
ls -dl /opt/radarr/Radarr /opt/radarr
systemctl cat radarr
```
On a test install it works fine here.

shlatchz, 2024-10-17:
```
root@DietPi:~# ls -dl /opt/radarr/Radarr /opt/radarr
systemctl cat radarr
ls: cannot access '/opt/radarr/Radarr': No such file or directory
drwxr-xr-x 2 radarr dietpi 32768 Oct  2 15:35 /opt/radarr
# /etc/systemd/system/radarr.service
[Unit]
Description=Radarr (DietPi)
Wants=network-online.target
After=network-online.target remote-fs.target transmission-daemon.service qbittorrent.service rtorrent.service nzbget.service deluged.service aria2.service sabnzbd.service

[Service]
SyslogIdentifier=Radarr
User=radarr
UMask=002
LogsDirectory=radarr
ExecStart=/opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr
TimeoutStopSec=20
KillMode=process
Restart=on-failure

# Hardening
ProtectSystem=strict
ProtectHome=true
PrivateDevices=true
ProtectKernelTunables=true
ProtectControlGroups=true
ReadWritePaths=-/opt/radarr -/mnt -/media -/var/log/radarr -/tmp

[Install]
WantedBy=multi-user.target

# /etc/systemd/system/radarr.service.d/dietpi-process_tool.conf
# WARNING: Do not manually edit this file, use "dietpi-services" to adjust values!
[Service]
CPUSchedulingPolicy=idle
IOSchedulingClass=idle

# /etc/systemd/system/radarr.service.d/dietpi-services_edit.conf
[Unit]
#Description=Radarr Daemon (DietPi)
#Wants=network-online.target
#After=network-online.target dietpi-boot.service

[Service]
#SyslogIdentifier=Radarr
User=radarr
#UMask=002
Environment="TMPDIR=/tmp"
#ExecStart=/usr/bin/mono -O=-aot /opt/Radarr/Radarr.exe -nobrowser -data=/mnt/dietpi_userdata/radarr
#TimeoutStopSec=20
#KillMode=process
#Restart=on-failure

# Hardening
#ProtectSystem=strict
#ProtectHome=true
#PrivateDevices=true
#ProtectKernelTunables=true
#ProtectControlGroups=true
#ReadWritePaths=/opt/Radarr /mnt /media /var/log/radarr /tmp

[Install]
#WantedBy=multi-user.target
```

shlatchz, 2024-10-17:
how can I fix this without losing my metadata?

Joulinar, 2024-10-17:
can you share following pls

`ls la /opt/radarr/`

MichaIng, 2024-10-17:
You can reinstall it to fix it, which preserves config and data:
```sh
dietpi-software reinstall 145
```
However, would be indeed interesting to see the content of the directory, to maybe get a clue why the executable is missing.

MichaIng, 2024-10-18:
Ah right there is a message that an existing installation was found an will be preserved, right? We really need a repair command, or do it OOTB incase of Servarr apps. Please do this:
```sh
rm -R /opt/radarr
dietpi-software reinstall 145
```

shlatchz, 2024-10-18:
works! thx

MichaIng, 2024-10-18:
Great, I'll mark the issue as closed then.

closed 2024-10-18 by MichaIng