MichaIng/DietPi #7239: [software update request] new update is available for Vaultwarden

yuukiAme, 2024-10-12:
Hi,

Can you guys update the repo and compile the newest version of Vaultwarden?
Thank you very much
[Release note](https://github.com/dani-garcia/vaultwarden/releases/tag/1.32.1) for version 1.32.1.

MichaIng, 2024-10-12:
Just done :)

yuukiAme, 2024-10-13:
Thank you very much. Quickly as always :)

MichaIng, 2024-10-14:
v1.32.2 build is already running, just in case one might ask 😉: https://github.com/MichaIng/DietPi/actions/runs/11327418865

yuukiAme, 2024-11-11:
@MichaIng 
New CVE in Vaultwarden v1.32.3. 
Could you please update the Vaultwarden to v1.32.4 as soon as possible?
Thank you very much.

sonnatager, 2024-11-11:
When i try to update vaultwarden, it says always i am up to date:
[image]

Joulinar, 2024-11-11:
Yes as well need to build a new package first

sonnatager, 2024-11-11:
I saw you have triggered a new run for vaultwarden: https://github.com/MichaIng/DietPi/actions/runs/11774771490

That means, a new package is available?

So when i run
`apt update`
`apt upgrade vaultwarden
`
there should be a new update?

I have an Odroid N2 (aarch64)

Joulinar, 2024-11-11:
No, packages are stored within testing directory only. Like "[https://dietpi.com/downloads/binaries/bookworm/testing](https://dietpi.com/downloads/binaries/bookworm/testing/)

We would need to move them to production still. Something @MichaIng need to do.

MichaIng, 2024-11-11:
Just done, please test whether everything works fine for you with the new version.

yuukiAme, 2024-11-19:
@MichaIng 

"another one" in DJ Khaled's voice

[v1.32.5](https://github.com/dani-garcia/vaultwarden/releases/tag/1.32.5) fixes further CVE.

Could you please update Vaultwarden again? 

Thank you very much.

MichaIng, 2024-11-19:
No need to ping us here each time. I am subscribed to vaultwarden releases, and trigger builds as fast as I have access to GitHub: https://github.com/MichaIng/DietPi/actions/runs/11892355002

closed 2024-11-19 by MichaIng

Joulinar, 2025-02-06:
once package is available on our package servers, you can use regular Debian `apt` command like

```
apt update
```