MichaIng/DietPi #7241: Issue with Auto-Start Option 17: Custom Script Not Executed

loss-and-quick, 2024-10-14:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm 1`
- Kernel version | `Linux  6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- SBC model | `RPi Zero W (armv6l)`
- Power supply used | 5V 2.1A PowerBank
- SD card used |  Samsung EVO Plus

#### Additional Information (if applicable)
- Software title | DietPI
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID | `737d8199-70ad-42e2-b71c-241ec47b48ef`

#### Steps to reproduce

1. Choose option 17 in autostart menu
2. Edit custom.sh

#### Expected behaviour

- Run commands from custom.sh

#### Actual behaviour

- Not running commands, just autologin

#### Extra details

- No info about option 17

loss-and-quick, 2024-10-14:
My `/var/lib/dietpi/dietpi-autostart/custom.sh`:
``` bash
#!/bin/bash
# DietPi-AutoStart custom script
# Location: /var/lib/dietpi/dietpi-autostart/custom.sh
(poetry run -C /home/dietpi/earing python3 -u earing/earing/main.py | while IFS= read -r line; do echo "earning: $line"; done | logdy-core forward 8123) &
touch /home/dietpi/test
exit 0
```

MichaIng, 2024-10-14:
Many thanks for your report.

What do you see on the console after auto-login? There should be this line:
```
[ INFO ] Starting DietPi-AutoStart custom script...
```
followed by the output of the `custom.sh` script.

loss-and-quick, 2024-10-15:
@MichaIng I'm sorry, but I switched to Alpine. When the problem arose, I searched all the files in the DietPi source code at /var/lib/dietpi/dietpi-autostart/custom.sh and realized that option 17 is not implemented at all. Specifically, when selecting option 17, a console-getty service is created, along with another similar one with comparable content, and nothing else is created. Additionally, there is nothing in the DietPi logs related to autostart.

I suspect that option 17 is fundamentally unimplemented.

MichaIng, 2024-10-15:
It is implemented here, like all autologin + foreground options 😉: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L85

loss-and-quick, 2024-10-15:
> It is implemented here, like all autologin + foreground options 😉: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L85

But I tried to search in system using grep, and didn't find anything like that

I figured only that agetty creates shell...

MichaIng, 2024-10-15:
That should be in `/boot/dietpi/dietpi-login, which is loaded from `/etc/bashrc.d/dietpi.bash`, which is loaded from `/etc/bash.bashrc`, which is loaded on every interactive `bash` session, which is loaded (by default) on `agetty` logins.

closed 2025-06-02 by MichaIng