MichaIng/DietPi #7242: Orange Pi Zero 3 | 1GB model shows up as a 2GB model after a reboot at random

bigross8, 2024-10-14:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bookworm
- Kernel version | Linux zero3 6.6.44-current-sunxi64 SMP Sat Aug  3 06:54:42 UTC 2024 aarch64 GNU/Linux
- SBC model | Orange Pi Zero 3
- Power supply used | 5V 3A Raspberry Pi Adaptor
- SD card used | SanDisk Ultra 64GB

#### Steps to reproduce
1. Check RAM size (either with free, htop or RAM usage in dietpi-banner
2. Reboot and repeat

#### Expected behaviour
Show up as having 1GB of RAM on a 1GB model (919 MiB)

#### Actual behaviour
Intermittently shows up as having 2GB of RAM (1923 MiB)

#### Extra details
Randomly on a 6th reboot, the board thinks it has 2GB of RAM:

[image]

Then, I needed to reboot twice in order to show up with the correct amount of RAM:

[image]

MichaIng, 2024-10-14:
Many thanks for your report.

The RAM size is obtained by the bootloader. How old is your system/image, and in case did you flash the new bootloader build?
```sh
/boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
reboot
```

bigross8, 2024-10-14:
I just got it running and set up today. I have not tried flashing the bootloader, I tried the commands you provided and rebooted. The board showed up with 2GB of RAM once again.

bigross8, 2024-10-14:
Yes I do have the adapter, this is the output:
```
U-Boot SPL 2024.01-armbian-2024.01-S866c-P667c-Ha9af-V5c1c-Bda0a-R448a (Oct 05 2024 - 02:23:53 +0000)
DRAM base address is defined as 0x40000000
DRAM has 16 b/raw, 10 b/col, 4 B/width, 1 #rank and 8 #bank
DRAM top address must be less than 0x80000000
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.2(debug):armbian
NOTICE:  BL31: Built : 02:22:56, Oct  5 2024
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0b2eb8, model: OrangePi Zero3
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for erratum 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for erratum 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.
ns16550_serial serial@5000000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

U-Boot 2024.01-armbian-2024.01-S866c-P667c-Ha9af-V5c1c-Bda0a-R448a (Oct 05 2024 - 02:23:53 +0000) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero3
DRAM:  2 GiB
Core:  57 devices, 25 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Allwinner mUSB OTG (Peripheral)
Net:   Unsupported value 13, using default (13)
Unsupported value 13, using default (13)
eth0: ethernet@5020000using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in
MAC de:ad:be:ef:00:01
HOST MAC de:ad:be:ef:00:00
RNDIS ready
, eth1: usb_ether
starting USB...
Bus usb@5200000: USB EHCI 1.00
Bus usb@5200400: USB OHCI 1.0
scanning bus usb@5200000 for devices... 2 USB Device(s) found
scanning bus usb@5200400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 1 Storage Device(s) found
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2765 bytes read in 2 ms (1.3 MiB/s)
## Executing script at 4fc00000
327 bytes read in 1 ms (319.3 KiB/s)
23443464 bytes read in 969 ms (23.1 MiB/s)
30496581 bytes read in 1260 ms (23.1 MiB/s)
35795 bytes read in 5 ms (6.8 MiB/s)
Working FDT set to 4fa00000
Moving Image from 0x40080000 to 0x40200000, end=418f0000
## Loading init Ramdisk from Legacy Image at 4ff00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
[... 87 lines cut ...]
Working FDT set to 4fa00000
Moving Image from 0x40080000 to 0x40200000, end=418f0000
## Loading init Ramdisk from Legacy Image at 4ff00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    30496517 Bytes = 29.1 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
Working FDT set to 4fa00000
   Loading Ramdisk to 482ea000, end 49fff705 ... OK
   Loading Device Tree to 00000000482de000, end 00000000482e9bd2 ... OK
Working FDT set to 482de000

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
...
```

I put these outputs into diffchecker and these are the lines that are different between them:

[image]

MichaIng, 2024-10-14:
Thanks, well outlined the difference. Weird that it detects 16 b/row sometimes, and 15 b/row the other times, as this should be a static hardware value 🤔.

MichaIng, 2024-10-14:
So this is in the SPL. I don't know U-Boot code well. Could not find the sentences of these logs in the code. But there are some functions in the interface which are meant to be defined by boards: https://github.com/u-boot/u-boot/blob/master/include/init.h#L90-L91

That one is not defined, but in general DRAM init code for this SoC is here: https://github.com/u-boot/u-boot/blob/master/arch/arm/mach-sunxi/dram_sun50i_h616.c

... ahh and now I find these messages in an Armbian patch: https://github.com/armbian/build/blob/main/patch/u-boot/u-boot-sunxi/opizero3-1.5GB-trim-from-u-boot-v2024.01.patch

The upper part is needed to detect and assign 1.5 GB RAM correctly. The bottom part however ... ah, it just adds debugging messages. The issue hence is `config->rows` which is hence not related to that patch, but wrong in the first place.

The values are derived here: https://github.com/u-boot/u-boot/blob/master/arch/arm/mach-sunxi/dram_sun50i_h616.c#L1363
I do not 100% understand the code, but it generally loops through a possible number of bits per column and row, derives the memory address/offset that would match the begin of RAM, and tests whether this is really the case, by writing and comparing values to the two addresses. This actually seems to be a correct method to me. The only value we do not see in logs, which could affect the outcome, is `config->bus_full_width`, which can be 0 or 1. If it should be 1 but is sometimes falsely detected as 0, it would lead to rows being detected as 1 bit larger, indeed. However, it would also make columns 1 bit larger, which is not the case. ... ah and `u8 width = config->bus_full_width ? 4 : 2;` => as the width is reported to be 4 in both your logs, `bus_full_width` must be 1 in both cases.

Here is the two functions which test whether the two addresses are the same: https://github.com/u-boot/u-boot/blob/master/arch/arm/mach-sunxi/dram_helpers.c#L28-L64
I do not see how this can fail either. Maybe if writing to both addresses fails, and both their values initially were different. But I guess the whole SPL would the fail.

MichaIng, 2024-10-14:
I mailed the author of the U-Boot patch, in the hope he might have some idea how this can go randomly wrong.

vickash, 2024-10-18:
@MichaIng thanks for mentioning. Will keep an eye on this thread. If there's anything I can test on my Zero 2W to help, let me know.

jozefSCK, 2025-03-01:
not sure if it could influence thsi behavior, but pin 272 is used by kernel, however has to be programable GPIO. 
line 272:      unnamed       kernel   input  active-high [used]

jozefSCK, 2025-03-01:
Temporary solution for this issue.
# orange-pi-3-zero-1GB-version
script which chceck if ram is bigger that it should. so it restarts system. It will solve (not idealy) issue with ram size in Orange pi 3 Zero 1GB

1. Create the script

Save your script as /usr/local/bin/ram_check.sh and make it executable:
```
sudo nano /usr/local/bin/ram_check.sh
```
Paste your script inside and save (Ctrl + X, then Y, then Enter).

Make it executable:
```
sudo chmod +x /usr/local/bin/ram_check.sh
```
2. Create a systemd service

Create a new systemd service file:
```
sudo nano /etc/systemd/system/ramCheckSize.service
```
Paste the following content:
```
[Unit]
Description=Check RAM size at boot and reboot if >1GB
After=network.target

[Service]
ExecStart=/usr/local/bin/ram_check.sh
Type=oneshot
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```
Save and exit.
3. Enable and start the service

Run these commands to reload systemd, enable the service at boot, and start it immediately:
```
sudo systemctl daemon-reload
sudo systemctl enable ramCheckSize.service
sudo systemctl start ramCheckSize.service
```
4. Check the status

To verify if the service ran successfully:
```
sudo systemctl status ramCheckSize.service
```
Now, every time the system boots, it will check the RAM size and restart if it’s greater than 1GB.

jozefSCK, 2025-03-02:
Thanx for your corrections. I am just begginer. I am learning with AI assistance. I know it is not perfect at all, but still something. Here is ram_check.sh script.
```
#!/bin/bash

ram_size=$(free -m | grep Mem | awk '{print $2}')

if [ $ram_size -gt 1024 ]; then
    echo "RAM is bigger than 1 GB ($ram_size MB). System is restarting..."
    sudo reboot
else
    echo "RAM is 1 GB ($ram_size MB). As it should."
fi
```

jozefSCK, 2025-03-02:
@MichaIng 
All-in-one solution looks fine. tried couple of times.
```
[Unit]
Description=Check RAM size at boot and reboot if >1GB
DefaultDependencies=no
Before=systemd-remount-fs.service

[Service]
Type=oneshot
ExecStart=/bin/bash -c '(( $(free -m | mawk "/^Mem:/{print \$2;exit}") > 1024 )) && reboot'

[Install]
WantedBy=multi-user.target
```

MichaIng, 2025-05-06:
I pushed new kernel and U-Boot packages to our APT repository. Could you test whether this fixes things, the U-Boot upgrade in particular?
```sh
sudo apt update
sudo apt update
sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
sudo reboot
```

jozefSCK, 2025-05-15:
> I pushed new kernel and U-Boot packages to our APT repository. Could you test whether this fixes things, the U-Boot upgrade in particular?
> 
> sudo apt update
> sudo apt upgrade
> sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
> sudo reboot

Sorry, now using diffrent os and have no free sdcard for such a thing. But when recieved new sd...i will do test.

MichaIng, 2025-05-15:
Does this other distro not suffer from the by times false RAM size detection? In case its bootloader is generally based on mainline U-Boot, we might derive the needed patch from its sources.

vickash, 2025-05-15:
@MichaIng, sorry, haven't kept up with this.

> sudo apt update
> sudo apt upgrade
> sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
> sudo reboot

Just didt this on my 1GB board, and restarted a few times. First 5 or so showed 1GB, then this:
[image]

jozefSCK, 2025-05-23:
Sorry for so late respong Michal. I am using Armbiang bookworm stable. I havent noticed any issue with ram size.

MichaIng, 2025-05-23:
@jozefSCK 
Hmm, if it was fixed in Armbian, it would be fixed in DietPi as well, as we use the same kernel and U-Boot sources. It is as well Linux 6.12, right?

MichaIng, 2025-05-23:
Yeah, probably coincidence then. Or did you forgot and have the RAM size check service in place, to apply an automatic reboot if detected falsely? 😉

vickash, 2025-05-24:
> cd /tmp
> wget https://dietpi.com/downloads/binaries/testing/linux-u-boot-orangepizero2w-current.deb
> sudo dpkg -i linux-u-boot-orangepizero2w-current.deb
> sudo /boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
> sudo reboot

@MichaIng just did this to my Zero 2W 1GB. 15 reboots without issue. Seems to be fixed finally.🤞

MichaIng, 2025-05-25:
Spoke too soon. It breaks support for the 1.5 GB variants at least, and another user reports unbootable system with it on 1 GB variant: https://github.com/MichaIng/DietPi/issues/7545#issuecomment-2907823366

Slowly I am thinking to throw support for these to SBCs to the bin. With such critical issues open all over the place ...

milkywade, 2025-05-25:
Hi - I performed above actions for Orange Pi Zero 3, my 1gb-that-identifies-as-2gb board still shows 2gb.
I don't have a USB-UART adapter but could connect the board to a screen via hdmi, would that help ?

MichaIng, 2025-05-25:
U-Boot sadly does not start displays usually, at least never in the relevant early SPL stage. So your board would then be really interesting to debug, to see how a complete repetitive test pattern can fail so completely to detect that a RAM size cannot be larger in the current test iteration. That would mean a very basic false assumption in the code. But it cannot be debugged without access to the serial debug console.

But at least both of your boards boot now with the new U-Boot version?

milkywade, 2025-05-25:
Booting is finally working again, I wanted to reinstall my system from scratch onto a new SD card so I guess I finally got myself up to do it.

milkywade, 2025-05-25:
wait, USB-UART is those chips you need to program an ESP32 ? I might have one. Where do I find the pinout for the opi?

MichaIng, 2025-05-25:
> wait, USB-UART is those chips you need to program an ESP32 ?

High likely yes, basically everything that has no video output. The Orange Pi Zero 3 has a dedicated 3-pin header for the debug UART. Ah, here it is shown well: http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_Zero_3#How_to_use_the_debugging_serial_port
Only GND, RX and TX need to be connected

ntn888, 2025-05-27:
> Sorry for so late respong Michal. I am using Armbiang bookworm stable. I havent noticed any issue with ram size.

Hi I tried running with below Armbian version. But still had issue (on 6th boot):

`v25.8 rolling for Orange Pi Zero3 running Armbian Linux 6.12.23-current-sunxi64`.

I'm not sure if this is an older distro version?

MichaIng, 2025-05-27:
@ntn888 
This is still the old one. The new one would be U-Boot 2025.04, as shown in my post above. Did you flash the new version via
```sh
sudo /boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
```
?

ntn888, 2025-05-27:
This one right? `https://dietpi.com/downloads/binaries/linux-u-boot-orangepizero3-current.deb`? I will try it

MichaIng, 2025-05-27:
In the meantime you can just
```sh
apt update
apt upgrade
```

ntn888, 2025-05-27:
> In the meantime you can just
> 
> apt update
> apt upgrade

okay i will try.. thanks

ntn888, 2025-05-27:
Okay.. I tried 10 reboots, all correct with 1GB for me..
```
U-Boot SPL 2025.04-armbian-2025.04-S3482-Pf089-H8869-V3d5b-Bb703-R448a-dirty (May 25 2025 - 14:46:18 +0000)
DRAM base address is defined as 0x40000000
DRAM has 15 b/raw, 10 b/col, 4 B/width, 1 #rank and 8 #bank
DRAM top address must be less than 0x40000000
DRAM: 1024 MiB
Trying to boot from MMC1
```

BTW: the system does auto update on first boot so I had to do nothing.. but the issue is if it boots with wrong value, it'll corrupt the image (which happened to me earlier).. so if I dont check the uart and just login it might break.. hope you're aware..

MichaIng, 2025-05-27:
> BTW: the system does auto update on first boot so I had to do nothing..

That is good, I already wanted to ask about it, because you said

> Also trying out the new DietPI version 9.13, still incorrectly detected..

so I was afraid that for some reason the U-Boot package did not get upgraded or the binary not flashed.

Great, one good sign. More tests of other users will show whether it is really bullet-prove now, or whether it has just lowered the chance for the failure without fixing it entirely.

MichaIng, 2025-05-29:
Yeah exactly. Same for Orange Pi Zero 3. Btw what is the `BOOTDIR` variable used for? I removed it as well, but was wondering: https://github.com/MichaIng/build/commit/21cadcc

MichaIng, 2025-05-29:
I just checked in Armbian code. It seems to be only relevant at compile time, the directory to which the U-Boot sources are cloned to. So if you have a build system used to generate images or assets for different boards, defining this differently at the board config means that you do not override the default U-Boot source directory for this board family. So you trade `git clone` time for disk space. Irrelevant for us, since we build everything on a fresh GitHub runner instance anyway.