MichaIng/DietPi #7243: Failed to Update from v9.3.0 to v.9.7.1.

oystermon, 2024-10-14:
#### Details:
- Date           | Mon Oct 14 19:05:22 BST 2024
- DietPi version | v9.3.0 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux Hamachi 5.10.0-28-amd64 #1 SMP Debian 5.10.209-2 (2024-01-31) x86_64 GNU/L             inux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. I ran dietpi-update as usual
#### Expected behaviour:

- dietpi should update
#### Actual behaviour:

- Fails to update
#### Extra details:

- Apologies I haven't been updating regularly as v9.3.0 has been rock solid..
#### Additional logs:
```
Get:1 http://download.opensuse.org/repositories/shells:/fish:/release:/3/Debian_11  InRelease [1556              B]
Hit:2 https://download.docker.com/linux/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye InRelease
Hit:4 https://deb.debian.org/debian bullseye-updates InRelease
Hit:5 https://dietpi.com/apt bullseye InRelease
Hit:6 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:7 https://deb.debian.org/debian bullseye-backports InRelease
Err:1 http://download.opensuse.org/repositories/shells:/fish:/release:/3/Debian_11  InRelease
  The following signatures were invalid: EXPKEYSIG 2CE2AC08D880C8E4 shells:fish OBS Project <shells:             fish@build.opensuse.org>
Fetched 1556 B in 2s (774 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previo             us index files will be used. GPG error: http://download.opensuse.org/repositories/shells:/fish:/rele             ase:/3/Debian_11  InRelease: The following signatures were invalid: EXPKEYSIG 2CE2AC08D880C8E4 shell             s:fish OBS Project <shells:fish@build.opensuse.org>
E: Failed to fetch http://download.opensuse.org/repositories/shells:/fish:/release:/3/Debian_11/InRe             lease  The following signatures were invalid: EXPKEYSIG 2CE2AC08D880C8E4 shells:fish OBS Project <sh             ells:fish@build.opensuse.org>
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2024-10-15:
You seems to have added a 3rd party repository where the key would need to be updated.

MichaIng, 2024-10-15:
Why a 3rd party repo for fish at all? Debian has it natively.

However, here is the key, depending on where you stored it:
```sh
wget 'https://download.opensuse.org/repositories/shells:/fish:/release:/3/Debian_11/Release.gpg' -O /etc/apt/trusted.gpg.d/opensuse.gpg
```

oystermon, 2024-10-15:
Thanks for the prompt reply...

Sorry not very familiar with this, but no idea where I got fish from? Is there a command to trace where it comes from?

MichaIng, 2024-10-15:
There is not really a way to know how a file was created, other than modification date as a hint 🤔. Yeah, in that case your can just remove the related file from `/apt/sources.list.d` and in case the related (expired) key from `/etc/apt/trusted.gpg.d`, if it can be identified without doubt.

But to be clear, those OpenSUSE repositories is trustable, so all good with that fish version. Probably it is newer than what Debian offers, especially the older Debian Bullseye.

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the problem persists.

closed 2025-05-31 by MichaIng