MichaIng/DietPi #7245: Please add support for NanoPi M6

r3jack, 2024-10-17:
### Creating an image request
NanoPi M6 4G~32G(with/without LCD)

#### Formal device information
- Device name | NanoPi M6
- Official product URL | https://wiki.friendlyelec.com/wiki/index.php/NanoPi_M6
- GitHub resource URL (if available) | https://github.com/friendlyarm
- Image download URL | https://drive.google.com/drive/folders/1FoBbP_nPkMehwBj4wHwsbRU-QGjEdeEP

#### Is the SBC officially supported by the Debian installer?
- Not sure.

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Product wiki provide debian bullseye image.

r3jack, 2024-10-19:
> First step done: [a4d273d](https://github.com/MichaIng/DietPi/commit/a4d273d)

NanoPi M6 32GB without LCD works great, thx.

StephanStS, 2024-12-23:
Doc change: See https://github.com/MichaIng/DietPi-Docs/pull/1040

closed 2024-12-25 by MichaIng