MichaIng/DietPi #7248: Chromium low memory warning

entozoon, 2024-10-18:
Running chromium on say, a raspberry pi 3 model A in kiosk mode kicks out a warning dialog that you can't prevent.

[image]

I found out it, [with some help](https://issues.chromium.org/issues/373891800#comment5), it's the `/usr/bin/chromium-browser` script that does it.

Does this script come from DietPi? Or is it from the [RPi-Distro debian script](https://github.com/RPi-Distro/chromium/blob/6b522e37f4bcb41fa8aafc5cf9e8ed913d324395/debian/scripts/chromium#L59)? I can see it's the same.

Either way I'd recommend gouging out this warning because, well, not only does it run reasonably well depending on your purposes, but it shouldn't have interstitial dialogs when booting into kiosk mode.

Joulinar, 2024-10-18:
we don't build Chromium ourselves. This is the default from official Debian/RPI source

entozoon, 2024-10-18:
Okay cool, I'll move my issue over there. Thank you

Django1982, 2024-10-20:
You can supress this message, when you set "want_memcheck=0" in the mentioned file from entozoon.
In my case this setting is on line 122.

entozoon, 2024-10-22:
Yeah or a `--no-memcheck` flag as mentioned in the [RPi-Distro issue](https://github.com/RPi-Distro/chromium/issues/51#issuecomment-2422232700) (that was also closed, thanks to the tip

MichaIng, 2024-10-26:
It is an RPi-only warning. I do not find out reasonably at all, hence we'll disable it by default on DietPi: https://github.com/RPi-Distro/chromium/commit/7f411d6ad53c1e8be8f6b864341ec27772826442

MichaIng, 2026-11-07:
Done with https://github.com/MichaIng/DietPi/commit/2d07fff for next release.

closed 2026-11-07 by MichaIng