MichaIng/DietPi #7250: Home Assistant: FFmpeg fails [Errno 2] No such file or directory: 'ffmpeg'

Glareascum, 2024-10-19:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm 0

- Kernel version | `uname -a`
Linux neowise 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)

#### Additional Information (if applicable)
- Software title | Home Assistant
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. Launch dietpi-software and install home assistant as provided from docs

#### Expected behaviour
No errors

#### Actual behaviour
Web interface is not reachable, journalctl shows me error about `ERROR (MainThread) [haffmpeg.core] FFmpeg fails [Errno 2] No such file or directory: 'ffmpeg'`

#### Extra details
Journalctl logs (in reverse)
```
Oct 19 02:04:07 neowise Home Assistant[16966]: 2024-10-19 02:04:07.211 ERROR (SyncWorker_2) [aiodhcpwatcher] Cannot watch for dhcp packets without a functional packet filter: libpcap is not available. Cannot >
Oct 19 02:04:04 neowise Home Assistant[16966]: 2024-10-19 02:04:04.333 WARNING (MainThread) [haffmpeg.tools] Error starting FFmpeg.
Oct 19 02:04:04 neowise Home Assistant[16966]: FileNotFoundError: [Errno 2] No such file or directory: 'ffmpeg'
Oct 19 02:04:04 neowise Home Assistant[16966]:     raise child_exception_type(errno_num, err_msg, err_filename)
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/subprocess.py", line 1955, in _execute_child
Oct 19 02:04:04 neowise Home Assistant[16966]:     self._execute_child(args, executable, preexec_fn, close_fds,
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/subprocess.py", line 1026, in __init__
Oct 19 02:04:04 neowise Home Assistant[16966]:                  ^^^^^^^^^^^^^^^^^
Oct 19 02:04:04 neowise Home Assistant[16966]:     self._proc = subprocess.Popen(
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/unix_events.py", line 820, in _start
Oct 19 02:04:04 neowise Home Assistant[16966]:     self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/base_subprocess.py", line 36, in __init__
Oct 19 02:04:04 neowise Home Assistant[16966]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Oct 19 02:04:04 neowise Home Assistant[16966]:     transp = _UnixSubprocessTransport(self, protocol, args, shell,
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/unix_events.py", line 211, in _make_subprocess_transport
Oct 19 02:04:04 neowise Home Assistant[16966]:                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Oct 19 02:04:04 neowise Home Assistant[16966]:     transport = await self._make_subprocess_transport(
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/base_events.py", line 1743, in subprocess_exec
Oct 19 02:04:04 neowise Home Assistant[16966]:                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Oct 19 02:04:04 neowise Home Assistant[16966]:     transport, protocol = await loop.subprocess_exec(
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/subprocess.py", line 224, in create_subprocess_exec
Oct 19 02:04:04 neowise Home Assistant[16966]:                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Oct 19 02:04:04 neowise Home Assistant[16966]:     self._proc = await asyncio.create_subprocess_exec(
Oct 19 02:04:04 neowise Home Assistant[16966]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/site-packages/haffmpeg/core.py", line 130, in open
Oct 19 02:04:04 neowise Home Assistant[16966]: Traceback (most recent call last):
Oct 19 02:04:04 neowise Home Assistant[16966]: 2024-10-19 02:04:04.318 ERROR (MainThread) [haffmpeg.core] FFmpeg fails [Errno 2] No such file or directory: 'ffmpeg'
Oct 19 02:04:03 neowise Home Assistant[16966]: 2024-10-19 02:04:03.095 WARNING (MainThread) [aiohttp_fast_zlib] zlib_ng and isal are not available, falling back to zlib, performance will be degraded.
Oct 19 02:04:01 neowise Home Assistant[16966]: 2024-10-19 02:04:01.813 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This compon>
Oct 19 02:03:54 neowise systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
```

Joulinar, 2024-10-19:
I guess `ffmpeg` would need to be installed

MichaIng, 2024-10-19:
I saw errors like this as well, but HA worked well. They added a lot of dependencies, unnecessary for the majority of users 🤔. The test installations via GitHub workflow from few days ago did start up and HA was listening on it's port at least. But will do one in a local VM to test.

Glareascum, 2024-10-19:
Yep, that's weird. Anyway, `sudo apt ffmpeg` solved the issue and all It's working fine now.

closed 2024-10-19 by Glareascum