MichaIng/DietPi #7251: Radarr update fails the second time

Arghh, 2024-10-19:
After updating Radarr over the Web-UI to latest version (from [5.11.0.9244](https://github.com/Radarr/Radarr/releases/tag/v5.11.0.9244) to [5.12.2.9335](https://github.com/Radarr/Radarr/releases/tag/v5.12.2.9335) Radarr stopped working.

After reinstalling with:
` dietpi-software reinstall 145`  the problem still exist.

I uninstalled radarr over dietpi-software menu and installed it again over the same menu. Now it works. 

Reporting this issue because this seems to happen now every time radarr has an update.

Service error:
`[FAILED] DietPi-Services | × radarr.service - Radarr (DietPi)
     Loaded: loaded (/etc/systemd/system/radarr.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-10-19 14:03:56 CEST; 6min ago
   Duration: 182ms
    Process: 28742 ExecStart=/opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr (code=exited, status=154)
   Main PID: 28742 (code=exited, status=154)
        CPU: 538ms`

In journal I see:
`Oct 19 14:03:56 DietPi4 systemd[1]: Failed to start radarr.service - Radarr (DietPi).
Oct 19 14:11:47 DietPi4 systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 19 14:11:47 DietPi4 Radarr[30499]: The application to execute does not exist: '/opt/radarr/Radarr.dll'.
Oct 19 14:11:47 DietPi4 systemd[1]: radarr.service: Main process exited, code=exited, status=154/n/a`

Update log:
`Oct 19 14:03:49 DietPi4 Radarr[9168]: [Error] CommandExecutor: Error occurred while executing task ApplicationUpdate
Oct 19 14:03:49 DietPi4 Radarr[9168]: [v5.11.0.9244] System.IO.IOException: Directory not empty : '/tmp/radarr_update/'
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at System.IO.FileSystem.RemoveDirectoryInternal(DirectoryInfo directory, Boolean recursive, Boolean throwOnTopLevelDirectoryNotFound)
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at System.IO.FileSystem.RemoveDirectory(String fullPath, Boolean recursive)
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at System.IO.Directory.Delete(String path, Boolean recursive)
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at NzbDrone.Common.Disk.DiskProviderBase.DeleteFolder(String path, Boolean recursive) in ./Radarr.Common/Disk/DiskProviderBase.cs:line 304
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at NzbDrone.Core.Update.InstallUpdateService.InstallUpdate(UpdatePackage updatePackage) in ./Radarr.Core/Update/InstallUpdateService.cs:line 121
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at NzbDrone.Core.Update.InstallUpdateService.Execute(ApplicationUpdateCommand message) in ./Radarr.Core/Update/InstallUpdateService.cs:line 289
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at NzbDrone.Core.Messaging.Commands.CommandExecutor.ExecuteCommand[TCommand](TCommand command, CommandModel commandModel) in ./Radarr.Core/Messaging/Commands/CommandEx>
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at CallSite.Target(Closure , CallSite , CommandExecutor , Object , CommandModel )
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid3[T0,T1,T2](CallSite site, T0 arg0, T1 arg1, T2 arg2)
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at CallSite.Target(Closure , CallSite , CommandExecutor , Object , CommandModel )
Oct 19 14:03:49 DietPi4 Radarr[9168]:    at NzbDrone.Core.Messaging.Commands.CommandExecutor.ExecuteCommands() in ./Radarr.Core/Messaging/Commands/CommandExecutor.cs:line 44
Oct 19 14:03:53 DietPi4 systemd[1]: radarr.service: Main process exited, code=killed, status=9/KILL
Oct 19 14:03:53 DietPi4 systemd[1]: radarr.service: Failed with result 'signal'.
Oct 19 14:03:53 DietPi4 systemd[1]: radarr.service: Unit process 28684 (Radarr.Update) remains running after unit stopped.
Oct 19 14:03:53 DietPi4 systemd[1]: radarr.service: Consumed 1h 1min 48.364s CPU time.
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Scheduled restart job, restart counter is at 1.
Oct 19 14:03:54 DietPi4 systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Consumed 1h 1min 48.595s CPU time.
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Found left-over process 28684 (Radarr.Update) in control group while starting unit. Ignoring.
Oct 19 14:03:54 DietPi4 systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Oct 19 14:03:54 DietPi4 systemd[1]: Started radarr.service - Radarr (DietPi).
Oct 19 14:03:54 DietPi4 Radarr[28738]: The application to execute does not exist: '/opt/radarr/Radarr.dll'.
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Main process exited, code=exited, status=154/n/a
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Failed with result 'exit-code'.
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Unit process 28684 (Radarr.Update) remains running after unit stopped.
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Scheduled restart job, restart counter is at 2.
Oct 19 14:03:54 DietPi4 systemd[1]: Stopped radarr.service - Radarr (DietPi).
Oct 19 14:03:54 DietPi4 systemd[1]: radarr.service: Found left-over process 28684 (Radarr.Update) in control group while starting unit. Ignoring.
Oct 19 14:03:54 DietPi4 systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Oct 19 14:03:54 DietPi4 systemd[1]: Started radarr.service - Radarr (DietPi).`

#### Required Information
- DietPi version | 9.7.1
- Distro version | bookworm
- Kernel version | Linux DietPi4 6.1.21-v8+ 

#### Additional Information (if applicable)
- Software title | Radarr
- updated over Browser UI
- Can this issue be replicated on a fresh installation of DietPi? - happens while updating. So no.
- 
#### Steps to reproduce
Update radarr over the UI.

#### Expected behaviour
Radarr should update without problems.

MichaIng, 2024-10-26:
Thanks for your report. There was a similar case sone week ago. The update seems to abort in the middle. Maybe this can happen if memory or `/tmp` is full. I'll they to replicate.

Arghh, 2024-10-26:
Thank you. I also think I could start the Update twice over the UI. Start it and than after about 30 seconds start it again. Maybe that is something. But my main problem is the broken state this leaves the software. Reinstall should fix this or not?

MichaIng, 2024-10-26:
Yeah, we implemented the reinstall conservatively, not touching an existing install, but only install dependencies, service file and stuff around, printing a hint to use the web UI updater instead. That message also gives that hint to remove `/opt/radarr` manually for a repair. But that conservative approach causes more issues than it solves, as the web UI updater does not thing else when what we would do, all migration steps are done afterwards on first start with new executables. And the message is easily overseen by the wall of other messages. I'll adjust this for the Servarr install options. Our docs point to web UI updater already, so it makes sense to have a reinstall doing an actual reinstall, repairing broken installs as well OOTB.

kmau, 2024-11-27:
Hi, today also appened to me as well (with the same _Directory not empty : '/tmp/radarr_update/'_ message in Radarr logs etc). I'm also secure that I've launced twice the update from web UI:

1. I've clicked on the Update/Install (I don't remember the exact name)
2. In the left bottom of the screen I've see the download/backup message
3. Before the finalization of the update, I've clicked again on the Update/Install button (yes, I'm _dumb_, but I didn't remember if the installation started automatically or if it was a first download step and then a second one to confirm the installation). 

Now with a new install from the tarball I solved it, but I wanted to share what happened.

MichaIng, 2024-12-01:
I changed `dietpi-software` now a way that `dietpi-software reinstall` always updates/repairs the instance. This skipping it, pointing to the internal updater, causes more issues than that it helps, as the reinstall is exactly required, when the internal updater fucks things up: https://github.com/MichaIng/DietPi/commit/23a5345

@kmau 
No problem. Let us know, probably link your issue on the Radarr repo here. All Servarr apps are affected by this, though I can never replicate it, so it seems to be happening in rare circumstances as of some sort of race condition or so. So it is interesting for us/our users as well, whether Radarr/Servarr devs find the underlying reason.