MichaIng/DietPi #7252: DietPi-Software: installation of Node-RED (122) fails

mvanbrab, 2024-10-19:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm 1
- Kernel version | `uname -a`
Linux dietpi1 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi Zero W (armv6l)
- Power supply used | (EG: 5V 1A RAVpower)
5V 2.5A STONTRONICS
- SD card used | (EG: SanDisk ultra)
Transcend Premium 400x 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Node-RED
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`
693b1d7d-ae95-4e05-b5cc-5966cbd2efb2

#### Steps to reproduce

1. as root user, start dietpi-software
2. search for software "Node-RED" (122) and select it for installation.
3. install

#### Expected behaviour

- Installation should work

#### Actual behaviour

- This error message appears:
```text
[ INFO ] DietPi-Software | runuser -u nodered -- npm i --cache /tmp/tmp.TJcTLucnGs --no-audit node-red@latest, please wait...
npm notice
npm notice New minor version of npm available! 10.5.0 -> 10.9.0
npm notice Changelog: <https://github.com/npm/cli/releases/tag/v10.9.0>
npm notice Run `npm install -g npm@10.9.0` to update!
npm notice
npm ERR! code ECONNRESET
npm ERR! network Client network socket disconnected before secure TLS connection was established
npm ERR! network This is a problem related to network connectivity.
npm ERR! network In most cases you are behind a proxy or have bad network settings.
npm ERR! network
npm ERR! network If you are behind a proxy, please make sure that the
npm ERR! network 'proxy' config is set properly.  See: 'npm help config'
```

#### Extra details

- I retried several times last week, from a fresh image and from an existing image.
- I'm not behind a proxy
- Network connectivity is OK (tested with dietpi-config)

MichaIng, 2024-10-26:
Hmm the error indicates a network issue, but a connection check is fine before the install. Maybe npmjs servers have an issue 🤔. I'lly to replicate.

mvanbrab, 2024-11-06:
Hi @MichaIng ,

I tried one more time from a fresh image today and I could at least finalize the installation of Node-RED this time, so there may have been some intermittent connectivity issue during all my previous trials, or some changes may have happened in the online resources meanwhile.

**However, the installation result was not as expected: the service fails to start.**

I tried to diagnose:

```bash
root@dietpi1:~# systemctl status node-red
× node-red.service - Node-RED (DietPi)
     Loaded: loaded (/etc/systemd/system/node-red.service; enabled; preset: enabled)
     Active: failed (Result: signal) since Wed 2024-11-06 18:00:44 CET; 2min 34s ago
   Duration: 5.203s
    Process: 3629 ExecStart=/mnt/dietpi_userdata/node-red/node_modules/.bin/node-red -u /mnt/dietpi_userdata/node-red (code=killed, signal=ILL)
   Main PID: 3629 (code=killed, signal=ILL)
        CPU: 4.466s

Nov 06 18:00:39 dietpi1 systemd[1]: Started node-red.service - Node-RED (DietPi).
Nov 06 18:00:44 dietpi1 systemd[1]: node-red.service: Main process exited, code=killed, status=4/ILL
Nov 06 18:00:44 dietpi1 systemd[1]: node-red.service: Failed with result 'signal'.
Nov 06 18:00:44 dietpi1 systemd[1]: node-red.service: Consumed 4.466s CPU time.
root@dietpi1:~#
```

I tried to further investigate by executing the ExecStart command found in `/etc/systemd/system/node-red.service`. It resulted in a single line error message:

```bash
root@dietpi1:~#/mnt/dietpi_userdata/node-red/node_modules/.bin/node-red -u /mnt/dietpi_userdata/node-red
Illegal instruction
```

For what it's worth, I found a [post on the Node-RED forum](https://discourse.nodered.org/t/cant-start-node-red-4-after-re-install-on-pi-zero-w/89322), describing exactly the same behaviour, also on a **RPi Zero W**, but outside DietPi.

- A conclusion could be [this entry](https://discourse.nodered.org/t/cant-start-node-red-4-after-re-install-on-pi-zero-w/89322/11) in that post.
- There may be a solution as [this entry](https://discourse.nodered.org/t/cant-start-node-red-4-after-re-install-on-pi-zero-w/89322/20) in that post says.

Maybe that solution needs to be integrated in DietPi?

MichaIng, 2024-11-07:
Our Node.js installer https://github.com/MichaIng/nodejs-linux-installer installs the unofficial Node.js builds on ARMv6 already. Problem is the bcrypt module, it seems.

I'll run some tests on my Zero W.

MichaIng, 2024-11-07:
Node-RED v4 adds a new dependency `@node-rs/bcrypt`, which does not support ARMv6:
- https://www.npmjs.com/package/node-red?activeTab=dependencies
- https://www.npmjs.com/package/@node-rs/bcrypt

We hence need to install Node-RED v3 on ARMv6: https://github.com/MichaIng/DietPi/commit/2187d65

To do that manually:
```sh
systemctl stop node-red
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm i --no-audit node-red@3
systemctl start node-red
```

mvanbrab, 2024-11-08:
The manual workaround works perfectly on my RPi Zero W!
Thank you very much @MichaIng !

closed 2024-12-25 by MichaIng