MichaIng/DietPi #7253: Raspberry Pi 5 Wifi driver / Access Point

oxoocoffee, 2024-10-19:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm 0 `
- Kernel version | `Linux dietpi-ai 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux`
- SBC model | `RPi 5 Model B (aarch64)`
- Power supply used | (5V 27W adapter)
- SD card used | (SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (ieee80211 driver)
- Was the software title installed freshly or updated/migrated? 3 months ago
- Can this issue be replicated on a fresh installation of DietPi? Not sure

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ... Enable Wifi in diet-configure
2. ... Install and configure Access Point using hostapd and dnsmasq

#### Expected behaviour

- ...

#### Actual behaviour

-  After booting I see messages popping outing console. (run level 3) with out login in or after.

#### Extra details

- I was not able to 
[image]

MichaIng, 2024-10-26:
Thanks for reporting. I'll try to replicate.

How did you configure `hostapd`. Since you installed `dnsmasq`, I guess you did not install the WiFi Hotspot option via `dietpi-software`, right?

oxoocoffee, 2024-10-27:
I did installed it using dietpi-software, I I remembered something failed during installation. I do not remember what since I had to do something else soon after. When I did comeback to it I fixed it manually. What specific information you looking for?

oxoocoffee, 2024-11-15:
Any updates of this issue? Any additional information I can provide?

Joulinar, 2024-11-16:
I have performed a test installation on my demo RPi5 without any issues. During the installation there is an error message about the failed start of the `isc-dhcp-server` service.

```
[FAILED] DietPi-Software | systemctl start isc-dhcp-server
```

But this is to be expected and `isc-dhcp-server` works after the reboot. There are no kernel error messages in `dmesg` and my mobile phone connects to the DietPi hotspot without any problems.

MichaIng, 2024-11-21:
> When I did comeback to it I fixed it manually. What specific information you looking for?

How did you fix it? Which exact settings did you change, in case?

oxoocoffee, 2024-11-22:
I will try to setup a AP on clean installation in about 2-3 days when I have access back to rpi5. I will capture all steps I took and see if I can reproduce it again. If I stumble into an error, I will post the details here.

oxoocoffee, 2024-11-25:
While I was first trying to find a working WiFi dongle to get `wlan0` and not needing enabling internatl WiFI, which is in metal case, I am using the using in this issue #7290. 

[Edimax Wi-Fi 4 802.11n Adapter for PC N150 Nano USB Adapter](https://www.amazon.com/dp/B08F2ZNC6J)

- I created a fresh image using latest v9.8.0 image for rPI 4.
- No pre initial boot config file modification
- During first boot, took all defaults and nothing was changed except host name in `/etc/hostname` and `/etc/hosts`
- After initial setup was completed as as root use, I created a new user test account `runner`
- Added `runner` to following groups: `dialout, sudo, users, tcpdump, wireshark` to `/etc/groups`, then rebooted
- Logged in to `runner` account and checked for `wlan0` using above USB dongle

   ```bash
   $ ifconfig -a
   wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
   ```

- Next is `sudo dietpi-config`
- Option 7 and 8 left unmodified.
   
   ```
    Ethernet     : Available | [On] | Connected
    WiFi         : Available | [Off] | Disconnected
    Onboard WiFi : [Off]
    ●─ Additional Options ───────────●
    IPv6         : [On]
    Proxy        : [Off]
   ```
- Next is `sudo dietpi-software` and selected `Browse Software` option. Be default no software option was enabled/selected. Booting to `run level 3`.

- Next, I selected `vim(20), tcpdump(15), WiFi Hotspot (60)` and pressed `Confirm`
   ```
   [  OK  ] DietPi-Software | Initialised database
   [  OK  ] DietPi-Software | Reading database
   [ SUB1 ] DietPi-Set_hardware > wifimodules (enable)
   [  OK  ] DietPi-Set_hardware | rm /etc/modprobe.d/dietpi-disable_wifi.conf
   [ INFO ] DietPi-Set_hardware | Please wait, enabling WiFi modules...
   [  OK  ] DietPi-Set_hardware | modprobe cfg80211
   [  OK  ] DietPi-Set_hardware | modprobe brcmfmac
   [  OK  ] DietPi-Set_hardware | modprobe brcmutil
   [ INFO ] DietPi-Set_hardware | Checking for required APT packages: iw wireless-tools wpasupplicant wireless-regdb
   [  OK  ] wifimodules enable | Completed
   ```

Since I am not intending to enable internal WiFi, are the above `modprobe` calls are needed?

- Before selecting `Install` to start installation, I opened another ssh session from another window just to check if I see another `wlan1` for the `mod probe` and check is the internal wifi was enabled just by selecting `WiFi Hotspot` and silently this was enabled. Running `ifconfig -a` still shows only one `wlan0` from external USB dongle. And I still see

   ```
    Ethernet     : Available | [On] | Connected
    WiFi         : Available | [Off] | Disconnected
    Onboard WiFi : [Off]
    ●─ Additional Options ───────────●
    IPv6         : [On]
    Proxy        : [Off]
   ```

- Select `<Install>` and then select `<Ok>` to start installation. Please see the attached log file for the errors. These are the same errors I got on this first attempt when I created this issue but using internal WiFi only. So it looks like it is the same issue regardless is using internet Wifi or external Wifi dongle.

[dietpi-coftware.log](https://github.com/user-attachments/files/17895806/dietpi-coftware.log)

Here is also output for `sudo journalctl -xeu isc-dhcp-server.service`

[isc-dhcp-server.service.log](https://github.com/user-attachments/files/17895836/isc-dhcp-server.service.log)

Also did checked most recent `dmsg` entries and I see following
[... 47 lines cut ...]
[Mon Nov 25 00:48:50 2024] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[Mon Nov 25 00:48:50 2024] task:RTW_CMD_THREAD  state:D stack:0     pid:4175  ppid:2      flags:0x00000008
[Mon Nov 25 00:48:50 2024] Call trace:
[Mon Nov 25 00:48:50 2024]  __switch_to+0xf8/0x1e0
[Mon Nov 25 00:48:50 2024]  __schedule+0x2a8/0x830
[Mon Nov 25 00:48:50 2024]  schedule+0x60/0x100
[Mon Nov 25 00:48:50 2024]  schedule_timeout+0x2cc/0x358
[Mon Nov 25 00:48:50 2024]  wait_for_completion+0x84/0x168
[Mon Nov 25 00:48:50 2024]  rtw_cmd_thread+0x68/0x1b0 [r8188eu]
[Mon Nov 25 00:48:50 2024]  kthread+0xfc/0x110
[Mon Nov 25 00:48:50 2024]  ret_from_fork+0x10/0x20
[Mon Nov 25 00:50:51 2024] INFO: task RTW_CMD_THREAD:4175 blocked for more than 604 seconds.
[Mon Nov 25 00:50:51 2024]       Tainted: G         C         6.1.21-v8+ #1642
[Mon Nov 25 00:50:51 2024] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[Mon Nov 25 00:50:51 2024] task:RTW_CMD_THREAD  state:D stack:0     pid:4175  ppid:2      flags:0x00000008
[Mon Nov 25 00:50:51 2024] Call trace:
[Mon Nov 25 00:50:51 2024]  __switch_to+0xf8/0x1e0
[Mon Nov 25 00:50:51 2024]  __schedule+0x2a8/0x830
[Mon Nov 25 00:50:51 2024]  schedule+0x60/0x100
[Mon Nov 25 00:50:51 2024]  schedule_timeout+0x2cc/0x358
[Mon Nov 25 00:50:51 2024]  wait_for_completion+0x84/0x168
[Mon Nov 25 00:50:51 2024]  rtw_cmd_thread+0x68/0x1b0 [r8188eu]
[Mon Nov 25 00:50:51 2024]  kthread+0xfc/0x110
[Mon Nov 25 00:50:51 2024]  ret_from_fork+0x10/0x20
```

oxoocoffee, 2024-11-25:
I forgot to add current screen messages. All the commands in previous posts were performed over ssh sessions using default starts on the boot `/usr/sbin/dropbear`

[image]

MichaIng, 2024-12-01:
@oxoocoffee 
That the DHCP server service start fails on first install is expected, since no valid config exists yet, and the interface may be down. Don't ask me why the APT maintainer scripts do not this expected case properly. We would need to install the `isc-dhcp-server` package after everything else (config generation, configuring WiFi interface), to avoid it 🤔.

However, this does not cause issues. The kernel errors are however really annoying. Those are "warnings", right?
```sh
dmesg -l 4
```
I guess it is an issue with the old Realtek driver. Please try whether the new kernel solves it: https://github.com/MichaIng/DietPi/issues/6676

And you are right, the Broadcom modules do not need to be loaded in your particular case. However, they were blacklisted before, and we cannot know which driver is needed, so we need to load all blacklisted ones to assure every WiFi adapter is functional.

oxoocoffee, 2024-12-07:
I got it fixed. Yup, there were no default configuration provided. After I did set it up it is up.
So this issue can be closed.

But I have a some additional question. Looks like rPI4 Wifi supports AC?
I should be able to setup `hostapd` in `n` mode. 

```ini
interface=wlan0
driver=nl80211
ssid=*******
wpa_passphrase=*********
country_code=US
hw_mode=g
channel=1
ieee80211n=1
ieee80211ac=0
ieee80211ax=0
wmm_enabled=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
logger_syslog=1
logger_syslog_level=2
```

If I change `hw_mode=g` to `hw_mode=n` it failed to start. Am I missing something?

MichaIng, 2024-12-12:
@oxoocoffee 
`hw_mode=n` does not exist. It is a little confusing, but `hw_mode=g` is 2.4 GHz while otherwise only `hw_mode=a` exists for 5 GHz. `ieee80211n=1` toggles WiFi 4/n.

You can toggle all these (WiFi 4/5/6, 2.5 and 5 GHz) in `dietpi-config` adapter > WiFi settings btw, and it will/should set all flags in the `hostapd` config correctly.

oxoocoffee, 2025-03-14:
@MichaIng 

Thank you for your info. I got sidetracked to a different project. Today I am back working on this item. I will have more information a bit later to report with some suggestions after I will try it. I am trying few other USB Adapters. Three of them are not recognized out of the box, so I will try to look at them a bit later. Before I go to my main question below, I like to ask you your opinion about the item below.

---

All  cases below are running on upgraded DietPi

```plaintext
DietPi v9.11.2
Device model : RPi 4 Model B (aarch64)
Linux rpi-cam-ap 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
```

```bash
$ rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Tue Feb 11 05:00:13 PM UTC 2025 (1739293213)
    LATEST: Tue Feb 11 05:00:13 PM UTC 2025 (1739293213)
```

Internal WiFi is disabled and I have this **USB WiFi Adapter** made my `Alpha Network` model `AC600`. The output from `lsusb` is below.

```bash
"Bus 001 Device 005: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]"
```

Looks like kernel module is not included in the main tree. I found this repo [8812au-20210820](https://github.com/morrownr/8812au-20210820) which seems promising, and I will take a crack at maybe tomorrow. If I get it working do you have any mechanism to include it in the future releases of DietPi? If anything I could create a deb file and post it here. Any comments?

---

Now, back to my main question for tonight.

I have to other adapters which do work out of the box. But I am observing a strange behavior. I am speculating it might be some `power saving` kicking in?

On my `MBP M4`, which sits on the same desk winning `WiFi Explorer` I see this behavior

[image]

The screenshot above was taken when AP was using adapter below.

1. Looks like the max rate is 72Mbps only!!! and will not go higher
2. I am not sure yet what or how to fix these constant dimples marked with **red arrows** above.
3. When starting hostapd `sudo systemctl start hostapd`, it takes more than a min before it is visible on any other machine WiFi list. So I am also looking at how to shorten this time.

Please note, at this time there are no clients connected to it yet.

Here are some different output for this device

```bash
$ lsusb
Bus 001 Device 006: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
```

```bash
$ lsusb -t
Port 1: Dev 6, If 0, Class=Vendor Specific Class, Driver=rtw_8821cu, 480M
```

```bash
$ lsmod | grep rtw
rtw88_8821cu           12288  0
rtw88_8821c            81920  1 rtw88_8821cu
rtw88_8822bu           12288  0
rtw88_8822b           221184  1 rtw88_8822bu
rtw88_usb              16384  2 rtw88_8822bu,rtw88_8821cu
rtw88_core            176128  3 rtw88_8821c,rtw88_usb,rtw88_8822b
mac80211             1032192  2 rtw88_core,rtw88_usb
cfg80211              983040  2 rtw88_core,mac80211
```

Looks like `power save` is on

```bash
$ iw wlan0 get power_save
Power save: on
```

---

Now, if I swap a mother adapter, with all the same AP settings / configurations, the speed is much better at 40MHz, but there are the same `dips` marked with `red arrows` above.

[image]

Here is `lsusb` output for this adapter

```bash
$ lsusb
Bus 001 Device 007: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
```

```bash
$ lsusb -t
Port 1: Dev 7, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 480M
```

---

Here is my `/etc/hostapd/hostapd.conf` used with both above adapters.

```ini
# Supports both 802.11g and 802.11n (hw_mode=g, ieee80211n=1)
# Maximizes security (wpa=2, rsn_pairwise=CCMP, no TKIP)
# Optimized for performance (ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40])
# Compatible with both old and new devices (require_ht=0 is not needed unless forcing n-only mode)
# Keeps SSID visible (ignore_broadcast_ssid=0)

# WiFi Interface
interface=wlan0

# Driver for the WiFi chipset
driver=nl80211

# WiFi Network Name (SSID)
ssid=*******

# Country Code for Regulatory Compliance
country_code=US

# Set WiFi Mode (g for 2.4GHz)
hw_mode=g

# Set WiFi Channel (1-11 in US)
channel=1

# Enable 802.11n (WiFi 4)
ieee80211n=1

# Disable 5GHz WiFi Standards (AC = WiFi 5, AX = WiFi 6)
ieee80211ac=0
ieee80211ax=0

# Enable WiFi Multimedia (WMM) for QoS
wmm_enabled=1

# MAC Address Filtering (0 = Disabled)
macaddr_acl=0

# Beacon Interval. Default is 100 TU (Time Units), which is 100 ms.
beacon_int=50

# Authentication Algorithms (1 = Open System)
auth_algs=1

# Broadcast SSID (0 = Visible, 1 = Hidden)
ignore_broadcast_ssid=0

# WPA2 Security Settings
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase***********
rsn_pairwise=CCMP

# Enable Short Guard Interval and 40MHz HT Mode for Performance
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40]

# Logging Settings
logger_syslog=1
logger_syslog_level=2

# Enable Hostapd Control Interface for hostapd_cli
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
```

Any comments, suggestions? IPv6 is disabled in `dietpi-config`

Cheers

MichaIng, 2025-04-19:
Does it work and change something when disabling the power savings?
```sh
sudo iw wlan0 set power_save off
```
This should be done automatically on network configuration, but maybe it fails, or it fails as long as the interface is not up already.

In the meantime there is Linux v6.12 available for RPi which might add drivers or enhance support for modern WiFi chips.

And as you say the hotspot takes long to show up for clients, do the logs give a hint?
```sh
journalctl -u hostapd
```

MichaIng, 2025-05-31:
Closing this for now. Feel free to reopen if issue persists with recent kernel.

closed 2025-05-31 by MichaIng

Victor456-001, 2026-08-27:
Can wifi drivers for Raspberry Pi 4 with Windows 10 ARM be on boardcom?

Joulinar, 2026-08-27:
What is the relationship to DietPi on this question. We don't manage or maintain drivers and firmware for RPi devices