MichaIng/DietPi #7254: Update RKNPU Driver 0.9.8 for Orange Pi 5 Plus

My-Name-Is-Mac, 2024-10-21:
### Creating a feature request

Orange Pi 5 Plus

#### Is your feature request related to a problem? Please describe:
- new RKNPU Driver 0.9.8 just released

#### Describe the solution you'd like:
- please merge driver from https://github.com/airockchip/rknn-llm/tree/main/rknpu-driver

MichaIng, 2024-10-26:
Here our (Armbian) sources: https://github.com/armbian/linux-rockchip/tree/rk-6.1-rkr3/drivers/rknpu

It is already newer than what Radxa provides. Hmm, I actually wonder where Rockchip themselves host their original sources. At least here is no Linux 6.1 branch: https://github.com/rockchip-linux/kernel

However, https://github.com/airockchip is an official Rockchip orga as well, so I see no problem to merge this manually via patch, until Armbian merged a an update to their Rockchip Linux repo/branch.

Btw, is there any particular benefit of 0.9.8 over 0.9.7 you need?

My-Name-Is-Mac, 2024-10-27:
Thank you for the information

RKLLM versiom 1.1.0 and 1.1.1 required RKNPU driver 0.9.8.

Thx
Regards

MichaIng, 2024-10-29:
Hmm, at Armbian the same was attempted already, leading to build failure: https://github.com/armbian/linux-rockchip/pull/265