MichaIng/DietPi #7257: update DietPi from v9.4.2 to v9.8.0 fails

klankschap, 2024-10-24:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bookworm

- Kernel version | `uname -a`

Linux npi 6.6.30-current-sunxi64 #1 SMP Thu May  2 14:32:50 UTC 2024 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

NanoPi NEO Plus2 (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

5V 

- SD card used | (EG: SanDisk ultra)

SanDisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)

- Was the software title installed freshly or updated/migrated?
both

- Can this issue be replicated on a fresh installation of DietPi?
 yes

- Bug report ID | `echo $G_HW_UUID`

e8265b8a-e9fa-40c7-8de2-dae518b5667a

#### Steps to reproduce

1.   sudo dietpi-update
  or sudo apt update 

2. ...

#### Expected behaviour

- it should update to v9.8

#### Actual behaviour

- it keeps trying to download th epackages, but ends up failing with connection timeouts.

#### Extra details

- 

$ sudo apt update
Hit:1 https://dietpi.com/apt bookworm InRelease                          
Hit:2 https://dietpi.com/apt all InRelease                               
Get:3 https://deb.debian.org/debian bookworm InRelease [151 kB]
Get:4 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:5 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:6 https://deb.debian.org/debian bookworm-backports InRelease [59.0 kB]
Get:7 https://deb.debian.org/debian bookworm/main arm64 Packages [8689 kB]
Ign:7 https://deb.debian.org/debian bookworm/main arm64 Packages                                                                                                                                   
Get:8 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:8 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:10 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages T-2024-10-24-0205.40-F-2024-10-24-0205.40.pdiff [5372 B]                                                           
Get:10 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages T-2024-10-24-0205.40-F-2024-10-24-0205.40.pdiff [5372 B]                                                           
Get:11 https://deb.debian.org/debian bookworm-backports/main arm64 Packages T-2024-10-24-1404.10-F-2024-10-24-1404.10.pdiff [2534 B]                                                               
Get:11 https://deb.debian.org/debian bookworm-backports/main arm64 Packages T-2024-10-24-1404.10-F-2024-10-24-1404.10.pdiff [2534 B]                                                               
Get:12 https://deb.debian.org/debian bookworm/non-free arm64 Packages [75.8 kB]                                                                                                                    
Get:13 https://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages [5832 B]                                                                                                            
Get:14 https://deb.debian.org/debian bookworm/contrib arm64 Packages [45.7 kB]                                                                                                                     
Get:7 https://deb.debian.org/debian bookworm/main arm64 Packages [8689 kB]                                                                                                                         
Ign:7 https://deb.debian.org/debian bookworm/main arm64 Packages                                                                                                                                   
Get:7 https://deb.debian.org/debian bookworm/main arm64 Packages [8689 kB]                                                                                                                         
Ign:7 https://deb.debian.org/debian bookworm/main arm64 Packages                                                                                                                                   
Get:7 https://deb.debian.org/debian bookworm/main arm64 Packages [8689 kB]                                                                                                                         
Ign:7 https://deb.debian.org/debian bookworm/main arm64 Packages                                                                                                                                   
Get:7 https://deb.debian.org/debian bookworm/main arm64 Packages [8689 kB]
Err:7 https://deb.debian.org/debian bookworm/main arm64 Packages                                                                                                                                   
  Connection timed out [IP: 151.101.38.132 443]
Get:7 https://deb.debian.org/debian bookworm/main arm64 Packages [11.9 MB]                                                                                                                         
Ign:7 https://deb.debian.org/debian bookworm/main arm64 Packages                                                                                                                                   
Get:7 https://deb.debian.org/debian bookworm/main arm64 Packages [11.9 MB]
Err:7 https://deb.debian.org/debian bookworm/main arm64 Packages                                                                                                                                   
  Connection timed out [IP: 2a04:4e42:9::644 443]
Ign:7 https://deb.debian.org/debian bookworm/main arm64 Packages
Err:7 https://deb.debian.org/debian bookworm/main arm64 Packages
  Connection timed out [IP: 2a04:4e42:9::644 443]
Fetched 55.5 kB in 19min 17s (47 B/s)
Reading package lists... Done
N: Repository 'https://deb.debian.org/debian bookworm InRelease' changed its 'Version' value from '12.5' to '12.7'
E: Failed to fetch https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages  Connection timed out [IP: 2a04:4e42:9::644 443]
E: Some index files failed to download. They have been ignored, or old ones used instead.

$ ping -6 2a04:4e42:9::644
PING 2a04:4e42:9::644(2a04:4e42:9::644) 56 data bytes
64 bytes from 2a04:4e42:9::644: icmp_seq=1 ttl=59 time=6.66 ms
64 bytes from 2a04:4e42:9::644: icmp_seq=2 ttl=59 time=4.52 ms
64 bytes from 2a04:4e42:9::644: icmp_seq=3 ttl=59 time=6.66 ms
64 bytes from 2a04:4e42:9::644: icmp_seq=4 ttl=59 time=3.90 ms
64 bytes from 2a04:4e42:9::644: icmp_seq=5 ttl=59 time=2.66 ms
64 bytes from 2a04:4e42:9::644: icmp_seq=6 ttl=59 time=4.00 ms

$ ping -4 51.101.38.132
PING 51.101.38.132 (51.101.38.132) 56(84) bytes of data.
^C
--- 51.101.38.132 ping statistics ---
4 packets transmitted, 0 received, 100% packet loss, time 3064ms

Joulinar, 2024-10-24:
Do you have a special network security device that could close connections for some reason? This looks pretty strange. At first it works for the first 5-6 connections, but shortly after that it times out 🤔

klankschap, 2024-10-24:
nope

MichaIng, 2024-10-26:
Does it work doing just a "Retry" from the error handler menu?

Else, can you enter a subshell and check network details:
```sh
ip a
ip r
```

klankschap, 2024-10-26:
a retry did not make any difference after a dietpi-update fail
neither did  apt update

# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 02:01:e2:89:bf:fa brd ff:ff:ff:ff:ff:ff
    altname end0
    inet 192.168.178.46/24 brd 192.168.178.255 scope global dynamic eth0
       valid_lft 719992sec preferred_lft 719992sec
    inet6 2a10:3781:4b93:1:1:e2ff:fe89:bffa/64 scope global dynamic mngtmpaddr 
       valid_lft 7052sec preferred_lft 3452sec
    inet6 fe80::1:e2ff:fe89:bffa/64 scope link 
       valid_lft forever preferred_lft forever

# ip r
default via 192.168.178.1 dev eth0 
192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.46

klankschap, 2024-10-26:
Would it be a work around to download the files using a different machine, then place them onto the SD card in a /tmp folder, then point apt to that folder?
How would that work in practice?

MichaIng, 2024-10-26:
Does it work to download the list manually?
```sh
cd /tmp
wget https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages
```
If it is the list only, and we name and place it correctly, it could even work. But I fear that downloading the packages results in the same connection timeout.

Seems to be some issue between you and the Debian mirror the CDN is choosing or you. Not sure which one it is, as there is no reverse DNS record given. Does the browser change to a different URL when opening this link, or show the mirror near the bottom of the page? https://deb.debian.org/debian/

However, easiest should be when you switch to a different mirror. Can be done from the error handler. `dietpi-config` > Network Options: Misc > APT mirror.

klankschap, 2024-10-27:
changing to a different mirror (de / uk) did not make a difference.
i can get the Packages.gz using a different machine in the network
 wget https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages.gz

but as the dietpi hasn't made it through its first initial setup, there is no way to get it there via scp, or is there?

MichaIng, 2024-10-29:
The file alone does not help anyway, as the individual packages would likely fail the same way. So manually downloading via `wget` or `curl` does not work either? I mean you can just test to download a package:
```sh
cd /tmp
wget https://deb.debian.org/debian/pool/main/d/dash/dash_0.5.12-9_amd64.deb
```
or a different package from our server:
```sh
wget https://dietpi.com/downloads/binaries/sprd-bluetooth.deb
```
Maybe there is a general network issue. @StephanStS you have a NanoPi NEO Plus2 as well right? Does it still work well for you with APT updates etc with recent kernel?

I'll also update all our images now, maybe the newer kernel helps.

MichaIng, 2024-10-29:
Hmm downloading the packages works, but the text file does still not?
```sh
cd /tmp
wget https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages
```
Weird 🤔.

klankschap, 2024-10-29:
> wget https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages
> --2024-10-29 15:11:15--  https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages
> Resolving deb.debian.org (deb.debian.org)... 2a04:4e42:9::644, 151.101.38.132
> Connecting to deb.debian.org (deb.debian.org)|2a04:4e42:9::644|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2024-10-29 15:11:16 ERROR 404: Not Found.

I guess the Packages file does not exist.
Is there an extension missing?
Packages.gz seems to take forever to download

```
# wget https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages.gz
--2024-10-29 15:14:20--  https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/Packages.gz
Resolving deb.debian.org (deb.debian.org)... 2a04:4e42:9::644, 151.101.38.132
Connecting to deb.debian.org (deb.debian.org)|2a04:4e42:9::644|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 11934473 (11M) [application/x-gzip]
Saving to: ‘Packages.gz’

Packages.gz            0%[                     ]  40.74K  --.-KB/s    eta 6h 54m

```

> Weird 🤔.

Magic ;-)

MichaIng, 2024-10-29:
Right, Debian provides the archived versions of this file only, as this is what APT downloads by default anyway, the `xz` archive, if available, with highest priority.

The output btw is confusing, not showing that fact:
```console
root@NanoPiR5S:/tmp# apt update
Get:1 https://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [59.0 kB]
Get:5 https://dietpi.com/apt bookworm InRelease [3529 B]
Get:6 https://dietpi.com/apt all InRelease [26.2 kB]
Get:7 https://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages [5832 B]
Get:8 https://deb.debian.org/debian bookworm/non-free arm64 Packages [75.8 kB]
Get:9 https://deb.debian.org/debian bookworm/contrib arm64 Packages [45.7 kB]
Get:10 https://deb.debian.org/debian bookworm/main arm64 Packages [8689 kB]
Get:11 https://deb.debian.org/debian bookworm-updates/contrib arm64 Packages [768 B]
Get:12 https://deb.debian.org/debian bookworm-updates/non-free-firmware arm64 Packages [616 B]
Get:13 https://deb.debian.org/debian bookworm-updates/non-free arm64 Packages [12.0 kB]
Get:14 https://deb.debian.org/debian bookworm-updates/main arm64 Packages [2468 B]
Get:15 https://deb.debian.org/debian-security bookworm-security/main arm64 Packages [187 kB]
Get:16 https://deb.debian.org/debian bookworm-backports/non-free-firmware arm64 Packages [3852 B]
Get:17 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages [4964 B]
Get:18 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages [10.1 kB]
Get:19 https://deb.debian.org/debian bookworm-backports/main arm64 Packages [250 kB]
Get:20 https://dietpi.com/apt bookworm/main arm64 Packages [2172 B]
Get:21 https://dietpi.com/apt all/nanopir5s arm64 Packages [2288 B]
Get:22 https://dietpi.com/apt all/nanopir5s all Packages [528 B]
Fetched 9637 kB in 4s (2520 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@NanoPiR5S:/tmp#
```
Also from the file size `8689 kB` (matching your log) you can see that it is in fact the `xz` archive: https://deb.debian.org/debian/dists/bookworm/main/binary-arm64/
The `gz` archive would be 11 MiB already, the plain text file 49 MiB.

That it hangs seems to be the issue, probably the same with the `xz` archive. Just weird that this happens with different mirrors, while downloading a package from the same mirror works fine. Can you try a really different one from here? https://www.debian.org/mirror/list

klankschap, 2024-10-29:
> Can you try a really different one from here? https://www.debian.org/mirror/list

Its bizarre  indeed. Changing to a different mirror doesn't seem to make a difference. I tried nl de uk.
The neopi doesn't seem to keep the line in order.
From the same local lan i can run different raspi to properly update their debian.

Is there maybe a network timing issue baked into the dietpi release for the neopi plus2? 

Searching along a different rout, is it possible to flash the neopi plus2 into emmc directly from a different linux machine?

klankschap, 2024-10-29:
```
# apt upgrade
The following packages will be upgraded:
  armbian-firmware linux-image-current-sunxi64                                            2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
14 not fully installed or removed.                                                        Need to get 0 B/123 MB of archives.
After this operation, 2234 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16724 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_24.11.0-trunk-dietpi1_all.deb ...
Unpacking armbian-firmware (24.11.0-trunk-dietpi1) over (24.8.0-trunk-dietpi2) ...        dpkg-deb (subprocess): decompressing archive '/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb' (size=91604592) member 'data.tar': lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb (--unpack):
 cannot copy extracted data for './lib/firmware/brcm/brcmfmac4356-sdio-nanopi-m4v2.bin' to '/lib/firmware/brcm/brcmfmac4356-sdio-nanopi-m4v2.bin.dpkg-new': unexpected end of file or stream                                              Preparing to unpack .../linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb ...
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 'prerm' starting.
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 'prerm' finishing.
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 'preinst' starting.   Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 'preinst' finishing.
Unpacking linux-image-current-sunxi64 (24.11.0-trunk-dietpi1) over (24.8.0-trunk-dietpi2) ...
dpkg-deb (subprocess): decompressing archive '/var/cache/apt/archives/linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb' (size=31346892) member 'data.tar': lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive /var/cache/apt/archives/linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb (--unpack):
 cannot copy extracted data for './boot/vmlinuz-6.6.44-current-sunxi64' to '/boot/vmlinuz-6.6.44-current-sunxi64.dpkg-new': unexpected end of file or stream
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 'postrm' starting.
Removing obsolete initramfs images
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 'postrm' finishing.
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 'postinst' starting.
Removing obsolete initramfs images
Armbian: update last-installed kernel symlink to 'Image'...
'/boot/Image' -> 'vmlinuz-6.6.31-current-sunxi64'
Armbian: Debian compat: linux-update-symlinks install 6.6.31-current-sunxi64 boot/vmlinuz-6.6.31-current-sunxi64
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 'postinst' finishing.
Errors were encountered while processing:
 /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb
 /var/cache/apt/archives/linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

StephanStS, 2024-10-29:
Könnte das eine volle Disk oder eine defekte Disk sein?

---------------------

Stephan Schultze

Glöserwiesenweg 4

97816 Lohr a. Main

Tel: 09352/89837

---------------------

klankschap, 2024-10-29:
On 29/10/2024 19:05, StephanStS wrote:
> Könnte das eine volle Disk oder eine defekte Disk sein?

Not sure, as it happens on different SD cards.
The timing errors seems to be related to the local switch for now.
That is, the update works, and the upgrade partially.

MichaIng, 2024-10-29:
Weirdly this looks now very different to this report here: #7261
However, you are using a different SBC with different kernel, and I just successfully tested the very same kernel and firmware upgrade (also some previous versions of both packages) on my Odroid N2+. So for now I would say coincidence.

But can you check for kernel errors?
```sh
dmesg -l 0,1,2,3
```

klankschap, 2024-10-29:
On 29/10/2024 22:20, MichaIng wrote:
> But can you check for kernel errors?
> 
> dmesg -l 0,1,2,3

with a fresh download from
https://dietpi.com/downloads/images/DietPi_NanoPiNEOPlus2-ARMv8-Bookworm.img.xz

it passed the initial setup (almost)

[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
   armbian-firmware linux-image-current-sunxi64
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
14 not fully installed or removed.
Need to get 123 MB of archives.
After this operation, 2234 kB of additional disk space will be used.
Get:1 https://dietpi.com/apt all/nanopineoplus2 all armbian-firmware all 
24.11.0-trunk-dietpi1 [91.6 MB]
Get:2 https://dietpi.com/apt all/nanopineoplus2 arm64 
linux-image-current-sunxi64 arm64 24.11.0-trunk-dietpi1 [31.3 MB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 123 MB in 15s (8262 kB/s)
(Reading database ... 16724 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_24.11.0-trunk-dietpi1_all.deb ...
Unpacking armbian-firmware (24.11.0-trunk-dietpi1) over 
(24.8.0-trunk-dietpi2) ...
dpkg-deb (subprocess): decompressing archive 
'/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb' 
(size=91604592) member 'data.tar': lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive 
/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb 
(--unpack):
  cannot copy extracted data for 
'./lib/firmware/brcm/brcmfmac4356-sdio-nanopi-m4v2.bin' to 
'/lib/firmware/brcm/brcmfmac4356-sdio-nanopi-m4v2.bin.dpkg-new': 
unexpected end of file or stream
Preparing to unpack 
.../linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb ...
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 
'prerm' starting.
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 
'prerm' finishing.
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 
'preinst' starting.
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 
'preinst' finishing.
Unpacking linux-image-current-sunxi64 (24.11.0-trunk-dietpi1) over 
(24.8.0-trunk-dietpi2) ...
dpkg-deb: error: <decompress> subprocess was killed by signal 
(Segmentation fault)
dpkg: error processing archive 
/var/cache/apt/archives/linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb 
(--unpack):
  cannot copy extracted data for './boot/vmlinuz-6.6.44-current-sunxi64' 
to '/boot/vmlinuz-6.6.44-current-sunxi64.dpkg-new': unexpected end of 
file or stream
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 
'postrm' starting.
Removing obsolete initramfs images
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 
'postrm' finishing.
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 
'postinst' starting.
Removing obsolete initramfs images
Armbian: update last-installed kernel symlink to 'Image'...
'/boot/Image' -> 'vmlinuz-6.6.31-current-sunxi64'
Armbian: Debian compat: linux-update-symlinks install 
6.6.31-current-sunxi64 boot/vmlinuz-6.6.31-current-sunxi64
Armbian 'linux-image-current-sunxi64' for '6.6.31-current-sunxi64': 
'postinst' finishing.
Errors were encountered while processing:
  /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb
  /var/cache/apt/archives/linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade

then the dmesg

***@***.***:~# dmesg -l 0,1,2,3
***@***.***:~#

klankschap, 2024-10-29:
starting with a fresh installer download, unmodified, it passes the initial setup, almost.

i remove the two packages that fail

`/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb
/var/cache/apt/archives/linux-image-current-sunxi64_24.11.0-trunk-dietpi1_arm64.deb
`
then relauch the upgrade

`
root@DietPi:~# rm /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb

root@DietPi:~# apt-get -y --with-new-pkgs upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  armbian-firmware
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
15 not fully installed or removed.
Need to get 91.6 MB of archives.
After this operation, 820 kB of additional disk space will be used.
Get:1 https://dietpi.com/apt all/nanopineoplus2 all armbian-firmware all 24.11.0-trunk-dietpi1 [91.6 MB]
Fetched 91.6 MB in 7s (12.3 MB/s)                                                                       
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16735 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_24.11.0-trunk-dietpi1_all.deb ...
Unpacking armbian-firmware (24.11.0-trunk-dietpi1) over (24.8.0-trunk-dietpi2) ...
dpkg-deb: error: <decompress> subprocess was killed by signal (Segmentation fault)
dpkg: error processing archive /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb (--unpack):
 cannot copy extracted data for './lib/firmware/ath12k/WCN7850/hw2.0/amss.bin' to '/lib/firmware/ath12k/WCN7850/hw2.0/amss.bin.dpkg-new': unexpected end of file or stream
Errors were encountered while processing:
 /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~# 
`

`root@DietPi:~# apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://dietpi.com/apt bookworm InRelease
Hit:6 https://dietpi.com/apt all InRelease     
Reading package lists... Done                  
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
root@DietPi:~# apt list --upgradable
Listing... Done
armbian-firmware/all 24.11.0-trunk-dietpi1 all [upgradable from: 24.8.0-trunk-dietpi2]
N: There are 3 additional versions. Please use the '-a' switch to see them.
root@DietPi:~# apt list --upgradable -a
Listing... Done
armbian-firmware/all 24.11.0-trunk-dietpi1 all [upgradable from: 24.8.0-trunk-dietpi2]
armbian-firmware/all,now 24.8.0-trunk-dietpi2 all [installed,upgradable to: 24.11.0-trunk-dietpi1]
armbian-firmware/all 24.8.0-trunk-dietpi1 all
armbian-firmware/all 24.5.0-trunk all
`

`
 Linux version 6.6.31-current-sunxi64 (armbian@next) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP Fri May 17 10:02:40 UTC 2024
`
magic for sure...

MichaIng, 2024-10-29:
Is it the same for you when you try to manually download and install that package:
```sh
cd /tmp
wget https://dietpi.com/apt/dists/all/nanopineoplus2/binary-all/armbian-firmware_24.11.0-trunk-dietpi1.deb
sha256sum armbian-firmware_24.11.0-trunk-dietpi1.deb
dpkg -i armbian-firmware_24.11.0-trunk-dietpi1.deb
```
The `sha256sum` should return `5ef8c82df7222b2a792f5f366f2d2dfda7b047c868b101a60334da4f8fd00531`.

klankschap, 2024-10-29:
`5ef8c82df7222b2a792f5f366f2d2dfda7b047c868b101a60334da4f8fd00531  armbian-firmware_24.11.0-trunk-dietpi1.deb`

`
/tmp# dpkg -i armbian-firmware_24.11.0-trunk-dietpi1.deb
(Reading database ... 16735 files and directories currently installed.)
Preparing to unpack armbian-firmware_24.11.0-trunk-dietpi1.deb ...
Unpacking armbian-firmware (24.11.0-trunk-dietpi1) over (24.8.0-trunk-dietpi2) ...
dpkg-deb (subprocess): decompressing archive 'armbian-firmware_24.11.0-trunk-dietpi1.deb' (size=91604592) member 'data.tar': lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive armbian-firmware_24.11.0-trunk-dietpi1.deb (--install):
 cannot copy extracted data for './lib/firmware/qcom/sm8550/ayn/odin2/adsp.mbn' to '/lib/firmware/qcom/sm8550/ayn/odin2/adsp.mbn.dpkg-new': unexpected end of file or stream
Errors were encountered while processing:
 armbian-firmware_24.11.0-trunk-dietpi1.deb
`

klankschap, 2024-10-30:
what is the difference between 
eg updating dietpi1 over dietpi22 ?

(24.11.0-trunk-dietpi1) over (24.8.0-trunk-dietpi2)

MichaIng, 2024-10-31:
So it is definitely something wrong with the `dpkg` extractor, and I guess it is the same with the other issue I linked above.

`24.11.0` is the Armbian build system version. We append an on suffix for every rebuild, usually implying a newer kernel version or updated firmware like in this case.

MichaIng, 2024-10-31:
Tested it on the Orange Pi Zero 3, which uses the same kernel, and the initial update from our image works well. If you have a spare SD card, could you check whether a current fresh DietPi image does the initial update fine for you?

And anther thing, does reinstalling other packages work, like `dpkg` itself?
```sh
apt install --reinstall dpkg
```

klankschap, 2024-10-31:
On 31/10/2024 21:42, MichaIng wrote:
> Tested it on the Orange Pi Zero 3, which uses the same kernel, and the 
> initial update from our image works well. If you have a spare SD card, 
> could you check whether a current fresh DietPi image does the initial 
> update fine for you?

just downloadded a fresh copy and it passed all the ways through the 
initial setup

> Building dependency tree...
> Reading state information...
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> [  OK  ] DietPi-Update | APT autopurge
> [  OK  ] DietPi-Update | systemctl daemon-reload
> [  OK  ] DietPi-Update | Incremental patching to v9.8.0 completed
> [ INFO ] DietPi-Update | Checking for new available live patches
> [  OK  ] DietPi-Update | eval echo 1 > /boot/dietpi/.install_stage
> 
>  DietPi-Update
> ─────────────────────────────────────────────────────
>  Phase: Completed
> 
> [ INFO ] DietPi-Update | Current version : v9.8.0
> [ INFO ] DietPi-Update | Latest version  : v9.8.0
> [  OK  ] DietPi-Update | sync
> [ INFO ] DietPi-Update | A reboot is done to finalise the kernel upgrade

whatever you did, it made the difference.
Much appreciated indeed!

> 
> And anther thing, does reinstalling other packages work, like |dpkg| itself?
> 
> apt install --reinstall dpkg

yes.
I also managed to switch dropbear to openssh :-)

Now i'll try to flash the emmc with the recent image.

.F

MichaIng, 2024-10-31:
And does reinstalling a larger package work?
```sh
apt install --reinstall armbian-firmware
```

klankschap, 2024-10-31:
On 31/10/2024 23:47, MichaIng wrote:
> And does reinstalling a larger package work?
> 
> apt install --reinstall armbian-firmware

nope, but first flashed the img into /dev/mmcblk2 and boot from there.
That works.
Installed python3

***@***.***:~# python3
Python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
 >>>

***@***.***:~# apt install --reinstall armbian-firmware
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not 
upgraded.
Need to get 91.6 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://dietpi.com/apt all/nanopineoplus2 all armbian-firmware all 
24.11.0-trunk-dietpi1 [91.6 MB]
Fetched 91.6 MB in 12s (7331 kB/s) 

debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16884 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_24.11.0-trunk-dietpi1_all.deb ...
Unpacking armbian-firmware (24.11.0-trunk-dietpi1) over 
(24.11.0-trunk-dietpi1) ...
dpkg-deb (subprocess): decompressing archive 
'/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb' 
(size=91604592) member 'data.tar': lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive 
/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb 
(--unpack):
  cannot copy extracted data for 
'./lib/firmware/qcom/sc8180x/SAMSUNG/W767/qcmpss8180_XEF.mbn' to 
'/lib/firmware/qcom/sc8180x/SAMSUNG/W767/qcmpss8180_XEF.mbn.dpkg-new': 
unexpected end of file or stream
Errors were encountered while processing:
  /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

did a reboot and tried again:

(Reading database ... 16965 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_24.11.0-trunk-dietpi1_all.deb ...
Unpacking armbian-firmware (24.11.0-trunk-dietpi1) over 
(24.11.0-trunk-dietpi1) ...
dpkg-deb (subprocess): decompressing archive 
'/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb' 
(size=91604592) member 'data.tar': lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive 
/var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb 
(--unpack):
  cannot copy extracted data for 
'./lib/firmware/rtl_bt/rtl8822cs_fw.bin' to 
'/lib/firmware/rtl_bt/rtl8822cs_fw.bin.dpkg-new': unexpected end of file 
or stream
Errors were encountered while processing:
  /var/cache/apt/archives/armbian-firmware_24.11.0-trunk-dietpi1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

.F

MichaIng, 2024-11-02:
Still no kernel errors?
```sh
dmesg -l 0,1,2,3
```

Does the same work on SD card?

And can you try it with a smaller recent package from our APT server:
```sh
cd /tmp
wget https://dietpi.com/apt/dists/all/orangepizero3/binary-arm64/sprd-bluetooth_0.0.1_arm64.deb
dpkg -i sprd-bluetooth_0.0.1_arm64.deb
```
Just maybe there is some issue with the way those packages are packed on the Ubuntu runner here on GitHub, an issue which shows up only in some rare circumstances, for whatever reason.

klankschap, 2024-11-03:
On 02/11/2024 23:09, MichaIng wrote:
> Still no kernel errors?
> 
> dmesg -l 0,1,2,3

***@***.***:~# dmesg -l 0,1,2,3
[    4.190228] usb 9-1: device descriptor read/64, error -62
[    4.517877] usb 9-1: device descriptor read/64, error -62
[    5.017793] usb 9-1: device descriptor read/64, error -62
[    5.805791] usb 9-1: device descriptor read/64, error -62
[    6.561793] usb 9-1: device not accepting address 4, error -62
[    7.177788] usb 9-1: device not accepting address 5, error -62
[    7.183674] usb usb9-port1: unable to enumerate USB device
[ 1155.001909] usb 9-1: device descriptor read/64, error -62
[ 1155.305872] usb 9-1: device descriptor read/64, error -62
[ 1155.797868] usb 9-1: device descriptor read/64, error -62
[ 1156.101912] usb 9-1: device descriptor read/64, error -62
[ 1156.845897] usb 9-1: device not accepting address 8, error -62
[ 1157.454212] usb 9-1: device not accepting address 9, error -62
[ 1157.460817] usb usb9-port1: unable to enumerate USB device
[60158.962628] usb 9-1: device descriptor read/64, error -62
[60160.019139] usb 9-1: device descriptor read/64, error -62
[60160.511376] usb 9-1: device descriptor read/64, error -62
[60160.815529] usb 9-1: device descriptor read/64, error -62
[60161.559892] usb 9-1: device not accepting address 12, error -62
[60162.168205] usb 9-1: device not accepting address 13, error -62
[60162.174324] usb usb9-port1: unable to enumerate USB device
[60178.608426] usb 9-1: device descriptor read/64, error -62
[60179.684972] usb 9-1: device descriptor read/64, error -62
[60180.177208] usb 9-1: device descriptor read/64, error -62
[60180.481364] usb 9-1: device descriptor read/64, error -62
[60181.225701] usb 9-1: device not accepting address 16, error -62
[60181.834033] usb 9-1: device not accepting address 17, error -62
[60181.840142] usb usb9-port1: unable to enumerate USB device
[60240.043148] usb 9-1: device descriptor read/64, error -62
[60241.119748] usb 9-1: device descriptor read/64, error -62
[60241.611925] usb 9-1: device descriptor read/64, error -62
[60241.916089] usb 9-1: device descriptor read/64, error -62
[60242.660447] usb 9-1: device not accepting address 21, error -62
[60243.268748] usb 9-1: device not accepting address 22, error -62
[60243.274878] usb usb9-port1: unable to enumerate USB device

> Does the same work on SD card?

No this runs from emmc

> 
> And can you try it with a smaller recent package from our APT server:
> 
> cd /tmp
> wget https://dietpi.com/apt/dists/all/orangepizero3/binary-arm64/sprd-bluetooth_0.0.1_arm64.deb

  wget 
https://dietpi.com/apt/dists/all/orangepizero3/binary-arm64/sprd-bluetooth_0.0.1_arm64.deb
--2024-11-03 01:00:26-- 
https://dietpi.com/apt/dists/all/orangepizero3/binary-arm64/sprd-bluetooth_0.0.1_arm64.deb
Resolving dietpi.com (dietpi.com)... 2a06:98c1:3121::, 2a06:98c1:3120::, 
188.114.97.0, ...
Connecting to dietpi.com (dietpi.com)|2a06:98c1:3121::|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 64828 (63K) [application/vnd.debian.binary-package]
Saving to: ‘sprd-bluetooth_0.0.1_arm64.deb’

sprd-bluetooth_0.0. 100%[==================>]  63.31K  --.-KB/s    in 
0.003s

2024-11-03 01:00:26 (19.0 MB/s) - ‘sprd-bluetooth_0.0.1_arm64.deb’ saved 
[64828/64828]

> dpkg -i sprd-bluetooth_0.0.1_arm64.deb

# dpkg -i sprd-bluetooth_0.0.1_arm64.deb
Selecting previously unselected package sprd-bluetooth.
(Reading database ... 23625 files and directories currently installed.)
Preparing to unpack sprd-bluetooth_0.0.1_arm64.deb ...
Unpacking sprd-bluetooth (0.0.1) ...
Setting up sprd-bluetooth (0.0.1) ...
Configuring sprd-bluetooth systemd service ...
Created symlink 
/etc/systemd/system/multi-user.target.wants/sprd-bluetooth.service → 
/lib/systemd/system/sprd-bluetooth.service.

> Just maybe there is some issue with the way those packages are packed on 
> the Ubuntu runner here on GitHub, an issue which shows up only in some 
> rare circumstances, for whatever reason.
> 

# apt update -y && apt upgrade -y

Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease 
[48.0 kB]
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [59.0 kB]
Hit:5 https://dietpi.com/apt bookworm InRelease
Hit:6 https://dietpi.com/apt all InRelease
Fetched 162 kB in 4s (42.9 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done

seems to work...

.F

MichaIng, 2024-11-03:
Okay, then I suspect the large size of the linux image and firmware packages to be related to the issue. Do larger packages from Debian fail as well?
```sh
apt install openjdk-17-jre-headless
```

There seem to be an issue with the USB device, but should not be related to installing a package on the eMMC. But to rule it out, if you detach (in case unmount) that USB device, does it make a difference?

The issue originally appeared when booting from SD card, and now also when booting from eMMC, so that does not seem to make a difference, right?

klankschap, 2024-11-03:
On 03/11/2024 19:17, MichaIng wrote:
> Okay, then I suspect the large size of the linux image and firmware 
> packages to be related to the issue. Do larger packages from Debian fail 
> as well?
> 
> apt install openjdk-17-jre-headless

this installs just fine

> 
> There seem to be an issue with the USB device, but should not be related 
> to installing a package on the eMMC. But to rule it out, if you detach 
> (in case unmount) that USB device, does it make a difference?

not sure wha5t you mean.
the unit is headless, attached via ssh only.
initial boot via SD
after initial-setup, i changed the ssh server to openssh to have scp.
then scp the img to a ~/tmp folder, then dd that image to /dev/mmcblk2
then reboot and do an initial setup again.

> The issue originally appeared when booting from SD card, and now also 
> when booting from eMMC, so that does not seem to make a difference, right?

The initial issue was likely related to the switch in the LAN causing 
timeouts in the downloading. I don't know if in the mean time the 
content of the online images have changed.

At present the unit is working as intended

.F

MichaIng, 2025-06-02:
Closing this issue. Feel free to reopen if the problem still exists.

closed 2025-06-02 by MichaIng