MichaIng/DietPi #7260: YDNS doesn't work by default

NatureHog, 2024-10-28:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm 0
- Kernel version | `uname -a`
Linux NaturePi 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3
(2024-10-08) aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 5 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
Official RPi5 Power Supply
- SD card used | (EG: SanDisk ultra)
[SanDisk GamePlay 512GB](https://shop.sandisk.com/products/memory-cards/microsd-cards/sandisk-gameplay-microsd-card?sku=SDSQXAV-512G-GN6XN)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Use the built-in DietPi-DDNS script with Provider set to YDNS
2. Error, even when server replies "good"
3. Change Provider to custom, using ydns.io as address
4. Works flawlessly

#### Expected behaviour

- Built-in option shouldn't give an error

#### Actual behaviour

- Built-In option gives an error, regardless of the servers reply
- Custom option with the same API URL works without problems.

#### Extra details

- Since I only have an RPi5 on hand, I couldn't try with other devices / not the "testing" branch.
- Sorry if formatting of this report is off, currently using GH Mobile...

MichaIng, 2024-10-29:
Many thanks for your report. The result is "good" now? It was "ok" in the past. Do you also see "nochg" when running the test/request a second time, as indicator that the update was successful but your IP address did not change? Then they might have updated their API to match the one of Dynu.

NatureHog, 2024-10-29:
Yes, it'd show "nochg", just like you said.
Glad I could help.

NatureHog, 2024-10-29:
Just submitted #7262 which _should_ solve this, but haven't got the time to test it, yet.

MichaIng, 2024-10-31:
Fix merged for next release. Many thanks 🙂 !

closed 2024-10-31 by MichaIng