MichaIng/DietPi #7263: Configure screen resolution/orientation in Kiosk mode.

Nurgak, 2024-10-31:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
In Kiosk mode, sometimes it is useful to set a custom resolution and orientation, rather than the maximum and landscape.

However, currently the Kiosk mode will force the screen resolution to the maximum and orientation to landscape. The only way to change it is via `xrand`, such as `xrandr -d :0 -o right -s 1280x900`. This needs to be called every time when Kiosk mode starts.

I'm sure there is a more elegant way to do it, but I have not been successful finding it. The settings in `/boot/config.txt` (RPi-specific?) only apply for the command line, not in Kiosk mode. Generally speaking, the browser itself should always be maximized in Kiosk mode, so [this issue is related](https://github.com/MichaIng/DietPi/issues/6866).

#### Describe the solution you'd like:
Ideally, I'd like the settings in `/boot/config.txt` to apply in Kiosk mode, both for the screen size as well as orientation.

#### Describe alternatives you've considered:
I considered modifying the Kiosk mode startup script in `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`, to add `xrandr -d :0 -o right -s 1280x900`, but I don't like it because this cannot be configured before first boot.

#### Additional context
My use-case is to display some JavaScript animation. More pixels to refresh is more demanding, and will slow down the animation. Having a more "pixelated" screen is not an issue. This is why a lower screen resolution is useful for me (also portrait orientation).

Related request: I'd like the `CHROMIUM_OPTS` of the Kiosk mode to be exposed (uncommented) so it could be configured before first boot in `dietpi.txt`. This is useful as it allows to hide the cursor (and [other improvements](https://dietpi.com/forum/t/non-interactive-rpi-kiosk-improvements/14886)).

MichaIng, 2024-10-31:
Here is a request about this: https://github.com/MichaIng/DietPi/issues/5585

In addition, since KMS is mandatory for Chromium, can you try whether it works (instead of using any X config or `xrandr`) to add
```
HDMI-A-1:768x1024,rotate=90
```
to `/boot/cmdline.txt` (space-separated). In case it is not the right HDMI display, find the correct one in `/sys/class/drm/`, and adjust the resolution as intended. I am actually not sure whether the values need to be inverted with `rotate=90`.

Nurgak, 2024-11-05:
@MichaIng 
Thank you for the reference and the information. I have tried different options in `/boot/cmdline.txt`, but none work the way I expect. Before Chromium starts the screen resolution is maxed out no matter what options I use.

However, I found the option that does what I need:

* `dietpi.txt`
   ```yaml
   SOFTWARE_CHROMIUM_RES_X=720
   SOFTWARE_CHROMIUM_RES_Y=1280
   ```
* `config.txt`
   ```yaml
   framebuffer_width=720
   framebuffer_height=1280
   display_hdmi_rotate=1
   ```

In `dietpi-config` set Display Options -> Display Resolution to `720p`. When set to `vc4-kms-v3d` it does not work.

I wonder if I can get [graphical acceleration working in Chromium](https://lemariva.com/blog/2020/08/raspberry-pi-4-video-acceleration-decode-chromium) without setting `vc4-kms-v3d`...

Nurgak, 2024-11-08:
Just to update this question with my experience so far:
- Chromium does not need any extra configuration in `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh` to have OpenGL acceleration enabled (RPi4).
   - However, upon the first boot/headless install the RPi needs to be **rebooted** for OpenGL to be available for Chromium.
   - Make sure the `DISPLAY` environment variable is **not set**, or OpenGL will not be available.
- There is no `dietpi.txt`, `cmdline.txt` or `config.txt` options that would allow screen resolution/orientation definition for Chromium in Kiosk mode, it will be reset to the maximum in landscape upon the script execution.
   - The only way to change it is to call `xrandr -d :0 -o right -s 1280x900` once Chomium is started.
- The Chromium auto-start script (`/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`) has **flawed instructions** on how to customize the options: `CHROMIUM_OPTS+=' --...'` does not work (options are not appended, they are ignored).
   - It should be `CHROMIUM_OPTS+="${CHROMIUM_OPTS} --..."`.
   - I would make a PR, but I don't know where the Chromium auto-start script template is hosted.

cadavre, 2024-11-14:
@Nurgak can you share your diff of `chromium-autostart.sh`? I'm trying to achieve the same result, but resolution is not changing.

Nurgak, 2024-11-15:
@cadavre I did not touch `chromium-autostart.sh`, ideally I would not want to as it's a generated file.

The resolution can only be set using `xrandr -d :0 -o right -s 1280x900` from the command line. Perhaps this command can be added to `chromium-autostart.sh`?

MDAR, 2024-12-19:
The resolution for the autostart chromium file is pulled from the dietpi.txt file, in the Chromium section

Nurgak, 2024-12-20:
@MDAR 
If you are taking about the `SOFTWARE_CHROMIUM_RES_X/Y` then yes, these fix the Chromium size, however that is different from the _display_ resolution, which defaults to the maximum the screen can support. So you can set a lower Chromium resolution, but then Chromium will only take up part of the screen.

My goal was to set the display size _and_ Chromium size so Chromium would be full-screened while the display is at a lower resolution than its maximum, which is currently not possible when using `vc4-kms-v3d`.

MichaIng, 2025-07-18:
In the meantime, `dietpi-display` allows to adjust display resolution for KMS/DRM enabled.

closed 2025-07-18 by MichaIng