MichaIng/DietPi #7264: Home Assistant installation needs bluez

stevexyz, 2024-10-31:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=9 G_DIETPI_VERSION_SUB=8 G_DIETPI_VERSION_RC=0 G_GITBRANCH='master' G_GITOWNER='MichaIng'
- Distro version | bookworm 0
- Kernel version | Linux jumpy5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
- SBC model | RPi5
- Power supply used | 5V 4A 
- SD card used | std

#### Additional Information (if applicable)
- Software title | Home Assistant
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. Install Home Assistant

#### Expected behaviour
- RPi bluetooth working

#### Actual behaviour
- HA reports error

#### Extra details
- I've solved manually installing bluez package

MichaIng, 2024-10-31:
There is a menu entry in `dietpi-config` to enable Bluetooth. Installing `bluez` alone would not enable it. Do you actually use any Bluetooth functionality in HA? It sadly changed the past months/year to throw a lot of errors about missing this or that, including FFmpeg etc, while most of these features are only used by a minority of users, at least are (of course) not required for HA core functionality. We intentionally ignore everything that does not make HA actually crash.

stevexyz, 2024-11-11:
It got few equipment in my house but not sure if all via wifi or some via bluetooth (few have both options), but at least it does not give error.
Anyway I can understand your way of reasoning, and probably this is the way to keep DietPi "clean", but then not sure if a big and complex app as HA is a good match deployed as a set of packages or better keep it as docker container anyway...

MichaIng, 2024-11-14:
Here is startup logs of fresh Home Assistant on my RPi 5 with WiFi enabled:
```
Nov 14 22:34:23 DietPi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
Nov 14 22:34:26 DietPi Home Assistant[31284]: 2024-11-14 22:34:26.449 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Nov 14 22:34:29 DietPi Home Assistant[31284]: 2024-11-14 22:34:29.175 WARNING (MainThread) [aiohttp_fast_zlib] zlib_ng and isal are not available, falling back to zlib, performance will be degraded.
Nov 14 22:35:27 DietPi Home Assistant[31284]: 2024-11-14 22:35:27.720 ERROR (ImportExecutor_0) [homeassistant.components.camera.img_util] Error loading libturbojpeg; Camera snapshot performance will be sub-optimal
Nov 14 22:35:27 DietPi Home Assistant[31284]: Traceback (most recent call last):
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/site-packages/homeassistant/components/camera/img_util.py", line 100, in __init__
Nov 14 22:35:27 DietPi Home Assistant[31284]:     TurboJPEGSingleton.__instance = TurboJPEG()
Nov 14 22:35:27 DietPi Home Assistant[31284]:                                     ^^^^^^^^^^^
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/site-packages/turbojpeg.py", line 300, in __init__
Nov 14 22:35:27 DietPi Home Assistant[31284]:     self.__find_turbojpeg() if lib_path is None else lib_path)
Nov 14 22:35:27 DietPi Home Assistant[31284]:     ^^^^^^^^^^^^^^^^^^^^^^^
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/site-packages/turbojpeg.py", line 967, in __find_turbojpeg
Nov 14 22:35:27 DietPi Home Assistant[31284]:     raise RuntimeError(
Nov 14 22:35:27 DietPi Home Assistant[31284]: RuntimeError: Unable to locate turbojpeg library automatically. You may specify the turbojpeg library path manually.
Nov 14 22:35:27 DietPi Home Assistant[31284]: e.g. jpeg = TurboJPEG(lib_path)
Nov 14 22:35:27 DietPi Home Assistant[31284]: 2024-11-14 22:35:27.945 ERROR (MainThread) [haffmpeg.core] FFmpeg fails [Errno 2] No such file or directory: 'ffmpeg'
Nov 14 22:35:27 DietPi Home Assistant[31284]: Traceback (most recent call last):
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/site-packages/haffmpeg/core.py", line 130, in open
Nov 14 22:35:27 DietPi Home Assistant[31284]:     self._proc = await asyncio.create_subprocess_exec(
Nov 14 22:35:27 DietPi Home Assistant[31284]:                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/subprocess.py", line 224, in create_subprocess_exec
Nov 14 22:35:27 DietPi Home Assistant[31284]:     transport, protocol = await loop.subprocess_exec(
Nov 14 22:35:27 DietPi Home Assistant[31284]:                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/base_events.py", line 1743, in subprocess_exec
Nov 14 22:35:27 DietPi Home Assistant[31284]:     transport = await self._make_subprocess_transport(
Nov 14 22:35:27 DietPi Home Assistant[31284]:                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/unix_events.py", line 211, in _make_subprocess_transport
Nov 14 22:35:27 DietPi Home Assistant[31284]:     transp = _UnixSubprocessTransport(self, protocol, args, shell,
Nov 14 22:35:27 DietPi Home Assistant[31284]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/base_subprocess.py", line 36, in __init__
Nov 14 22:35:27 DietPi Home Assistant[31284]:     self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/asyncio/unix_events.py", line 820, in _start
Nov 14 22:35:27 DietPi Home Assistant[31284]:     self._proc = subprocess.Popen(
Nov 14 22:35:27 DietPi Home Assistant[31284]:                  ^^^^^^^^^^^^^^^^^
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/subprocess.py", line 1026, in __init__
Nov 14 22:35:27 DietPi Home Assistant[31284]:     self._execute_child(args, executable, preexec_fn, close_fds,
Nov 14 22:35:27 DietPi Home Assistant[31284]:   File "/home/homeassistant/.pyenv/versions/3.12.7/lib/python3.12/subprocess.py", line 1955, in _execute_child
Nov 14 22:35:27 DietPi Home Assistant[31284]:     raise child_exception_type(errno_num, err_msg, err_filename)
Nov 14 22:35:27 DietPi Home Assistant[31284]: FileNotFoundError: [Errno 2] No such file or directory: 'ffmpeg'
Nov 14 22:35:27 DietPi Home Assistant[31284]: 2024-11-14 22:35:27.948 WARNING (MainThread) [haffmpeg.tools] Error starting FFmpeg.
[... cut ...]
Nov 14 23:00:54 DietPi Home Assistant[31284]: 2024-11-14 23:00:54.773 WARNING (MainThread) [homeassistant.helpers.translation] Failed to load integration for translation: Invalid domain mobile_app.notify
Nov 14 23:00:54 DietPi Home Assistant[31284]: 2024-11-14 23:00:54.773 WARNING (MainThread) [homeassistant.helpers.translation] Failed to load integration for translation: Invalid domain energy.sensor
```
Where do you see an error about Bluetooth? Can you paste it? The log/error message should contain the integration as well, which demands it.

stevexyz, 2024-11-21:
I got it on the user interface, I didn't check the logs

MichaIng, 2024-11-21:
Where in the UI does this show up exactly?

MichaIng, 2025-06-02:
Closing this issue. Feel free to reopen if the problem still exists.

closed 2025-06-02 by MichaIng