MichaIng/DietPi #7265: MineOS refuses to install

mikedebian, 2024-10-31:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=9
 G_DIETPI_VERSION_SUB=8
 G_DIETPI_VERSION_RC=0
 G_GITBRANCH='master'
 G_GITOWNER='MichaIng'

- Distro version | Bookworm
- Kernel version | Linux BLACKPI 6.1.0-26-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux
- SBC model | Native, x64 UEFI
- SSD install

#### Additional Information (if applicable)
- Software title | MineOS
- Was the software title installed freshly or updated/migrated? Installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Dietpi fresh install
- Bug report ID | 78d8132e-38f7-4e5a-9b2b-c3d10478f177

#### Steps to reproduce
1. dietpi-software install 53

#### Expected behaviour
- MineOS should be installed

#### Actual behaviour

- NPM errors out. Whiptail gives pages with errors about deprecated/removed functions.

#### Extra details

Works on my raspberry pi.

mikedebian, 2024-10-31:
Here is the [pastebin](https://pastebin.com/z0fTdmwG)

MichaIng, 2024-10-31:
Thanks for reporting. I started a quick test across all platforms and versions: https://github.com/MichaIng/DietPi/actions/runs/11619447508

MichaIng, 2024-10-31:
Weird, like a syntax error in the `userid` node module, installed as dependency, but weirdly it fails on Bookworm only, while the Node version is the same and module version should be the same. For `userid` a beta version is installed `userid@1.0.0-beta.9`, maybe it will solve itself once that one became stable.

.... nope weird: https://www.npmjs.com/package/userid?activeTab=versions
There is a stable version `1.0.0`, but the beta 9 was uploaded 9 seconds later, hence `npm` grabs that one las "latest" 1.0.0 instead 🤣. However, that is 4 years old, so somehow I do not think that the beta is the problem. But updating that dependency might solve things regardless. But still not sure why it can fail on Bookworm only, with identical Node versions.

mikedebian, 2024-10-31:
Linux shenanigans :smiling_face_with_tear:  Hope you are able to figure it out...

mikedebian, 2024-11-06:
> Weird, like a syntax error in the `userid` node module, installed as dependency, but weirdly it fails on Bookworm only, while the Node version is the same and module version should be the same. For `userid` a beta version is installed `userid@1.0.0-beta.9`, maybe it will solve itself once that one became stable.
> 
> .... nope weird: https://www.npmjs.com/package/userid?activeTab=versions There is a stable version `1.0.0`, but the beta 9 was uploaded 9 seconds later, hence `npm` grabs that one as "latest" 1.0.0 instead 🤣. However, that is 4 years old, so somehow I do not think that the beta is the problem. But updating that dependency might solve things regardless. But still not sure why it can fail on Bookworm only, with identical Node versions.

Any update on this?

MichaIng, 2024-11-09:
PR up to fix/workaround this, and additionally fixing/working around the login issue #5759 which required #4551 to be addressed: #7276 

Indeed, when just updating the `userid` module, all works well. What you could do, when facing the error, is select "Open shell" from the error handler, and run this command:
```sh
G_CONFIG_INJECT '"userid"' '    "userid": "1.2.5",' package.json
exit
```
Then "Retry".

To fix the login issue, create an new non-root user:
```sh
sudo useradd -rMU -d -d /mnt/dietpi_userdata/mineos/serverdata -s /usr/sbin/nologin mineos
sudo chpasswd --crypt-method SHA512 <<< 'mineos:dietpi'
```
Change "dietpi" to a different password. Then use user "mineos" with this password to login the MineOS web UI.

MichaIng, 2024-11-09:
Fix sent upstream: https://github.com/hexparrot/mineos-node/pull/558

mikedebian, 2024-11-09:
> ```shell
> sudo systemctl restart mineos
> ```

You have been busy! This worked great, thank you :)

closed 2024-12-25 by MichaIng