MichaIng/DietPi #7267: Yacy administration page errors out.

mikedebian, 2024-11-01:
### Creating a bug report/issue
- [✓] I have searched the existing open and closed issues

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=9
 G_DIETPI_VERSION_SUB=8
 G_DIETPI_VERSION_RC=0
 G_GITBRANCH='master'
 G_GITOWNER='MichaIng'

- Distro version | Bookworm
- Kernel version | `Linux BLACKPI 6.1.0-26-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux`
- SBC model | Native, x64 UEFI
- SSD install

#### Additional Information (if applicable)
- Software title | YaCy id: 133
- Was the software title installed freshly or updated/migrated? Installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Dietpi fresh install

#### Steps to reproduce
1. dietpi-software install 133

#### Expected behaviour
- Yacy loading and working 100%

#### Actual behaviour

- Yacy is unable to load the administration page.
- http://1.2.3.4:8090/ConfigBasic.html

#### Extra details

[Pastebin of error message ](https://pastebin.com/D9qhUWVg)

Joulinar, 2024-11-01:
hmm I did a test an a VM without issues. Installation succeeded and web site is reachable right after

service log
```
root@DietPiVM:~# journalctl -u yacy.service
Nov 01 10:38:20 DietPiVM systemd[1]: Starting yacy.service - YaCy (DietPi)...
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]: ****************** YaCy Web Crawler/Indexer & Search Engine *******************
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]: **** (C) by Michael Peter Christen, usage granted under the GPL Version 2  ****
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]: ****   USE AT YOUR OWN RISK! Project home and releases: http://yacy.net/   ****
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]: **  LOG of       YaCy: DATA/LOG/yacy00.log (and yacy<xx>.log)                **
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]: **  STOP         YaCy: execute stopYACY.sh and wait some seconds             **
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]: **  GET HELP for YaCy: join our community at https://searchlab.eu            **
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]: *******************************************************************************
Nov 01 10:38:20 DietPiVM startYACY.sh[2737]:  >> YaCy started as daemon process. Administration at http://localhost:8090 <<
Nov 01 10:38:20 DietPiVM systemd[1]: Started yacy.service - YaCy (DietPi).
root@DietPiVM:~#
```

service status
```
root@DietPiVM:~# systemctl status yacy.service
● yacy.service - YaCy (DietPi)
     Loaded: loaded (/etc/systemd/system/yacy.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-11-01 10:38:20 GMT; 3min 2s ago
       Docs: https://wiki.yacy.net/index.php/En:Start
    Process: 2737 ExecStart=/etc/yacy/startYACY.sh (code=exited, status=0/SUCCESS)
   Main PID: 2749 (java)
      Tasks: 69 (limit: 2349)
     Memory: 305.9M
        CPU: 12.602s
     CGroup: /system.slice/yacy.service
             └─2749 /usr/bin/java -Xmx600m -Xms180m -server ....
```

LISTEN ports
```
root@DietPiVM:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=471,fd=3))
tcp   LISTEN 0      128                *:8090            *:*    users:(("java",pid=2749,fd=177))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=471,fd=4))
root@DietPiVM:~#
```

Where did you have seen that error message

Joulinar, 2024-11-01:
ok probably similar to this one.  https://github.com/yacy/yacy_search_server/issues/579

I guess a password would need to be set first. Can be done according our online docs https://dietpi.com/docs/software/distributed_projects/#yacy-admin-interface-login

mikedebian, 2024-11-04:
That worked! I SSH in to the system and set the password to my own personal one and the page loads now. I did not know it was mandatory..

Joulinar, 2024-11-04:
@MichaIng maybe we should set password automatically during install? I can submit a PR if you agree.

mikedebian, 2024-11-04:
> e

Using the global password would be a good idea

Joulinar, 2024-11-07:
There is some light at the end of the tunnel, and we will update Yacy version + install process on next DietPi version.

closed 2024-12-25 by MichaIng