MichaIng/DietPi #7268: Please add support for NanoPi Zero2

r3jack, 2024-11-01:
### Creating an image request
NanoPi Zero2 1G~2G

#### Formal device information
- Device name | NanoPi Zero2
- Official product URL | https://wiki.friendlyelec.com/wiki/index.php/NanoPi_Zero2
- GitHub resource URL (if available) | https://github.com/friendlyarm
- Image download URL | https://drive.google.com/drive/folders/1TbhufnT6NITr35uEvn_UsvnZPx6PbMEL

#### Is the SBC officially supported by the Debian installer?
- Not sure.

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Product wiki provide debian bookworm image.

sander030, 2024-12-21:
yes please add support for nano pi zero 2

sander030, 2024-12-22:
> Please add the support. The Zero is also already supported.

nano pi zero 2?

[image]

Joulinar, 2024-12-22:
As long as there is no own image, you can use the Debian image from Friendlyelec (rk3528-sd-debian-bookworm-core-6.1-arm64-20241112.img) and turn it into a DietPi system afterwards. (generic device)

https://wiki.friendlyelec.com/wiki/index.php/NanoPi_Zero2#Downloads

sander030, 2024-12-22:
> As long as there is no own image, you can use the Debian image from Friendlyelec (rk3528-sd-debian-bookworm-core-6.1-arm64-20241112.img) and turn it into a DietPi system afterwards. (generic device)
> 
> https://wiki.friendlyelec.com/wiki/index.php/NanoPi_Zero2#Downloads
> 
> This is working on my NanoPi Z2 without issues.

nice, good to know^^

sander030, 2024-12-23:
> As long as there is no own image, you can use the Debian image from Friendlyelec (rk3528-sd-debian-bookworm-core-6.1-arm64-20241112.img) and turn it into a DietPi system afterwards. (generic [...]

[image]

 which git branch i have to select?

sander030, 2024-12-26:
> Collaborator

I’m getting error messages when trying to turn it into a DietPi. Do you happen to have a small guide?

Joulinar, 2024-12-26:
What exactly is your challenge? For me it was working without issues.

MichaIng, 2025-01-12:
I guess the FriendlyELEC image is of this type with 8 (or is it now 9?) partitions, no access to boot or kernel config, and an overlayfs as root filesystem, right?

sander030, 2025-01-12:
I haven't managed to get it to work so far, tried all three, none worked dev beta master > @Joulinar  how do you get it to run? Did you choose Dev, Beta or Master release of Generic Installer?

Joulinar, 2025-01-13:
just tested it again without issues. All default and standard, I used the option `Flash Official OS to eMMC` to install Debian image to eMMC initially.

kochegin, 2025-02-27:
Hello! Please add support for NanoPi Zero2

dheydeck, 2025-03-24:
An earlier comment that I made that the conversion script ran against `rk3528-eflasher-debian-bookworm-core-6.1-arm64-20250211.img` on the NanoPi Zero2 had not worked was, in fact, incorrect. The script worked brilliantly- it was just that the IP address had changed.
Still, an official image for the Zero2 would be most welcome.

FullyRemoteDev, 2025-05-01:
Yes. Please. An official image for the NanoPi zero2 would be nice 👍

clariion, 2025-09-02:
> Yes. Please. An official image for the NanoPi zero2 would be nice 👍

Need this more than ever :(

lbmh, 2025-09-18:
Please add support for the NanoPI Zero 2.

MichaIng, 2026-02-15:
Mainline Linux support is generally there, but somehow during tests Ethernet was not working. It is weird since the mainline kernel device tree explicitly adds Ethernet nodes and the related mailing list discussion states it is working. Not sure at this point what is missing, we'd need someone with serial console access to see what is actually going on, whether the driver is loaded, or missing, or what/where it failed to come up. This means to take the PCB out of the neat passive cooling case, sadly 😢. Maybe someone else bout it without case, or with one that leaves the GPIO header/UART pins accessible?

But I mean, it does work well with vendor Linux, including Ethernet, so it is nothing really urgent. Would be just nice to ship it with mainline Linux 6.18 instead of vendor Linux 6.1 that is package with 3rd party code and out-of-tree drivers that throw a bunch of kernel warnings and errors.