MichaIng/DietPi #7270: Image |  NanoPI R3S

stefan01, 2024-11-02:
### Creating an image request

#### Formal device information
- Device name | NanoPI R3S
- Official product URL | https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R3S
- GitHub resource URL (if available) | https://github.com/friendlyarm
- Image download URL | https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R3S#Official_image

#### Is the SBC officially supported by the Debian installer?
- Not Sure.

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Debian Bookworm from wiki link

#### If not, are there install instructions for Debian available?
- Yes, see wiki: https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R3S#Install_OS

azagramac, 2024-12-06:
Only comment that I hope they will release an official version of DietPi for the NanoPi R3S. 🙏

azagramac, 2024-12-23:
@MichaIng  https://github.com/armbian/build/pull/7556#issuecomment-2558426285 It seems that armbian is interested in support for the RS3 board

EvilOlaf, 2025-02-01:
"Interested"...not really. That was just out of boredom if I can do it.

closed 2025-10-19 by MichaIng