MichaIng/DietPi #7271: no update to 0.9.8

Andy3001, 2024-11-03:
#### Details:                                                                                                                                                                                                                                                                       
- Date           | Sun Nov  3 17:20:08 CET 2024                                                                                                                                                                                                                                     
- DietPi version | v9.4.2 (MichaIng/master)                                                                                                                                                                                                                                         
- Image creator  |                                                                                                                                                                                                                                                                  
- Pre-image      |                                                                                                                                                                                                                                                                  
- Hardware       | ROCK64 (aarch64) (ID=43)                                                                                                                                                                                                                                         
- Kernel version | `Linux rock64 6.6.16-current-rockchip64 #1 SMP PREEMPT Fri Feb 23 08:25:28 UTC 2024 aarch64 GNU/Linux`                                                                                                                                                           
- Distro         | bookworm (ID=7)                                                                                                                                                                                                                                                  
- Command        | `systemctl start systemd-timesyncd`                                                                                                                                                                                                                              
- Exit code      | 1                                                                                                                                                                                                                                                                
- Software title | DietPi-TimeSync                                                                                                                                                                                                                                                  
#### Steps to reproduce:                                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                         
1. ...                                                                                                                                                                                                                                                                              
2. ...                                                                                                                                                                                                                                                                              
#### Expected behaviour:                                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                        
- ...                                                                                                                                                                                                                                                                               
#### Actual behaviour:                                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                         
- ...                                                                                                                                                                                                                                                                               
#### Extra details:                                                                                                                                                                                                                                                                 
                                                                                                                                                                                                              
- ...                                                                                                                                                                                                                                                                               
#### Additional logs:                                                                                                                                                                                                                                                               
```                                                                                                                                                                                                                                                                                 
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.

MichaIng, 2024-11-03:
Did you install any other network time sync daemon? Some of them mask `systemd-timesyncd` via maintainer scripts, or may have a conflict with the `systemd-timesyncd` package.

If so, please switch time sync mode to "custom", via `dietpi-config` or
```sh
sudo /boot/dietpi/func/dietpi-set_software ntpd-mode 0
```

If there is no alternative time sync daemon, unmask `systemd-timesyncd`, in case assure it is installed:
```sh
sudo apt install systemd-timesyncd
systemctl unmask systemd-timesyncd
```
Then rerun the update.

Joulinar, 2024-11-03:
somethimes this happen after upgrade from Debian Bullseye to Debian Bookworm

MichaIng, 2024-11-03:
Ah right, as the package is no strict dependency anymore, on Bookworm.

bobodaclown1, 2024-11-17:
Thank you. I had same issue. I followed the steps above:
swith time sync mode to "custom"
and then the "sudo apt install..."
Worked like a champ. 
Thank you for the support. Running V9.8.9 on my ROCK64 that is supporting pihole for my home network.

MichaIng, 2024-11-19:
@bobodaclown1 it is either switching to "custom" mode, or installing `systemd-timesyncd`, not both. Either you installed a different time sync daemon already, hence do not need `systemd-timesyncd`, then you switch to "custom". Or you did not install any time sync daemon by yourself, then install `systemd-timesyncd`, but leave the time sync mode at daily, hourly or daemon, depending on the precision you need.

MichaIng, 2025-06-02:
Closing this issue. Feel free to reopen if still relevant.