MichaIng/DietPi #7278: ddns with IPv4 and IPv6

LOGIN-TB, 2024-11-11:
### Creating a feature request
in DDNS Config Menue (dietpi-ddns) we have 3 Options:

IPv4 : Force IPv4 usage for the DDNS update (best compatibility)
IPv6or4 : Do not force any IP family for the DDNS update
IPv6    : Force IPv6 usage for the DDNS update (for modern clients) 

It would be great to have the option to activate both protocols for ddns!

#### Is your feature request related to a problem? Please describe:
In these times it is very helpful if the hosted Servers are available through IPv4 and IPv6

#### Describe the solution you'd like:
Maybe its helpful to have a 4th option with IPv4 and IPv6 and a fallback to IPv4 if IPv6 is not available

#### Describe alternatives you've considered:
n/a

#### Additional context
n/a

Please consider to implement this feature.
Thanks
Holger

MichaIng, 2024-11-11:
Good day Holder! You mean to update both, IPv4 + IPv6 (if possible)? Makes sense: "IPv6and4". We could then do the IPv4 request first, and allow the IPv6 request afterwards to fail, if either the system does not even have an IPv6 address, or the DDNS server is not reachable via IPv6, or it does not accept to add an IPv6 address to the DNS entry.

LOGIN-TB, 2024-11-11:
Hi Micha,

Yes. This is exactly what I mean.  Update v4 and afterwards v6 with the option to fail. Sounds great,

Holger

MichaIng, 2024-12-15:
I implemented it just now: #7309

closed 2024-12-25 by MichaIng