MichaIng/DietPi #7279: Installed Ubuntu on Android using Linux Deploy and set up PiVPN with OpenVPN, using IPv6. The connection succeeds, but there is no network access. Could you please help me identify the issue? Thank you very much!

blueskyno6, 2024-11-11:
Here is the complete English translation of your issue description:

Environment Information
Operating System: Ubuntu Bionic (18.04) running on Android through Linux Deploy
Architecture: ARM64 (aarch64)
OpenVPN Version: OpenVPN 2.4.4 aarch64-unknown-linux-gnu
Network Setup: VPN server configured using OpenVPN. Client connects successfully but has no network access.
Configuration Details
Firewall: Added necessary firewall rules using iptables.
Routing: Set a rule to direct all traffic to the main routing table (ip rule add from all lookup main pref 1).
Client Device: iOS device connecting via OpenVPN client. The server and client establish a connection successfully.
Issue Description: After connecting, the client cannot access any network resources, including both internet and local network.

**### Log Information（Client OpenVPN Log）：**
[Nov 12, 2024, 00:00:01] TLS-TCP send error: Broken pipe
[Nov 12, 2024, 00:00:01] Transport Error: Transport error on '2409:****:****:9840:****:f353:****:****: NETWORK_SEND_ERROR
[Nov 12, 2024, 00:00:01] EVENT: TRANSPORT_ERROR Transport error on '2409:****:****:9840:****:f353:****:****: NETWORK_SEND_ERROR [ERR]
[Nov 12, 2024, 00:00:01] Client terminated, restarting in 5000 ms...
[Nov 12, 2024, 00:00:01] NIP: iOS reported network status unavailable
[Nov 12, 2024, 00:00:01] OS Event: NET UNAVAILABLE (PAUSE): Internet:NotReachable/-- -------
[Nov 12, 2024, 00:00:01] EVENT: PAUSE
[Nov 12, 2024, 00:00:02] NIP: iOS reported network status available
[Nov 12, 2024, 00:00:02] OS Event: NET AVAILABLE (RESUME): Internet:ReachableViaWWAN/WR t------ allow=1
[Nov 12, 2024, 00:00:05] RECONNECT TEST: Internet:ReachableViaWWAN/WR t------
[Nov 12, 2024, 00:00:05] ACTIVE PAUSE
[Nov 12, 2024, 00:00:06] RESUME TEST: Internet:ReachableViaWWAN/WR t------
[Nov 12, 2024, 00:00:06] STANDARD RESUME
[Nov 12, 2024, 00:00:06] EVENT: RESUME
[Nov 12, 2024, 00:00:06] EVENT: RECONNECTING
[Nov 12, 2024, 00:00:06] EVENT: RESOLVE
[Nov 12, 2024, 00:00:06] Contacting [2409:****:****:9840:****:f353:****:****]:443 via TCPv6
[Nov 12, 2024, 00:00:06] EVENT: WAIT
[Nov 12, 2024, 00:00:06] Connecting to [2409:****:****:9840:****:f353:****:****]:443 (2409:****:****:9840:****:f353:****:****) via TCPv6
[Nov 12, 2024, 00:00:06] EVENT: CONNECTING
[Nov 12, 2024, 00:00:06] Tunnel Options:V4,dev-type tun,link-mtu 1571,tun-mtu 1500,proto TCPv4_CLIENT,cipher AES-256-CBC,auth SHA256,keysize 256,key-method 2,tls-client
[Nov 12, 2024, 00:00:06] Creds: UsernameEmpty/PasswordEmpty
[Nov 12, 2024, 00:00:06] Sending Peer Info: IV_VER=3.10_qa IV_PLAT=ios IV_NCP=2 IV_TCPNL=1 IV_PROTO=2974 IV_MTU=1600 IV_CIPHERS=AES-128-CBC:AES-192-CBC:AES-256-CBC:AES-128-GCM:AES-192-GCM:AES-256-GCM:CHACHA20-POLY1305 IV_AUTO_SESS=1 IV_GUI_VER=net.openvpn.connect.ios_3.5.0-6000 IV_SSO=webauth,crtext
[Nov 12, 2024, 00:00:06] VERIFY OK: depth=1, /CN=Easy-RSA CA, signature: ecdsa-with-SHA256
[Nov 12, 2024, 00:00:06] VERIFY OK: depth=0, /CN=localhost_a95c90da-9082-4b30-b53d-7c2dc60002d7, signature: ecdsa-with-SHA256
[Nov 12, 2024, 00:00:06] SSL Handshake: peer certificate: CN=localhost_a95c90da-9082-4b30-b53d-7c2dc60002d7, 256 bit EC, group:prime256v1, cipher: ECDHE-ECDSA-AES256-GCM-SHA384  TLSv1.2 Kx=ECDH     Au=ECDSA Enc=AESGCM(256)            Mac=AEAD
[Nov 12, 2024, 00:00:06] Session is ACTIVE
[Nov 12, 2024, 00:00:06] EVENT: GET_CONFIG
[Nov 12, 2024, 00:00:06] Sending PUSH_REQUEST to server...
[Nov 12, 2024, 00:00:06] OPTIONS: 0 [redirect-gateway] [def1] 1 [dhcp-option] [DNS] [192.168.100.1] 2 [route-gateway] [10.176.139.1] 3 [topology] [subnet] 4 [ping] [15] 5 [ping-restart] [120] 6 [ifconfig] [10.176.139.2] [255.255.255.0] 7 [peer-id] [0] 8 [cipher] [AES-256-GCM]
[Nov 12, 2024, 00:00:06] PROTOCOL OPTIONS: cipher: AES-256-GCM digest: none key-derivation: OpenVPN PRF compress: NONE peer ID: 0 control channel: tls-crypt enabled
[Nov 12, 2024, 00:00:06] EVENT: ASSIGN_IP
[Nov 12, 2024, 00:00:06] NIP: preparing TUN network settings
[Nov 12, 2024, 00:00:06] NIP: init TUN network settings with endpoint: 2409:****:****:9840:****:f353:****:****
[Nov 12, 2024, 00:00:06] NIP: adding IPv4 address to network settings 10.176.139.2/255.255.255.0
[Nov 12, 2024, 00:00:06] NIP: adding (included) IPv4 route 10.176.139.0/24
[Nov 12, 2024, 00:00:06] NIP: redirecting all IPv4 traffic to TUN interface
[Nov 12, 2024, 00:00:06] NIP: adding DNS 192.168.100.1
[Nov 12, 2024, 00:00:06] NIP: allowFamily(AF_INET, 1)
[Nov 12, 2024, 00:00:06] NIP: allowFamily(AF_INET6, 1)
[Nov 12, 2024, 00:00:06] Connected via NetworkExtensionTUN
[Nov 12, 2024, 00:00:06] EVENT: CONNECTED [2409:****:****:9840:****:f353:****:****]:443 (2409:****:****:9840:****:f353:****:****) via /TCPv6 on NetworkExtensionTUN/10.176.139.2/ gw=[/] mtu=(default)

I hope this description helps in creating a clear and complete GitHub issue submission.

Joulinar, 2024-11-11:
I guess this is the wrong place for your question. We support DietPi which is based on Debian and not Ubuntu or Android.

Joulinar, 2024-11-11:
I guess this is the wrong place for your question. We support DietPi which is based on Debian and not Ubuntu or Adnroid.

closed 2024-11-11 by Joulinar