MichaIng/DietPi #7281: No Ethernet auto connect / reconnect

oxoocoffee, 2024-11-15:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9 G_DIETPI_VERSION_SUB=8 G_DIETPI_VERSION_RC=0 G_GITBRANCH='master' G_GITOWNER='MichaIng'`
- Distro version | `bookworm 0`
- Kernel version | `Linux rpi-cam-ap 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (SanDisk Ultra)

#### Additional Information (if applicable)
- Software title | (ifconfig)
- Was the software title installed freshly or updated/migrated? `Fresh`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

- Bug report ID | `d714ba79-6dc9-4de3-9515-fd8e92223326`

#### Steps to reproduce

1. Setup fresh latest DietPi SD card, boot and finish configuration. But do not enable WifI, just LAN port
2. Remove LAN cable and reboot, After network timeout after reboot, login to the unit
3. Plug in the network card
4. run `ifconfig` or `ip address show`. The IP address in never assigned from DHCP server. 
5. Running `sudo ifconfig eth0 down` followed `ifconfig eth0 up` never gets a new address. After full reboot, with linkable is plugged in will get IP address assigned.
6. 

#### Expected behaviour

- If I unplug the cable or of `sudo ifconfig eth0 down` and change IP address reservation on DHCP server and bring the network up or just plug in the cable, it should get new address.

#### Actual behaviour

- ...

#### Extra details

- Nothing happens 
This is the same case for rpi5 beta latest image of this writing

MichaIng, 2024-11-21:
Yes this is expected behaviour, basically the same if a WiFi network is not in rage. Such plug&play for network requires additional tools, like `ifplugd` for events on Ethernet cable connection: https://packages.debian.org/ifplugd
... oh that one supports WiFi as well, so yeah, the de facto standard on Debian, actually.

I am currently not sure what it does by default, but on our forum you will find some threads where it is used to automatically switch between WiFi and Ethernet, depending on connection/availability: https://dietpi.com/forum/search?q=ifplugd
From the configs, maybe you get the idea how just have it configuring `eth0` automatically, once a cable is connected, if this is not its default behaviour anyway. Else feel free to ask.

oxoocoffee, 2024-11-24:
Ok that make sense.

I will try to get it going manually.

If I get it going and working would you be interested to include a option to "add or enable" in future `dietpi-software` or `dietpi-config`?

MichaIng, 2024-12-12:
It is somewhat related to this existing request: #3254
But I do not find the time currently to implement such. Instead of adding it to `dietpi-wifi-monitor`, it made much more sense to instead create a TUI frontent for `ifplugd` (which supports WiFi in the meantime) to individually toggle re-connects or fallbacks to/from WiFi/Ethernet individually.

However, I'll close the issue here, to not have too many open.

closed 2024-12-12 by MichaIng

oxoocoffee, 2024-12-12:
Just so I understand, this will be added at some point to dietpi-wifi-monitor? That would be great.

MichaIng, 2024-12-12:
I would like to have it implemented, but I do not find time, and there are a lot more higher priority issues and requests stocked up. So no ETA, unless someone else starts working on it.

And as said, I would not implement it into `dietpi-wifi-monitor` anymore, but create a new script which does nothing else than creating `ifplugd` configs for some flexible use cases.