MichaIng/DietPi #7283: Dietpi-services restart will not work

AdamF100001, 2024-11-19:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | v9.8.0
- Distro version | bookworm
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | Raspberry Pi 4
- Power supply used | (EG: 5V 1A Standard power supply)
- SD card used | (EG: SanDisk ultra 32GB)

Issue Example 

root@DietPi:~# dietpi-services restart
-bash: /boot/dietpi/dietpi-services: Input/output error

Also, clearing a previous command doesn't work.

root@DietPi:~# clear
-bash: clear: command not found[image]

Thanks

Joulinar, 2024-11-19:
Can you share following info 

```
dmesg -l 0,1,2,3
```

AdamF100001, 2024-11-19:
root@DietPi:~# dmesg -l 0,1,2,3
[    0.589756] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.

AdamF100001, 2024-11-19:
Also, my pi has an uptime of around one week, and then SSH stops working, but the ethernet network port still has the LEDs flashing as if there is a connection.

I have to power cycle the pi to get it working again.

MichaIng, 2024-11-19:
Looks like some data loss along with the crash. Probably the PSU is insufficient or so.

Let's start with the easier one: Can you check about the `clear` command, does it exist?
```sh
ls -l /usr/bin/clear
```

AdamF100001, 2024-11-19:
root@DietPi:~# ls -l /usr/bin/clear
-rwxr-xr-x 1 root root 67680 May  7  2023 /usr/bin/clear

MichaIng, 2024-11-19:
The `arm64` version is surprisingly large, but looks good. Can you execute that one directly?
```sh
/usr/bin/clear
```
If so, please check our PATH:
```sh
echo $PATH
which clear
```

AdamF100001, 2024-11-19:
root@DietPi:~# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
root@DietPi:~# which clear
/usr/bin/clear

MichaIng, 2024-11-19:
That looks all good. And running `/usr/bin/clear` works, while `clear` still produces that "command not found" error? Does this produce any different path?
```sh
command -v clear
```

AdamF100001, 2024-11-19:
root@DietPi:~# command -v clear
/usr/bin/clear

AdamF100001, 2024-11-19:
/usr/bin/clear

Also cleared the terminal

MichaIng, 2024-11-19:
And `clear` still produces this error?

AdamF100001, 2024-11-19:
https://github.com/user-attachments/assets/1d1a6fae-0b53-4b09-906b-2a4c041ba8dd

Clear command cleared the screen. As shown in this video, I recorded running the command.

MichaIng, 2024-11-19:
Okay, so that one is working again.

Then let's check about the `dietpi-service` script. Does it still produce the input/output error?
```sh
dietpi-services restart
```
And if so, can you run it like that:
```sh
bash -x /boot/dietpi/dietpi-services restart
```
It will produce a lot of output until it run into the error. In case, only the last page is relevant.

AdamF100001, 2024-11-19:
dietpi-services restart

bash -x /boot/dietpi/dietpi-services restart

Please see video of command outputs.

https://github.com/user-attachments/assets/a3a0e3ca-1f81-458b-a6ba-636c70a72f14

AdamF100001, 2024-11-19:
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /lib/systemd/system/htpc-manager.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /usr/lib/systemd/system/htpc-manager.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /run/systemd/generator.late/htpc-manager.service ]]
+ continue
+ [[ -n '' ]]
+ unset -v 'aSERVICE_NAME[107]'
+ for i in "${!aSERVICE_NAME[@]}"
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /etc/systemd/system/node_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /usr/local/lib/systemd/system/node_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /lib/systemd/system/node_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /usr/lib/systemd/system/node_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /run/systemd/generator.late/node_exporter.service ]]
+ continue
+ [[ -n '' ]]
+ unset -v 'aSERVICE_NAME[108]'
+ for i in "${!aSERVICE_NAME[@]}"
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /etc/systemd/system/raspberrypi_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /usr/local/lib/systemd/system/raspberrypi_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /lib/systemd/system/raspberrypi_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /usr/lib/systemd/system/raspberrypi_exporter.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /run/systemd/generator.late/raspberrypi_exporter.service ]]
+ continue
+ [[ -n '' ]]
+ unset -v 'aSERVICE_NAME[109]'
+ for i in "${!aSERVICE_NAME[@]}"
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /etc/systemd/system/docker.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /usr/local/lib/systemd/system/docker.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /lib/systemd/system/docker.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /usr/lib/systemd/system/docker.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /run/systemd/generator.late/docker.service ]]
+ continue
+ [[ -n '' ]]
+ unset -v 'aSERVICE_NAME[110]'
+ for i in "${!aSERVICE_NAME[@]}"
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
+ [[ -e /etc/systemd/system/k3s.service ]]
+ continue
+ for j in /{{etc,usr/local/lib,lib,usr/lib}/systemd/system,run/systemd/generator.late}/"${aSERVICE_NAME[$i]}.service"
[... 803 lines cut ...]
+ ((  0 == 0  ))
+ output_string+='\e[90m[\e[0m\e[32m  OK  \e[90m]\e[0m '
+ ainput_string+=('\n')
+ Print 1
+ [[ 1 == 1 ]]
+ [[ -n DietPi-Services ]]
+ output_string+='\e[90mDietPi-Services |\e[0m '
+ (( i=1 ))
+ (( i<3 ))
+ output_string+='restart : cron'
+ (( i++ ))
+ (( i<3 ))
+ output_string+='\n'
+ (( i++ ))
+ (( i<3 ))
+ echo -ne '\e[0m\r\e[J\e[90m[\e[0m\e[32m  OK  \e[90m]\e[0m \e[90mDietPi-Services |\e[0m restart : cron\n\e[0m'
[  OK  ] DietPi-Services | restart : cron
+ unset -f Print
+ exit 0
+ G_EXIT
+ declare -F G_EXIT_CUSTOM
+ cd /tmp
+ [[ ! -d /tmp/DietPi-Services ]]
+ rm -R /tmp/DietPi-Services
root@DietPi:~#

MichaIng, 2024-11-19:
So this also works now. Guess the problems are solved? Maybe in temporary instability. You use the official RPi PSU or a good brand one?

AdamF100001, 2024-11-20:
I am currently using an official RPi PSU.

Could you please recommend and good alternative one so I can replace my current one?

Thanks

MichaIng, 2024-11-20:
Hmm, the official RPi PSU should provide sufficient power. Do you use and SD card or USB flash drive for the OS? Or other USB devices?

In any case, if something similar happens again, like the crash you had, report back. We can enable persistent system logging, to check what happened before the crash.

AdamF100001, 2024-11-20:
I have ordered a new Raspberry Pi 27W USB-C Power Supply.

It should arrive in a few days. I will report back in a weeks time.

Can you please keep this ticket open for the time being?

It boot off an SD Card 32GB SanDisk.

I also have a USB 3.0 SSD connected for media storage and a UPS HAT connected

MichaIng, 2024-11-21:
Is the USB SSD powered through the USB ports only, or via dedicated PSU?

AdamF100001, 2024-11-21:
> Is the USB SSD powered through the USB ports only, or via dedicated PSU?

It's a Crucial X6 2TB Portable SSD - Up to 800MB/s - PC and Mac - USB 3.2 USB-C External Solid State Drive - CT2000X6SSD9 - Black

Deal: Crucial X6 2TB Portable SSD - Up to 800MB/s - PC and Mac - USB 3.2 USB-C External Solid State Drive - CT2000X6SSD9 - Black https://amzn.eu/d/i0wt8c1

The replacement psu I ordered for my pi arrived today. I will test it out over the weekend.

I will report back mid week next week once it's been running for a few days.

MichaIng, 2024-11-21:
Ah, an NVMe SSD in a portable case, nice. Not sure how much power these draw, compared to a 2.5" SSD. If it is similar, it may be related to the instability, since the RPi USB ports do no provide enough power to drive such drives reliably. Sometimes it works, sometimes not, sometimes it works until a kernel upgrade, or power peak.

... @Joulinar is testing it just now with another NVMe SSD in a 😄 and indeed, the additional load is similar to a 2.5" SATA SSD powered via USB. So in case you face a crash again, you might test it without that SSD. Difficult in this case to power it externally, but sadly for RPi's there is not other reliable way.

AdamF100001, 2024-11-21:
I purchase the drive back in 2022 and I had no issues until now.

AdamF100001, 2024-11-29:
The new Raspberry Pi 27W USB-C Power Supply has been installed and powered up for nearly six days, and it's stable.

Please can you resolve this ticket.

Thanks 

[image]

AdamF100001, 2024-11-29:
Please can you reopen the ticket as the network connection on the raspberry pi just disconnected again.

I had to power cycle to get it working again.

Thanks

MichaIng, 2024-12-06:
You can close and reopen issues by yourself, isn't it? 😄 

However, as crash happened again, best is to enable permanent system logs, to see what is going on:
```sh
dietpi-software uninstall 103
mkdir /var/log/journal
reboot
```
After above steps `/var/log` is no tmpfs/RAM disk anymore, and `/var/log/journal` should contain the journal file(s) used by `journalctl`. When the crash happens again, check `journalctl` on next book. It should contain logs from the last boot session, and hopefully something about what happened before the crash.

You can limit size of the journal, like 2 files in rotation with the older one removed after 7 days, to have 7-14 days of logs at any time:
```sh
echo -e 'SystemMaxFiles=2\nMaxFileSec=7days' > /etc/systemd/journald.conf.d/99-size.conf
systemctl restart systemd-journald
```

AdamF100001, 2024-12-07:
The SD card will not boot up at all now.

I will re-image the card today and monitor full system logs as stated in you previous posts.

Thanks

MichaIng, 2024-12-12:
@AdamF100001 
You enabled permanent system logging already? The it is `journalctl`. The system logs contain timestamps, but also kind of delimiters between the boot sessions. That should help to find the part and last logs before the crash.

MichaIng, 2024-12-12:
This is from the current boot session only without a crash or even startup log. The `systemctl` logs start at `Dec 11 22:09:57`? Because after enabling permanent system logging and doing a reboot once, it should be cut only after 7 days.

AdamF100001, 2024-12-18:
The system is still accessible via SSH, but the services have stopped running.

dietpi-services restart generates this error.

root@DietPi:~# dietpi-services restart
-bash: /boot/dietpi/dietpi-services: Input/output error

Joulinar, 2024-12-18:
Hmm there seems to be I/O issue on the system.

If not already rebooted can you check for kernel error messages 

`dmesg -l 0,1,2,3`

AdamF100001, 2024-12-18:
root@DietPi:~# dmesg -l 0,1,2,3
-bash: dmesg: command not found

Joulinar, 2024-12-18:
You are using SD card? Probably it's going to die

AdamF100001, 2024-12-18:
Yes, it's an SD card I'm booting off.

I will replace it over the weekend, and hopefully, it should be stable.

Thanks

AdamF100001, 2024-12-30:
Hi

Please can you close this ticket.

I replaced the SD card, it's been running for over a week without any issues.

Thanks

closed 2024-12-30 by AdamF100001