MichaIng/DietPi #7284: Added on banner silly "Up since"

dentaku65, 2024-11-21:
On /boot/dietpi/.dietpi-banner_custom
Add:
`stat -c %w / |cut -f1 -d "."`

Enable on /boot/dietpi/.dietpi-banner
`aDESCRIPTION[10]='Up since'`

[image]

MichaIng, 2024-11-21:
Based on the command, this is to show how old the image is, right?

My first thought was: "but we have an uptime option already?", then recognised that this is something different 😄.
[image]

dentaku65, 2024-11-22:
Hi MichaIng,
yep, command shows DietPI first run and operate since. Can be considered "age".
And yes my DietPI is operating since "2018-03-13 23:51:07" without issue :-) 
Den

MichaIng, 2024-11-22:
To be precise, for new images, it is the time at which the image was generated, or more precisely, at which our build script creates some initial directories, before running `mmdebstrap` to install the initial Debian base system 🙂.

For a first boot timestamp, there are some log files, like `/var/tmp/dietpi/logs/fs_partition_resize.log` which could be used. But they are of course not guaranteed to exist in any case. `/boot/dietpi/.hw_model` might be also an idea. It is re-created on every boot, but never removed, AFAIK.

dentaku65, 2024-11-22:
Well, English is not my native language, I just simplified in "Up since". No date found in the logs on `/var/tmp/dietpi/logs`
A question: because I've always did the upgrade from version to version, why I don't have the up-to-date banner but the one still coming from an old release?

MichaIng, 2024-11-22:
I meant to use the birth date of those log files, like you currently to with `/`:
```sh
stat -c '%w' /var/tmp/dietpi/logs/fs_partition_resize.log | cut -f1 -d '.'
```

About your question: `/boot/dietpi/.dietpi-banner_custom` is a config file that is not touched with DietPi updates, so any custom banner entry survives updates, if this is what you mean. Only `/boot/dietpi/func/dietpi-banner` is updated.

dentaku65, 2024-11-22:
> I meant to use the birth date of those log files, like you currently to with `/`:
> 
> ```shell
> stat -c '%w' /var/tmp/dietpi/logs/fs_partition_resize.log | cut -f1 -d '.'
> ```

Got it the point.

> 
> About your question: `/boot/dietpi/.dietpi-banner_custom` is a config file that is not touched with DietPi updates, so any custom banner entry survives updates, if this is what you mean. Only `/boot/dietpi/func/dietpi-banner` is updated.

In this case I meant I don't have "Uptime" from your screenshot... but perhaps I commented out on some my experiment during the years which I forgot atm :-)

MichaIng, 2024-11-22:
Uptime is not enabled by default. Run `dietpi-banner` command to toggle options.

dentaku65, 2024-11-22:
Done!