MichaIng/DietPi #7285: Impossibility to update DietPi

boterocamilo, 2024-11-21:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- Date           | 2024-11-21 18:28:16
- Program name   | DietPi-Update
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- DietPi version | v9.7.1 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=0)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | RPi 4 Model B (aarch64)
- Power supply used |  5V 1A
- SD card used | SSD 1TB - WD Elements

#### Additional Information (if applicable)
- Software title | DietPi uptate to v.9.8.0
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? I did not try

- Bug report ID | `1568fffc-d7e8-4414-b734-9ede76bcaa07`

#### Steps to reproduce

1. Run dietpi-update
2. Error occurs.
3. It must exit since it does not install the update.

#### Expected behaviour

- Update DietPi

#### Actual behaviour

- It is not updating

#### Extra details

- #### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.nodesource.com/node_18.x nodistro InRelease
Hit:4 https://downloads.plex.tv/repo/deb public InRelease
Hit:5 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:6 https://deb.debian.org/debian bookworm-backports InRelease
Get:7 https://download.mono-project.com/repo/debian buster InRelease [5874 B]
Hit:8 https://dietpi.com/apt bookworm InRelease
Hit:9 https://dietpi.com/apt all InRelease
Ign:10 https://download.docker.com/linux/ubuntu bookworm InRelease
Hit:11 https://archive.raspberrypi.com/debian bookworm InRelease
Err:12 https://download.docker.com/linux/ubuntu bookworm Release
  404  Not Found [IP: 13.33.141.88 443]
Reading package lists...
E: The repository 'https://download.docker.com/linux/ubuntu bookworm Release' does not have a Release file.
```
- IMPORTANT NOTE: Docker was originally installed using DietPi-Software

Joulinar, 2024-11-21:
Not sure why but you added a docker repository for Ubuntu. And there the release Bookworm. First, DietPi is based on Debian and Bookworm is not a Ubuntu release. Any reason why you added it manually?

boterocamilo, 2024-11-21:
> Not sure why but you added a docker repository for Ubuntu. And there the release Bookworm. First, DietPi is based on Debian and Bookworm is not a Ubuntu release. Any reason why you added it manually?

Actually I do not remember installing it manually.
However, I did run the following:
`sudo rm /etc/apt/sources.list.d/docker.list`
Which contained the Ubuntu reference.

After, I ran `dietpi-update` and error appeared and selected `re-run`

Now I have the v9.8.0.

There is just one package that has been kept back (rpi-eeprom) and cannot be updated.

closed 2024-11-21 by boterocamilo

Joulinar, 2024-11-21:
> There is just one package that has been kept back (rpi-eeprom) and cannot be updated.

```
apt install --reinstall rpi-eeprom
```

MichaIng, 2024-11-21:
It requires one of the new firmware packages now, so this is expected, until migrating to new kernel/firmware stack: https://github.com/MichaIng/DietPi/issues/6676